solvespace/solvespace #570: Add missing constraints to toolbar

Timmmm, 2020-03-28:
SolveSpace version: 3.0~645353cb
Operating system: Windows 10

Some constraints are not in the toolbar, which makes them not very discoverable. I think they should all be added. As far as I can see the missing ones are:

* Length Ratio
* Length Difference
* At midpoint
* Lock point where dragged
* Comment

(Also maybe rename "Lock point where dragged" to "Current position"? Assuming that's what it does.)

whitequark, 2020-03-28:
I think they're fine as it is for two reasons:

  1. if we make the toolbar even taller, it will get cropped by the window on smaller screen resolutions, which is even worse;
  2. menus literally exist for discoverability.

whitequark, 2020-03-28:
What I think would be far more useful is implementing #121.

closed 2020-03-28 by whitequark