solvespace/solvespace #571: Segfault when removing bezier spline point

Timmmm, 2020-03-28:
### System information

SolveSpace version: 3.0~645353cb
Operating system: Windows 10

### Expected behavior

Add a bezier curve with a few segments. Right click one of the vertices, click "Remove spline point" or something like that. It shouldn't crash.

### Actual behavior

You get a segfault in `mouse.cpp:651` roughly here:

```
            Request *r = SK.GetRequest(gs.point[0].request());
            int index = r->IndexOfPoint(gs.point[0]);
            if((r->type == Request::Type::CUBIC && (index > 1 && index < r->extraPoints + 2)) ||
                    r->type == Request::Type::CUBIC_PERIODIC) {
                menu->AddItem(_("Remove Spline Point"), [&]() {
                    int index = r->IndexOfPoint(gs.point[0]);
```

Because `r` is 0xccccccccc which I think possibly means uninitialised in MSVC? Anyway it's not pointing to a valid object so it crashes later in `r->IndexOfPoint()`.

Timmmm, 2020-03-28:
I'm guessing you probably didn't mean to take a reference to `r` in that lambda! Changing the capture to `[&, r]` fixes it. Though really, passing raw pointers around like this is asking for trouble!

whitequark, 2020-03-28:
Yes, unfortunately the SolveSpace codebase has a mix of legacy code with more modern C++ and it doesn't always work well. Could you prepare a pull request? I didn't look into your fix closely but it sounds correct to me.

PR #572 (2020-03-28): Fix segfault in Remove Spline Point context menu
This was incorrectly capturing `r` by reference and using it after it left its scope. Changed to capture by value, and also explicitly capture `this` in case we were accidentally capturing any other scope variables by reference.

Fixes #571
commit 485fd012 (2020-03-28): Fix segfault in Remove Spline Point context menu

This was incorrectly capturing `r` by reference and using it after it left
its scope. Changed to  capture by value, and also explicitly capture `this`
in case we were accidentally capturing any other scope variables by reference.

Fixes #571
closed 2020-03-28 by whitequark