solvespace/solvespace #573: Allow constraints to apply to multiple objects

Timmmm, 2020-03-28:
### System information

SolveSpace version: 3.0~645353cb
Operating system: Windows 10

### Expected behavior

Select 3 circles, click "equal size" constraint, you get an error.

### Actual behavior

It should constrain all the circles to be the same size as the "first one".

### Additional information

Obviously there is some ambiguity in which circles to constrain to which others, but I feel like taking a random guess is better than making people add every constraint manually.

phkahler, 2020-03-28:
I've wondered about the best way to do that. If you have 10 circles, should one of them have 9 constraints - one to each of the others - or is it better to constrain circles 1,2   2,3,  3,4   4,5 .... etc?

To me it's a question about the constraint solver.

What you propose will still require the same number of constrain equations, but the UI will be nicer.

whitequark, 2020-03-28:
> I feel like taking a random guess is better than making people add every constraint manually.

I'm strongly opposed to taking a random guess. Software is unpredictable enough as it is without it rolling dice for you. Fortunately, in this case we can do it deterministically.

The main problem is that most constraints can't be handled in the same way as equality constraints.

Timmmm, 2020-03-29:
Looks like #530 is an attempt to fix this.

77maxikov, 2020-03-29:
Yes, it is about constraining multiple objects - helps me save some time

closed 2022-12-31 by phkahler