solvespace/solvespace #576: Add analysis for non-closed contours

Timmmm, 2020-03-29:
### System information

SolveSpace version: 3.0~64717101
Operating system: Windows 10

### Details

I recently had a sketch that *looked* like it was a closed contour, and nothing was obviously wrong, but I still got the error that it wasn't a closed contour. Eventually I had to just delete it and start again.

It would be nice if there was an analysis mode that highlighted the bits that were leaking. I thought maybe "Analysis->Show Naked Edges" would do that? But it didn't seem to. It's not clear what "Naked" means there either - probably needs renaming.

phkahler, 2020-03-29:
When you say a contour "looked" closed, that is the key. An example might be that there are segments with slightly different end points that look connected unless you zoom in. It may be difficult for software to figure out what was intended but didn't happen. An example slvs file would be helpful here.

whitequark, 2020-03-29:
> It would be nice if there was an analysis mode that highlighted the bits that were leaking.

That's already done: see the red error message (which you can use if you're in that group), and the "err" in the property browser (which you can use if you're in another group):

[image]

> I thought maybe "Analysis->Show Naked Edges" would do that? But it didn't seem to. It's not clear what "Naked" means there either - probably needs renaming.

"Show Naked Edges" highlights the edges of triangles comprising the solids that are not connected to adjacent triangles, as a result of a failure of a (extrusion, revolution, or boolean) operation over solids.

whitequark, 2020-03-29:
Did you perhaps inadvertently turn this off in the settings
[image]
?

phkahler, 2020-03-30:
@Timmmm Are you looking for a better indication of where the contour is open?

Timmmm, 2020-03-31:
Ah this is helpful thanks. To clarify:

1. It was obvious that there was a problem - I do get the "not closed contour, or not all same style" message.
2. I didn't realise that the location of that message was significant! This is not really obvious.
3. Because the non-closed bit was really small, the red highlight wasn't visible. It's also not a very obvious colour.

So I think the best solution would be:

1. Add an arrow to the error message, so it is more obvious that it is actually attached to something.
2. Make the red "unclosed" line a bit brighter. If you also make it wider than the green vertex squares, and give it a round end cap then it would show up even if the leak is very small.

Re "naked" - maybe rename to "Show broken edges"? "Show unclosed edges"? "Show disconnected edges"?

I'll try and make an example file tonight.

Timmmm, 2020-03-31:
Hmm I'm not sure what I did to do this, so I'll close it for now and reopen if it happens again.

whitequark, 2020-03-31:
> * Add an arrow to the error message, so it is more obvious that it is actually attached to something.

It's not trivial to add this because you have to position it nicely in the space so it wouldn't interfere with other entities. What about highlighting the point somehow, e.g. circling it?

> Re "naked" - maybe rename to "Show broken edges"? "Show unclosed edges"? "Show disconnected edges"?

I agree "naked" seems like an opaque term if you're approaching this from the perspective of a CAD user. I think "Show broken edges" sounds good, since you pretty much never want to have those in your model.

Timmmm, 2020-04-01:
> What about highlighting the point somehow, e.g. circling it?

Yeah that'd be good!

whitequark, 2020-04-01:
Let's reopen the issue then.