solvespace/solvespace #577: Add chamfer and fillet tools

Timmmm, 2020-03-30:
It should be possible to select an arbitrary set of edges in 3D, and then click "chamfer/fillet", and have it nicely modified. Sometimes you can do this in a sketch with arcs, but often it is very tedious or impossible.

Most professional CAD tools include these as basic operations. For some reason they are usually separate tools but I think it makes more sense to have it as one tool with a "chamfer style" option which is "straight" or "rounded" (and a radius option).

whitequark, 2020-03-31:
This is unlikely to ever happen. The only open-source backend that natively supports chamfers and fillets is OpenCASCADE, and it (a) has an incompatible license, and (b) is a rather poor quality codebase, since it exists to sell consulting services.

I've looked into extending SolveSpace's NURBS backend a few years ago, and my conclusion was that the amount of work was prohibitive for a volunteer-run open-source project, requiring several people employed full-time for multiple years.

closed 2020-03-31 by whitequark

ruevs, 2020-03-31:
Hmm... I've personally "seen" you answer this twice. Perhaps a FAQ is in order?

whitequark, 2020-03-31:
Good idea.

Timmmm, 2020-03-31:
That is a shame!

whitequark, 2020-03-31:
Yes, it's really unfortunate. A few years ago I was working much more closely on SolveSpace together with one full-time developer (@Evil-Spirit), and based on our discussions back then, I believe this is just not feasible for this project at this time.

Timmmm, 2020-04-01:
What were the main problems, just out of interest?

whitequark, 2020-04-01:
Mostly it's just a lot of work to handle every possible case, and it's also work that needs to be done either with someone specializing in NURBS, or someone willing to invest a *lot* of time to learn the necessary math. I'm really happy to have @phkahler on board, who has been able to make very nice improvements to the NURBS code, but features like chamfers or fillets require a lot more effort to work well.

phkahler, 2020-04-03:
@Timmmm I can suggest reading the entire thread for issue #453 to get a little understanding of the issues.

There are several problems, the primary one being the way surfaces and entities are represented internally be SolveSpace. Surfaces are completely separate from the editable/constrainable entities you can interact with. When you do an extrude or other operation that creates a solid volume, the new entities are created separately from the surfaces. When entities are dragged new surfaces are created. This disconnect means that the first problem for a generalized fillet or chamfer tool is to identify which surfaces are going to be modified by the operation - it might be possible, but internally it's not set up for that.

The more conventional challenges are defining how existing surfaces need to be trimmed for the new edge features, figuring out what to do at corners (there are many special cases and different ways to cut there), and making sure there is proper continuity along the chamfer with regard to the U,V coordinates interpolated along both edges. There are a lot of papers written on this stuff, so we could pick something and try but then we end up facing the problem of SS internal representation that makes these things harder.

Having said all that, I was working on some new tools that may work for some of the more common use cases. These will all work at the time of extrusion and include: adding draft angles to an extrusion, adding a chamfer of fillet to the top or bottom of an extrusion, and possibly a frustum tool. The issues with those (except frustum) is that the entities will not take on the new geometry of the surfaces. In some cases that's fine while in others it's not. I hope to get back to that effort to see what can be worked out.

fortinmike, 2020-12-07:
That is indeed a shame :(

For my personal (beginner) maker needs, it _looks like_ fillets and chamfers are the only tools missing in Solvespace for me to make the designs I have in mind, which is mostly small objects and mechanical parts. That's too bad because I like Solvespace quite a bit! Much nicer to use than FreeCAD in my opinion. The workflow and keyboard shortcuts are just great.

@phkahler I'd be very interested in tools to provide some fillet / chamfer capability even if it's not the "real deal". Consider this a +1 from me! Are you still working on this to some extent?

phkahler, 2020-12-07:
@fortinmike After v3.0 is done I want to get back to making new construction tools. I'm not sure which ones will make it but everything I've mentioned in various threads is still on my mind.

pinpox, 2023-03-08:
> @fortinmike After v3.0 is done I want to get back to making new construction tools. I'm not sure which ones will make it but everything I've mentioned in various threads is still on my mind.

3.1 is out, any news on adding fillet/chamfer tools?

ruevs, 2023-03-09:
#149 for cross reference.

g19fanatic, 2026-04-06:
@ruevs . The token budget isn't large at all. i think i've spent $15 in tokens on this so far. I use these toolsets for a lot of my development and have some custom tooling setups that enable quick turnarounds.

I think i'm almost to an mvp (have been iterating through bugs). The screenshot you show is already some ways behind too.
I'll clean everything up after the mvp is there. 

I wasn't going to mention anything or put in a PR until a felt it was useful _just_ after the simple cases.
Ya exposed me! hah

phkahler, 2026-04-06:
@ruevs  I looked at a few of the documentation files it generated. Wish we had some good documentation like that, but they're also full of things I'd want to correct, or add detail. Then the code seemed really long for what it's seems able to do.

@g19fanatic  Did you see my PR #1501 ??  It contains an outline of how I was starting to approach this. As I was getting into it, it turns out there are some details missing. A lot of book keeping is necessary too, and if planned out I think it can be done fairly simply.

One thing to not is that a chamfer will create two new curves offset within the surfaces joined by the original curve. A fillet will need those same two new curves, and can use the points of the original curve as the control points for the "middle" of the fillet (with proper weight modification).

All of this needs to be tested with some kind of fake decision about which edges to apply changes to. Having the capability and plumbing it into solvespace are IMHO two very separate projects. I don't think AI is up to the task in either case, but I'm curious to see how it goes.

ruevs, 2026-04-06:
> but they're also full of things I'd want to correct, or add detail.

Indeed me too. But at the same time I would NEVER spend the time and effort to write this kind of documentation.

> [@g19fanatic](https://github.com/g19fanatic) Did you see my PR [#1501](https://github.com/solvespace/solvespace/pull/1501) ?? It contains an outline of how I was starting to approach this.

The LLM saw it ;-) And "thought" about it a lot :-)

<details>
Code	File	Line	Column

1. phkahler's PR #1501 crash was due to COINC_OPP degeneracy from standard boolean	C:\PAR\Projects\SolveSpace\git\findings\39-risk-assessment.md	472	19

### #1501 — "WIP add functions for creating chamfers and fillets to the NURBS kernel" (Draft PR, Dec 10, 2024)	C:\PAR\Projects\SolveSpace\git\findings\14-github-community-requests.md	123	6

**URL:** https://github.com/solvespace/solvespace/pull/1501	C:\PAR\Projects\SolveSpace\git\findings\14-github-community-requests.md	124	56

**Key technical insights from PR #1501 discussion:**	C:\PAR\Projects\SolveSpace\git\findings\14-github-community-requests.md	157	35

- 8 ❤️ reactions on PR #1501 from multiple community members including maintainer ruevs	C:\PAR\Projects\SolveSpace\git\findings\14-github-community-requests.md	184	25

- The phkahler:fillet branch (PR #1501) has only 3 commits, last of which says "Will crash"	C:\PAR\Projects\SolveSpace\git\findings\14-github-community-requests.md	197	35

   - PR #1501 is draft but the scaffolding exists	C:\PAR\Projects\SolveSpace\git\findings\14-github-community-requests.md	228	10

| #1501 | WIP add functions for creating chamfers and fillets to the NURBS kernel | Draft PR | Dec 2024 | phkahler |	C:\PAR\Projects\SolveSpace\git\findings\14-github-community-requests.md	242	4

The SolveSpace community has clearly wanted chamfer/fillet since at least 2017. The feature is officially listed as "commonly requested" by maintainer ruevs. The core contributor phkahler has attempted an implementation (PR #1501, Dec 2024), identified the key algorithmic challenges, and offered architectural guidance. His final advice: "Next time I try I might limit myself to flat surfaces only just to get the basics down." This suggests the MVP strategy should be:	C:\PAR\Projects\SolveSpace\git\findings\14-github-community-requests.md	248	226

5. phkahler's PR #1501 concluded this approach tends to crash	C:\PAR\Projects\SolveSpace\git\findings\16-boolean-pipeline.md	265	19

**Critical decision:** phkahler's failed PR (#1501) revealed that using the standard	C:\PAR\Projects\SolveSpace\git\findings\21-chamfer-group-design.md	56	47

From phkahler's PR #1501 (the only prior attempt):	C:\PAR\Projects\SolveSpace\git\findings\21-chamfer-group-design.md	564	21

- GitHub PR #1501 by phkahler (findings/14-github-community-requests.md)	C:\PAR\Projects\SolveSpace\git\findings\21-chamfer-group-design.md	595	14

- Finding 14 — GitHub community requests (phkahler PR #1501 analysis)	C:\PAR\Projects\SolveSpace\git\findings\22-fillet-group-design.md	826	56

6. **phkahler's existing fillet branch (PR #1501)** — used the same face-pair approach.	C:\PAR\Projects\SolveSpace\git\findings\23-edge-entity-design.md	207	45

### PR #1501 — "WIP add functions for creating chamfers and fillets to the NURBS kernel"	C:\PAR\Projects\SolveSpace\git\findings\28-solvespace-chamfer-attempts.md	14	9

- **URL**: https://github.com/solvespace/solvespace/pull/1501	C:\PAR\Projects\SolveSpace\git\findings\28-solvespace-chamfer-attempts.md	15	58

This is the **complete content** of the new file created in PR #1501:	C:\PAR\Projects\SolveSpace\git\findings\28-solvespace-chamfer-attempts.md	38	65

### Confirmed Working in PR #1501	C:\PAR\Projects\SolveSpace\git\findings\28-solvespace-chamfer-attempts.md	243	30

   equivalent to SolveSpace's `MakeFromBoolean`, but the key insight from phkahler's PR #1501	C:\PAR\Projects\SolveSpace\git\findings\30-cgal-and-geometric-algorithms.md	375	90

3. This is exactly what phkahler's PR #1501 crashed on	C:\PAR\Projects\SolveSpace\git\findings\32-group-generate-trace.md	224	40

The cleanest approach (as phkahler confirmed via PR #1501) is:	C:\PAR\Projects\SolveSpace\git\findings\32-group-generate-trace.md	262	54

- **Result: boolean failure** (as confirmed by phkahler in PR #1501)	C:\PAR\Projects\SolveSpace\git\findings\32-group-generate-trace.md	634	64

- phkahler (PR #1501): "Trim polygon update is the hard part"	C:\PAR\Projects\SolveSpace\git\findings\32-group-generate-trace.md	648	17

The reason: the standard `MakeFromBoolean/MakeFromDifferenceOf` pipeline fails for chamfer because the chamfer surface is coincident with the adjacent original faces, causing `COINC_OPP` degeneracy in the boolean classifier (confirmed by phkahler's crashed PR #1501).	C:\PAR\Projects\SolveSpace\git\findings\36-synthesis-part1.md	164	262

10. **phkahler's PR #1501 confirms the hardest parts**: partial shell modification is very difficult; starting with flat surfaces only is the right MVP scope.	C:\PAR\Projects\SolveSpace\git\findings\36-synthesis-part1.md	310	22

| #1501 | Draft PR: WIP chamfer/fillet implementation | Dec 2024 |	C:\PAR\Projects\SolveSpace\git\findings\37-synthesis-part2.md	156	4

### PR #1501: The Only Implementation Attempt	C:\PAR\Projects\SolveSpace\git\findings\37-synthesis-part2.md	158	9

**Confirmed**: phkahler's PR #1501 crashed using boolean approaches.	C:\PAR\Projects\SolveSpace\git\findings\37-synthesis-part2.md	250	31

[... cut ...]
- Finding 28: phkahler PR #1501 analysis	C:\PAR\Projects\SolveSpace\git\findings\38-synthesis-part3.md	973	28

analysis, community discussions (phkahler PR #1501), and analogous implementations in	C:\PAR\Projects\SolveSpace\git\findings\39-risk-assessment.md	7	47

- phkahler confirmed in PR #1501 that vertex caps are extremely difficult	C:\PAR\Projects\SolveSpace\git\findings\39-risk-assessment.md	119	29

- **phkahler's `fillet` branch** (PR #1501): Contains 112 lines of `src/srf/chamfer.cpp`	C:\PAR\Projects\SolveSpace\git\findings\39-risk-assessment.md	387	39

</details>

ruevs, 2026-04-07:
@g19fanatic since issues on your fork are not enabled I'll post this here:
https://github.com/solvespace/solvespace/pull/1410#issuecomment-1731480056
It's a model with many corner cases (pun intended :-D ) but with only flat surfaces (I created it while implementing the `intersection` boolean operation, and as you see both `union` and `intersection` still have bugs).

You can change the second extrusion group `g005_extrude` to `difference` mode. It should provide plenty of fun corners to fillet and chamfer for you and the LLM ;-)

g19fanatic, 2026-04-08:
@ruevs   

[image]

works for both chamfers and fillets!
Still have more to do though