solvespace/solvespace #578: Feature Request: Entering a Negative Contraint Flips the Extrude Direction

kk6mrp, 2020-04-09:
Many times when extruding a surface it is difficult to move the extrusion to the other side of the workplane. For this reason it would be nice if I could enter a negative constraint when extruding for the purpose of flipping the extrusion direction. I don't mean to keep the value negative, but just as an indicator to SolveSpace to extrude the other way.

phkahler, 2020-04-10:
SolveSpace will extrude toward the viewer, so if you want an extrusion to go the other way just rotate the model so you are viewing the sketch from the other side prior to extruding it. Changing the extrusion type to "difference" also flips the direction, but will flip it back when unchecked.

gituuu, 2020-04-25:
I agree this would be really useful but as there is an easy workaround I personally don't mind it being put on the backburner.

closed 2026-01-02 by ruevs