solvespace/solvespace #587: Frequent build timeouts for arm64 snap

ppd, 2020-04-25:
Travis "deploy" stage for arm64 snap fails often with message: "No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself."

It's always in the CMake install stage, which is invoked by snapcraft. No build-error, just no output for ten minutes. There's nothing extraordinary going on and the hardware seems "OK fast", so it shouldn't choke on a copy of a font or icon.

WIP, obviously...

phkahler, 2020-04-25:
@ppd Thank you for spending so much time making SolveSpace available via Snap. I build from source, but the more ways there are for other people to get it the better!

ppd, 2020-04-26:
There's something wrong with stdout/stderr capturing on arm64. It's not really stalling, as even output from simple echo loops fails to show after a while. 

It would be simple to redirect snapcrafts output to a logfile, one would think. Then there's less output and maybe that will help. But no; apparmor confines its execution and makes delegation of file descriptors impossible, which means no piping etc.

It's not a huge deal. Travis will eventually fix their stuff without a doubt. Until then, it's maybe wise to pause the edge builds for arm64. It will fail all the time for no good reason and that's not so helpful, I think.

whitequark, 2020-04-26:
@ppd By any chance, would [this](https://github.com/solvespace/solvespace/blob/master/.travis.yml#L18) also help arm64?

ppd, 2020-04-26:
@whitequark Same problem as with the logfile, I think. Apparmor will shoot it down with sth like `Log: apparmor="DENIED" operation="file_inherit" profile="/usr/lib/snapd/snap-confine" [...]`

whitequark, 2020-04-26:
> Until then, it's maybe wise to pause the edge builds for arm64.

Let's do that. Let's also remove the builds for arm64 from the store so that people don't unintentionally download outdated buggy versions. (There's probably very few people who can run it in the first place at the moment, too.)

ppd, 2020-04-27:
Support request regarding snapcraft output redirection: https://forum.snapcraft.io/t/redirecting-snapcraft-output/16948

ppd, 2020-04-27:
Related Travis issue: https://travis-ci.community/t/output-is-truncated-heavily-in-arm64-when-a-command-hangs/7630

ppd, 2020-04-27:
@whitequark What about adding `allow_failures` to Travis cfg to prevent this hickup from polluting the CI results?

`edge` is by definition unstable or uncurated enough that we could tolerate them being a little out of date (sometimes) till the arm64 problem is resolved on the Travis side.

whitequark, 2020-04-27:
Sounds fine.

ppd, 2020-05-07:
Looks better now. On the other hand, OSX is always on the verge of timing out.

whitequark, 2020-05-21:
See #618.

ppd, 2020-10-17:
Memo: new workaround to try @ https://travis-ci.community/t/travis-build-gets-stalled-for-arm64-job-in-pandas-package/7678/8

ppd, 2020-10-18:
This is likely to be resolved in the not-too-distant future with a move to `arm64-graviton2` on travis-ci.com: https://github.com/solvespace/solvespace/issues/744#issuecomment-711051289

ppd, 2020-10-26:
With our move to `arm64-graviton2`, the original issue is resolved.

closed 2020-10-26 by ppd