solvespace/solvespace #588: Entities unselected on switching nearest view (isometric or orthogonal)

Symbian9, 2020-04-27:
### System information

**SolveSpace version:** [`f36ac500`](https://github.com/solvespace/solvespace/commit/f36ac500a14da169c2017947710fc6e13b712c4e)

**Operating system:** Debian 10 buster (x86_64)

### Expected behavior

Every time on change nearest view (isometric or orthogonal) all selected entities should be kept selected.

### Actual behavior

Every time on change nearest view (isometric or orthogonal) all selected entities automatically unselected.

FTR, With changing projection (enable/disable perspective projection) all selected items *kept selected*.

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

whitequark, 2020-04-27:
@jwesthues Was this behavior intentional, or is this just an omission?

jwesthues, 2020-04-28:
Intentional, but I'm good with whatever behavior you prefer.

ruevs, 2020-04-28:
I also like it as it is. But not too strongly.

whitequark, 2020-04-30:
It looks like there isn't consensus in favor of the change, and my baseline preference for arbitrary UI changes is to keep things as they were, since no one enjoys churn.

I'm going to close this, but if it turns out that a lot of people find the current behavior frustrating (if you're one of them, please leave a comment!), we can always revisit it in the future.

closed 2020-04-30 by whitequark