solvespace/solvespace #597: PNG images

ruevs, 2020-05-08:
### System information

SolveSpace version: (e.g. 3.0~3dd2fc00; go to Help → About...)

Operating system: (e.g. Debian testing)

### Expected behavior

What should have happened?

### Actual behavior

What actually happened?

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

whitequark, 2020-05-08:
Definitely not as-is, because the `/test` images are generated and shouldn't be treated as assets. The `/res` images can be, but the gains are minimal; I think I already ran optipng on them some time ago.

ruevs, 2020-05-08:
Oh I forgot about the `/test` sorry. So `/res` is reduced from 22.03kb to 13.7kb not bad as a percentage, but not worth it.

closed 2020-05-08 by ruevs