solvespace/solvespace #599: Few DOF shown in group where all entities are constrained

Symbian9, 2020-05-08:
### System information

- **SolveSpace version:** `3.0~f36ac500`
- **Operating system:** Debian 10 buster (x86_64)

### Expected behavior

If all constrained there should be `0` DOF.

### Actual behavior

Actually there are few DOF shown in group where all entities are fully constrained.

### Additional information

Take a look on `g008-*` group inside linked SLVS-file:

- https://gist.github.com/Symbian9/d9a84389e7dd04b8affaafa3f83a220c

[image]

whitequark, 2020-05-09:
> Take a look on `g008-*` group inside linked SLVS-file:

You shouldn't post SLVS files as text, since they aren't text files and get corrupted if you do that. Use a zip archive to attach them.

> Actually there are few DOF shown in group where all entities are fully constrained.

The behavior you're seeing is caused by constraining a point in the workplane to a line in 3d perpendicular to the workplane:

[image]

In other words, the degree of freedom is present analytically (the equations have free variables corresponding to this DOF), but not numerically (the sketch happens to have a shape that prevents the DOF from letting the point move).

If you were to alter the sketch so that the highlighted line is not perpendicular anymore, then the DOF would come into play. So this isn't a bug: it's showing you that your sketch can potentially exhibit more freedom in different circumstances. To avoid this, constrain the point not to a line but for example to a point in the same corner.

closed 2020-05-09 by whitequark

Symbian9, 2020-05-09:
> In other words, the degree of freedom is present analytically (the equations have free variables corresponding to this DOF), but not numerically (the sketch happens to have a shape that prevents the DOF from letting the point move).

As me it looks exactly like a bug, because there is only one possible position for point in plane constrained on line, which are  perpendicular to actual plane.

whitequark, 2020-05-09:
@Symbian9 How many degrees of freedom does a gyroscope in gimbal lock have? There isn't a single correct answer to this question; you can say "two" if you look at it numerically, or "three" if you do it analytically, and SolveSpace currently does the latter.

Inciedntally, nearly every sketch SolveSpace says has 0 degrees of freedom actually *does* have a nonzero amount of them. For example, consider this sketch with "0 DOF":
[image]

And this one, also with "0 DOF":
[image]

But wait, they're the same sketch! In fact you can relax the constraints (for convenience; not actually required but makes it much easier to do this) and change the shape of the 1st sketch to be like the 2nd. That's a degree of freedom, though discrete and not continuous like usual, so it's not counted.

To summarize, the DOF number shown by SolveSpace is an abstraction, and the choice of what exactly counts as a DOF is arbitrary. That's why this isn't a bug.