solvespace/solvespace #602: Opening via Finder displays an error related to a -psn parameter

whitequark, 2020-05-10:
See https://github.com/solvespace/solvespace/issues/398#issuecomment-612759454 and https://stackoverflow.com/questions/10242115/os-x-strange-psn-command-line-parameter-when-launched-from-finder.

closed 2020-05-10 by whitequark

stiff, 2020-05-10:
Now that you've reminded me about this, I realised that it's trivial to test it: just run from terminal, not from finder. And I can confirm that running from terminal without command line arguments app opens without any alerts.

So probably handling `psn` argument like here: https://github.com/freeorion/freeorion/blob/0165590532701ac6e38e2872105566eeac2f84d8/util/OptionsDB.cpp#L718 is enough.

whitequark, 2020-05-10:
Yep, already done.

commit 9951da89 (2020-05-10): macOS: ignore spurious `-psn_` command line argument.

Fixes #602.