solvespace/solvespace #611: Helix crashes

ruevs, 2020-05-12:
### System information

SolveSpace version: master e84fd4649b05fe2
Operating system: Windows 10 
Microsoft Visual C++ 2019 x64

### Expected behavior

Create a helix

### Actual behavior

Crashes here https://github.com/solvespace/solvespace/blob/master/src/group.cpp#L1033
With
```
Exception thrown at 0x756707F2 in solvespace.exe: Microsoft C++ exception: std::length_error at memory location 0x00EFC7E8.
Unhandled exception at 0x756707F2 in solvespace.exe: Microsoft C++ exception: std::length_error at memory location 0x00EFC7E8.
```
Call stack:
```
 	KernelBase.dll!756707f2()	Unknown
 	KernelBase.dll![Frames below may be incorrect and/or missing, no symbols loaded for KernelBase.dll]	Unknown
 	[External Code]	
>	solvespace.exe!SolveSpace::Group::CopyEntity(SolveSpace::IdList<SolveSpace::Entity,SolveSpace::hEntity> * el, SolveSpace::Entity * ep, int timesApplied, int remap, SolveSpace::hParam dx, SolveSpace::hParam dy, SolveSpace::hParam dz, SolveSpace::hParam qw, SolveSpace::hParam qvx, SolveSpace::hParam qvy, SolveSpace::hParam qvz, SolveSpace::hParam dist, SolveSpace::Group::CopyAs as) Line 1033	C++
 	solvespace.exe!SolveSpace::Group::Generate(SolveSpace::IdList<SolveSpace::Entity,SolveSpace::hEntity> * entity, SolveSpace::IdList<SolveSpace::Param,SolveSpace::hParam> * param) Line 649	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::GenerateAll(SolveSpace::SolveSpaceUI::Generate type, bool andFindFree, bool genForBBox) Line 243	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::GenerateAll(SolveSpace::SolveSpaceUI::Generate type, bool andFindFree, bool genForBBox) Line 197	C++
 	solvespace.exe!SolveSpace::Group::MenuGroup(SolveSpace::Command id, SolveSpace::Platform::Path linkFile) Line 343	C++
 	solvespace.exe!SolveSpace::Group::MenuGroup(SolveSpace::Command id) Line 68	C++
 	[External Code]	
 	solvespace.exe!SolveSpace::Platform::WindowImplWin32::WndProc(HWND__ * h, unsigned int msg, unsigned int wParam, long lParam) Line 1041	C++
 	[External Code]	
 	solvespace.exe!SolveSpace::Platform::RunGui() Line 1679	C++
 	solvespace.exe!main(int argc, char * * argv) Line 29	C++
 	solvespace.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 40	C++
```

### Additional information

Watch the movie ;-)
[HelixCrash.zip](https://github.com/solvespace/solvespace/files/4613823/HelixCrash.zip)

phkahler, 2020-05-12:
Does not happen on my Fedora setup. Did it really die on copying the font of an entity? That sounds like unitialized memory or something. Couldn this be due to the recent change of memory allocator?

whitequark, 2020-05-12:
@phkahler Have you tried to reproduce this issue with a normal build, or with a build with sanitizers enabled?

ruevs, 2020-05-12:
If I go back to 74103eee819d2 (just before 521473ee4bde8e) then it crashes here:
https://github.com/solvespace/solvespace/blob/fabffbab7929e1db59d87ebd34ec5001941e25ff/src/platform/utilwin.cpp#L60
```
Critical error detected c0000374
```
```
 	ntdll.dll!776093dd()	Unknown
 	ntdll.dll![Frames below may be incorrect and/or missing, no symbols loaded for ntdll.dll]	Unknown
 	[External Code]	
>	solvespace.exe!SolveSpace::MemAlloc(unsigned int n) Line 60	C++
 	solvespace.exe!SolveSpace::List<SolveSpace::SEdge>::ReserveMore(int howMuch) Line 218	C++
 	solvespace.exe!SolveSpace::List<SolveSpace::SEdge>::AllocForOneMore() Line 232	C++
 	solvespace.exe!SolveSpace::List<SolveSpace::SEdge>::Add(const SolveSpace::SEdge * t) Line 236	C++
 	solvespace.exe!SolveSpace::SEdgeList::AddEdge(SolveSpace::Vector a, SolveSpace::Vector b, int auxA, int auxB, int tag) Line 224	C++
 	solvespace.exe!SolveSpace::SContour::MakeEdgesInto(SolveSpace::SEdgeList * el) Line 564	C++
 	solvespace.exe!SolveSpace::SPolygon::MakeEdgesInto(SolveSpace::SEdgeList * el) Line 674	C++
 	solvespace.exe!SolveSpace::SPolygon::UvGridTriangulateInto(SolveSpace::SMesh * mesh, SolveSpace::SSurface * srf) Line 425	C++
 	solvespace.exe!SolveSpace::SSurface::TriangulateInto(SolveSpace::SShell * shell, SolveSpace::SMesh * sm) Line 436	C++
 	solvespace.exe!SolveSpace::SShell::TriangulateInto$omp$1() Line 1058	C++
```
Which I think is this:
https://github.com/solvespace/solvespace/pull/600#issuecomment-626263617

If I checkout up to and including 521473ee4bde8e then the crash changes to the one above. So @phkaler - yes, I think 521473ee4bde8e removes that crash and introduces this one.

phkahler, 2020-05-12:
> @phkahler Have you tried to reproduce this issue on a normal build, or on a build with sanitizers enabled?

I only run release builds, sorry. It's kind of siily I haven't adopted a decent IDE yet or used the debugger.  gedit and the terminal over here...

@ruevs yeah using OpenMP was a risk, but that first crash you had was in such a strange place, it doesn't seem related on the surface.

That second crash (that you posted) looks like it's in the merging of the triangle lists. That was not done by separate threads prior to  https://github.com/solvespace/solvespace/commit/e84fd4649b05fe250ba3fcb7bc4ccd73a764ba68  Does it crash if you go one prior to that?

whitequark, 2020-05-12:
> I only run release builds, sorry. It's kind of siily I haven't adopted a decent IDE yet or used the debugger. gedit and the terminal over here...

You don't need an IDE or a debugger to run a build with sanitizers. Try configuring with `cmake .. -DENABLE_SANITIZERS=ON` (probably in another build directory than your normal one, like `build-debug` or something), then reproducing the bug. If there is a memory issue you'll see it in the terminal.

phkahler, 2020-05-12:
OK, probably not important but when running cmake:

[paul@localhost build]$ cmake .. -DCMAKE_BUILD_TYPE=Release -DENABLE_SANITIZERS=ON
-- Using sanitizer options address;alignment;bounds;shift;signed-integer-overflow;integer-divide-by-zero;null;bool;enum;return
-- Using in-tree libdxfrw
-- Using in-tree flatbuffers
-- Using in-tree q3d
CMake Warning (dev) at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (SPACEWARE)
  does not match the name of the calling package (SpaceWare).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/FindSpaceWare.cmake:19 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:266 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done

Then immediately on running solvespace:

SolveSpace!

=================================================================
==180558==ERROR: AddressSanitizer: new-delete-type-mismatch on 0x6060000df640 in thread T0:
  object passed to delete has wrong type:
  size of the allocated type:   56 bytes;
  size of the deallocated type: 48 bytes.
    #0 0x7f85834e7175 in operator delete(void*, unsigned long) (/lib64/libasan.so.5+0x111175)
    #1 0x5775e2 in SolveSpace::Platform::TimerImplGtk::RunAfter(unsigned int) (/home/paul/projects/solvespace/build/bin/solvespace+0x5775e2)
    #2 0x4d85ee in SolveSpace::SolveSpaceUI::Init() (/home/paul/projects/solvespace/build/bin/solvespace+0x4d85ee)
    #3 0x49adb4 in main (/home/paul/projects/solvespace/build/bin/solvespace+0x49adb4)
    #4 0x7f85812521a2 in __libc_start_main (/lib64/libc.so.6+0x271a2)
    #5 0x4a68ad in _start (/home/paul/projects/solvespace/build/bin/solvespace+0x4a68ad)

0x6060000df640 is located 0 bytes inside of 56-byte region [0x6060000df640,0x6060000df678)
allocated by thread T0 here:
    #0 0x7f85834e5a97 in operator new(unsigned long) (/lib64/libasan.so.5+0x10fa97)
    #1 0x5773db in SolveSpace::Platform::TimerImplGtk::RunAfter(unsigned int) (/home/paul/projects/solvespace/build/bin/solvespace+0x5773db)

SUMMARY: AddressSanitizer: new-delete-type-mismatch (/lib64/libasan.so.5+0x111175) in operator delete(void*, unsigned long)
==180558==HINT: if you don't care about these errors you may set ASAN_OPTIONS=new_delete_type_mismatch=0
==180558==ABORTING

whitequark, 2020-05-12:
The first issue (CMake) is unimportant but something we should fix in the long run just to reduce noise.

The second issue (new-delete-type-mismatch) is a major memory issue that needs to be fixed.

whitequark, 2020-05-12:
@phkahler I forgot to mention that to get useful backtraces with sanitizer builds you should add `-DCMAKE_BUILD_TYPE=Debug` to the cmake command line.

whitequark, 2020-05-12:
> The second issue (new-delete-type-mismatch) is a major memory issue that needs to be fixed.

It looks like a bug in libsigc++ (which is a transitive dependency through GTK 3): https://github.com/libsigcplusplus/libsigcplusplus/issues/10. We can't do anything about it unfortunately so you should `export ASAN_OPTIONS=new_delete_type_mismatch=0` like it suggests you to work around the abort.

phkahler, 2020-05-12:
@whitequark Thanks I'll try that next. If I ignore the error it appears to run but then crashes if I try to Revolve or Helix (but not extrude) it's claiming a use-after-free in CopyEntity called from Group::Generate.

I'm impressed that these tools are finding issues that have gone undetected so long.

whitequark, 2020-05-12:
> I'm impressed that these tools are finding issues that have gone undetected so long.

Sanitizers are extremely powerful and valuable tools and it is almost impossible to write reliable C++ code without using them; completely impossible if a team of people with a varying degree of understaning C++ is working on a shared codebase. (Before sanitizers there was Valgrind, which is still occasionally useful, but it can fundamentally discover fewer issues in C/C++ codebases, and is much slower.)

The main downside is a small amount of false positives they have (e.g. TSan doesn't appear to be usable with the GNU OpenMP implementation, sadly), but almost everything they report tends to be a real issue after all.

whitequark, 2020-05-12:
Oh, and we do already run sanitizers on CI, but the scarce test coverage means many issues go undetected anyay.

phkahler, 2020-05-12:
Lathe is failing too.

phkahler, 2020-05-12:
Oh, lots of memory leaks too without doing anything, just run it and close the window.

whitequark, 2020-05-12:
> Oh, lots of memory leaks too without doing anything, just run it and close the window.

There are quite a few allocations in GTK and OpenGL that aren't freed, so leak checking is more usefully performed with Windows builds, unfortunately.

phkahler, 2020-05-12:
The leaks look mostly GTK related too. The crashes are not...

whitequark, 2020-05-12:
Yep, the crashes are exceedingly likely to point to a real issue.

phkahler, 2020-05-12:
Oh.... I think I did something dumb. I need to review some history. And by dumb I don't mean the typical "wrote a bug dumb", but the "ignored the comments that warn you about a hazard" kinda dumb.

phkahler, 2020-05-12:
@ruevs can you try this:
https://github.com/phkahler/solvespace/commit/f06fb4fc04389f687d688b955ca5d03ab5091abc

I probably introduced this when I removed extra entity copies from those 3 tools. There is a comment right there not to use *e after copy because it may not be valid. I changed to code for clarity but ignored that comment.

phkahler, 2020-05-12:
Yes, the bugs I just fixed have been there since: https://github.com/solvespace/solvespace/commit/45eb2468655dbf8f26b2a113810de0c62d3d0833

There seem to be a lot of people building from source but apparently not on Windows?

ruevs, 2020-05-12:
> 
> 
> @ruevs yeah using OpenMP was a risk, but that first crash you had was in such a strange place, it doesn't seem related on the surface.
> 
> That second crash (that you posted) looks like it's in the merging of the triangle lists. That was not done by separate threads prior to [e84fd46](https://github.com/solvespace/solvespace/commit/e84fd4649b05fe250ba3fcb7bc4ccd73a764ba68) Does it crash if you go one prior to that?

1. If I revert _only_ `void SShell::TriangulateInto(SMesh *sm)` to this: https://github.com/solvespace/solvespace/blob/7baf58588bebb2909a3179b42455c7615491bd09/src/srf/surface.cpp#L1052 version then it crashes in the first way. So the **_second crash_** is certainly due to (a race condition in?) the parallel code. Sorry I did not state this more clearly. I thought pointing to  _**this  https://github.com/solvespace/solvespace/pull/600#issuecomment-626263617**_  made it obvious.

2. As for Whitequarks' heap change https://github.com/solvespace/solvespace/commit/521473ee4bde8e1697211de30a1aa125cdc35ff8 it does not `git revert` cleanly so I need to spend more time to figure out if reverting it stops causing the **_first crash_**, Tomorrow.

phkahler, 2020-05-12:
@ruevs did you try the patch I linked. It's not in PR #612, waiting for you to verify it actually fixes your issue and not just the problems I found.

ruevs, 2020-05-12:
Just did. Your phkahler/issue611 fixes the "first crash".
I do build on Windows all the time (VS, MinGW, clang) but I hah not created new helix-es in a while :-) Opening existing one (obviously) works just fine.

closed 2020-05-12 by phkahler

ruevs, 2020-05-12:
Now I'll try with your patch, the heap change AND parallel SShell::TriangulateInto. Essentially the new master at 23dfd97285f46   ...

phkahler, 2020-05-12:
Oops. I linked this issue to the PR so it closed on merge. Please let us know if you still have the other problem.

whitequark, 2020-05-12:
Parallel triangulation won't ever work on Windows without removing HEAP_NO_SERIALIZE.

phkahler, 2020-05-12:
Is that done on master?

whitequark, 2020-05-12:
Of course.

ruevs, 2020-05-12:
It does not crash at all with current master https://github.com/solvespace/solvespace/commit/23dfd97285f466d3102995becbbb1bf1f75c8b31

So @whitequark your removal of the Win32 heap seems to fix (hide?) the ["OpenMP SShell::TriangulateInt" related crash](https://github.com/solvespace/solvespace/issues/611#issuecomment-627397255).

And @phkahler your https://github.com/solvespace/solvespace/pull/612 fixes the [original crash in this issue.](https://github.com/solvespace/solvespace/issues/611#issue-616403036)

And by the way I like your ["Split quads based on angle"](https://github.com/solvespace/solvespace/commit/c355a4730adf40aae3552bc5ec57f7615bb6d5bb) - now that I can try it I will play some more :-)

phkahler, 2020-05-12:
Awesome!

whitequark, 2020-05-12:
> So @whitequark your removal of the Win32 heap seems to fix (hide?) the ["OpenMP SShell::TriangulateInt" related crash](https://github.com/solvespace/solvespace/issues/611#issuecomment-627397255).

It doesn't "hide" anything. It fixes the root cause of the crash, which is the lack of serialization on heap access. In this context, serialization means using a mutex to synchronize heap access, which would cause the threads to access it one at a time, or *serially*.