solvespace/solvespace #615: Suggestion: Add radio buttons to rotate groups for combine method

phkahler, 2020-05-17:
I don't know it this is a good idea or not. As it exists when rotating a group, it picks up the last combine operation of the parent group - a regular extrusion will replicate N times and form a union of those. A hole extruded as difference will produce N holes when rotated. The idea here is to allow the full selection of Assemble, Union, Difference, and Intersection for rotate groups.

The attached model uses 2 extrudes - only to allow the second one to be an intersection. I then added 2 lines in Group::GenerateForStepAndRepeat() to allow the copies to be intersected:

        // And tack this transformed copy on to the return.
        if(soFar->IsEmpty()) {
            scratch->MakeFromCopyOf(&transd);
        } else if(forWhat == CombineAs::ASSEMBLE) {
            scratch->MakeFromAssemblyOf(soFar, &transd);
//        } else if(forWhat == CombineAs::INTERSECTION) {
//            scratch->MakeFromIntersectionOf(soFar, &transd);
        } else {
            scratch->MakeFromUnionOf(soFar, &transd);
        }

The experimental lines are commented out here. This is not what I propose, but rather adding the regular radio buttons on the text window. It'd be fine to inherit the operation from the parent group but being able to change it might be nice.

Make the changes above and load the following model to see how powerful this can be.
[dodecahedron.zip](https://github.com/solvespace/solvespace/files/4639507/dodecahedron.zip)

Also to @ruevs if you load that with unmodified master the booleans fail. It seem intersection works more reliably than union or difference. I have no idea why, but maybe it's a hint.

This concludes todays episode of making SolveSpace do crazy stuff. Thoughts?

whitequark, 2020-05-17:
IIRC we looked into it with @Evil-Spirit but there was some issue. I don't remember what it was anymore though.

whitequark, 2020-05-17:
@phkahler (Aside: I've edited the source block in your comment to be more readable, since it was pretty confusing the way it go laid out. Take a look at the [markdown introduction](https://guides.github.com/features/mastering-markdown/); the most useful part is just the <code>```</code> delimiters for code blocks.)

Evil-Spirit, 2020-05-17:
Can you attach an image of result and describe why this change help to make it?

Evil-Spirit, 2020-05-17:
Also. I propose a new combine method for groups - clip & aseemble. The current group should be clipped (b - a) and then assembled. This is perfect for wooden constructions like roofs.

phkahler, 2020-05-17:
@Evil-Spirit  But if you do B-A there is nothing left to assemble? Just part of B.

Evil-Spirit, 2020-05-17:
A stays and assembled with B-A

phkahler, 2020-05-17:
How is that different that A union B?

Evil-Spirit, 2020-05-17:
You have an assemble of two different parts instead of of one solid model. You can suppress solid model of prev group and see a surface of contact. The reason is the same why we choose assemble mode for extrusion groups

Evil-Spirit, 2020-05-17:
[image]

phkahler, 2020-05-17:
> Can you attach an image of result and describe why this improvement helps to make it?

I've simplified it a bit.

1) Start with a large rectangle - the only purpose of this is to create a 2-sided extrusion so the next one can be made as intersection.

2) Sketch on the end of the extrusion from 1:
[image]

3) extrude that as intersection with the previous extrusion to form this:
[image]

4) Rotate that object 5 times around the origin. With the code patch above it will be done as intersection, resulting in:

[image]

The previously attached file does it in 2 rotations. The first creates a truncated 5-sided pyramid and the second flips that over 180 degrees and intersects them.

This is not a typical construction method, and I'd rather implement more conventional tools that are more generally useful. However, adding the radio buttons would actually be consistent within the SolveSpace UI. A sort of generalization of what's already there. This object could actually be made in 3 groups - sketch - extrude - rotate (as intersection).

OTOH, when you extrude as difference to cut a hole, I think it's nice that a subsequent rotate group is done as difference to make more holes. Actually I'm not sure if that's what happens, does it union with the previous holes prior to taking the difference? Whatever it does, I would want this to somehow be consistent with it. If it can't be consistent with what exists, then it should not be done. OTOH again, doing these operations with existing code does nothing useful AFAICT.

ruevs, 2020-05-17:
Oh I love this. I would never have thought of making a dodecahedron this way, in fact I would have to think pretty hard to even get a vague clue how to draw one in SoveSpace at all... But this is pretty amazing.
I do not know how we are going to explain it to the users. Maybe an example like this in the manual.
[image]

Evil-Spirit, 2020-05-18:
I see. But this is not the same option - actually, we have two different combine options. First is operation for combining with the previous group, the second if for operation to combine inside the current group (step and repeat - like).

phkahler, 2020-05-18:
Yes @Evil-Spirit  it looks like a hole (extrude as difference or even lathe as difference) is first rotated and Union combined with the original hole and then the entire set is subtracted from whatever existed 2 groups prior. This seems like it would be confusing to a user, even with radio buttons who might say "it says union but it's making more holes, which is what I wanted but...."

@ruevs  This is really a hack. One thing I want to implement (and tried before) is frustum extrusions. That will allow you to make a pentagon into a 5-sided truncated pyramid, and the intersection of 2 of those can form a dodecahedron. A tool like that seems more generally useful than this. It will also create all the line entities on the edges of the frustum, which booleans don't do. Either way, intersection is very important.

Another way to make a dodecahedron (empty inside) is to extrude a pentagon to some thickness and then make an assembly from 12 of them. My recent experiment in "copy groups" allowed the same thing without the need to save and link the pentagonal extrusion, just make 11 copies of it.

whitequark, 2020-05-18:
I agree that this seems to be confusing enough to be more trouble than it's worth. Remember, we can't really remove features we add.

whitequark, 2020-06-23:
Since @phkahler, who requested this feature, thinks it's more trouble than it's worth, I'm going to close this. We can of course revisit it later if more use cases arise.

closed 2020-06-23 by whitequark