solvespace/solvespace #618: RFC: Migrate CI to GitHub Actions?

whitequark, 2020-05-21:
I've set up GitHub Actions for a different project and was pleasantly surprised by how well they work. This has some significant potential benefits for us, including:
  * Unifying the build configuration for Windows, Linux and macOS (strictly speaking we could do this on Travis or AppVeyor just as well);
  * Uploading artifacts for each build on all platforms, not just on Windows, simplifying the life of macOS and to lesser degree Linux users;
  * Significantly faster builds, since all of our submodules already live on GitHub and checkouts are a significant contributor to AppVeyor build times;
  * Better build sequencing, such as not building Snap packages if tests fail on Windows (which may indicate a Windows-specific problem, but quite often points to a generic problem that goes undetected on Linux build bots);

Any opinions or objections?

cc @ppd since this will impact Snap builds

Symbian9, 2020-05-21:
FTR, GitHub's actions artifacts (as I know) could not be downloaded by not logged users (e.g. there  are no anonymous downloading for artifacts). I may be wrong.

whitequark, 2020-05-21:
You are correct (I think they do it to avoid excessive bandwidth costs). This isn't a huge problem though since we can publish a rolling "nightly" pre-release, like [wasmtime does](https://github.com/bytecodealliance/wasmtime/blob/master/.github/workflows/main.yml#L500-L509).

ppd, 2020-05-21:
There's an upstream action for snapcraft: https://github.com/snapcore/action-build and it seems like arm64 and amd64 are both supported.

So this seems like a good idea, snap-wise.

phkahler, 2020-06-09:
@whitequark  you had mentioned that the current build are debug anyway, so not suitable for general use. Would this change include producing release builds?

Either way, if it makes life easier I say do it.

whitequark, 2020-06-09:
> Would this change include producing release builds?

We could far more easily set it up so that e.g. after debug builds succeed and go through testing, there is another round of release builds that are uploaded for general use.

ppd, 2020-07-02:
I must correct my statement regarding arm64 support: the only way to build on arm64 seems to self-host runners. Github-hosted runners are only amd64 as far as I can see (https://github.community/t/does-github-actions-support-multiple-os-architectures/16526).

phkahler, 2020-10-16:
Closing because it looks like we're going with Travis for now.

closed 2020-10-16 by phkahler

ppd, 2020-10-17:
We can come back to this if necessary. Travis has its own problems, but it's a) there and b) has brittle arm64 support as a USP.