solvespace/solvespace #619: Some parameters affected in "Property Browser" by changing "digits after decimal point to show"

Symbian9, 2020-05-21:
### System information

- **SolveSpace version:** `3.0~70ec7cc2`
- **Operating system:** Debian 10 buster

### Expected behavior

Changing *"digits after decimal point to show"* should not affect properties fields on *"Property Browser"*.

### Actual behavior

Some properties fields in "Property Browser" affected by changing *"digits after decimal point to show"*.

For example, if set *"digits after decimal point to show > distances: `0`"* some properties values lover than `1.0` (e.g. `0.5`) would be shown as `0`, but as those properties are important settings such thing should not happen:

[image]

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

whitequark, 2020-05-21:
Hm, that's an issue, but it's not clear to me what precision we should use here.

Symbian9, 2020-05-21:
> but it's not clear to me what precision we should use here.

I suggest `<= 8` digits as next:

- `x.1234567890` -> `x.12345678` — cut to 8 digits (maximum) without decimation;
- `x.0010000000` -> `x.001` — limit 8 digits + cut to last non-zero digit;
- `x.0000000000` -> `x` — cut to integer if first 8 digits are zero.

whitequark, 2020-05-21:
IIRC, changing the amount of decimals depending on the value itself is troublesome to do with the way the text window is currently implemented.

phkahler, 2023-08-10:
This was fixed in #884 at least well enough to not show 0 when a value is not 0.

closed 2023-08-10 by phkahler