solvespace/solvespace #622: Invalid file path when dragging .slvs file on .exe (Windows drag and drop argument handling)

gnbl, 2020-06-01:
`This is SolveSpace version 3.0~225f82a2` on `Windows 10` (from https://ci.appveyor.com/api/buildjobs/g2waan8v16gv22jc/artifacts/build%2Fbin%2FDebug%2Fsolvespace.exe)

1. Start program / open file by drag-dropping the file on the executable.
2. "File > Save As..." fails without error (file chooser dialog does not open)
3. "File > Open Recent" has entries like `\\?\<path>`

gnbl, 2020-06-03:
Same behavior when running `solvespace.exe <path>` in cmd.exe.

gnbl, 2020-06-03:
Argument handling appears to start in InitCli() at https://github.com/solvespace/solvespace/blob/master/src/platform/platform.cpp#L598

ruevs, 2020-10-16:
The problem is caused by this code:
https://github.com/solvespace/solvespace/blob/d72eba8039c52f5e8d3ab36ec2d8a7598dfd578b/src/platform/platform.cpp#L309

When a file is passed in as a command line argument the parsing on Windows adds the [raw handling prefix `\\?\\`](https://docs.microsoft.com/en-us/windows/win32/fileio/naming-a-file?redirectedfrom=MSDN#win32-file-namespaces) to the path. In other words the path becomes:
`\\?\C:\Users\...`
instead of:
`C:\Users\...`
However the file Open/Save/[Save As dialog boxes](https://docs.microsoft.com/en-us/windows/win32/api/commdlg/nf-commdlg-getsavefilenamea) do not seem to like this prefix (the path is stored in the [lpstrFile member of the `OPENFILENAMEA` structure](https://docs.microsoft.com/en-us/windows/win32/api/commdlg/ns-commdlg-openfilenamea)

When a file is opened with the "File | Open..." or "Open Recent" dialog boxes on the other hand the [`ofn` structure](https://github.com/solvespace/solvespace/blob/d72eba8039c52f5e8d3ab36ec2d8a7598dfd578b/src/platform/guiwin.cpp#L1536) contains a normal path - and that is why it works.

The easiest way would be to comment out https://github.com/solvespace/solvespace/blob/d72eba8039c52f5e8d3ab36ec2d8a7598dfd578b/src/platform/platform.cpp#L309 but we will lose the ability to open files from long paths (>MAX_PATH). Otherwise I will have to think about it some more.

closed 2020-10-16 by phkahler