solvespace/solvespace #625: IDF import support

azonenberg, 2020-06-03:
This would be very useful when using solvespace to design enclosures for PCBs.

https://en.wikipedia.org/wiki/Intermediate_Data_Format

phkahler, 2020-09-24:
For bare PCBs this file format should be fairly easy to read and could be "converted" to an actual SovleSpace file. The board outline and cut-outs are defined by line and arc entities that could be directly represented by SolveSpace sketch entities. The board thickness is given, so an extrusion could be created from the outline. Drilled holes are defined separately but could also be included as part of the 2D sketch. Keepout regions are defined the same way and could be imported as some other line style.

Components are another story, but they're not much more complicated than the PCB itself.

Google search for "IDF file format specification" turned up this:
http://www2.me.rochester.edu/courses/ME204/nx_help/en_US/graphics/fileLibrary/nx/tdoc_pcb/idf_v3.pdf

You know, for anyone that wants to write an importer and learn more about solvespace internals! ;-)

phkahler, 2020-09-25:
Wondering if @azonenberg or anyone else can append a simple IDF file. I'm working on an IDF importer that is initially only going to support the board outlines and drilled holes. I have a file that's just a rectangular board, but one with some round features and a cut-out also with both straight and curved edges would be nice. Also some drilled holes but not hundreds of them.

phkahler, 2020-09-27:
Having written a crude importer for IDF boards, I think the better approach will be Linking. The distinctions is this: Import creates a SolveSpace sketch (and maybe extrusions etc...) from an IDF file, but they will not be fully constrained and not exportable as IDF, so the best use would be to save it as .slvs and then link it into an assembly. Instead we should allow directly linking an IDF file.

phkahler, 2020-09-29:
Looking for feedback on a preliminary implementation in PR #725.  Also looking for a simple PCB design with routing keepouts and some components (in IDF format with both files).

phkahler, 2020-09-30:
@azonenberg  I merged IDF linking support to SolveSpace master. Part placement is still not there so I'm leaving this issue open.

azonenberg, 2020-09-30:
Will test soonish.  Been busy w/ other stuff and haven't had time to test.

For the most part I'm primarily interested in PCB outlines, not components.

phkahler, 2020-09-30:
> For the most part I'm primarily interested in PCB outlines, not components.

You may find it useful then! thanks for the response.

phkahler, 2021-05-16:
The circuit boards (.emn) can be linked in version 3.0 final but will not show board keepout regions. There is a new Pull Request that shows keepouts as construction lines which should be merged soon.

Parts files (.emp) are not supported yet. I have not been able to get a design from anyone with the part placements. Someone sent me a pair of files for a KiCAD project, but the part file was empty and the board file had no part placement info (makes sense if the part definitions aren't there).

Symbian9, 2021-05-17:
> Parts files (.emp) are not supported yet. I have not been able to get a design from anyone with the part placements. Someone sent me a pair of files for a KiCAD project, but the part file was empty and the board file had no part placement info (makes sense if the part definitions aren't there).

@ohkahler, Here are some IDF PCB examples on GitHub:

- `*.emn` - https://github.com/search/?q=idf+extension:emn&type=Code
- `*.emp` - https://github.com/search/?q=idf+extension:emp&type=Code

phkahler, 2023-08-10:
Closing since the PCBs can be linked.  Not many people are using it, and no one has requested that we add the ability to load the parts file too. Let's call this done until further notice.

closed 2023-08-10 by phkahler