solvespace/solvespace #628: Deleting a workplane with an arc locked to it crashes SolveSpace

johreh, 2020-06-06:
### System information

SolveSpace version: 3.0~225f82a2 

Operating system: Arch 

### Actual behavior
After constructing a new workplane inside an existing group and constructing an arc on that workplane, SolveSpace crashes when deleting the workplane.

### Additional information
Example file with conditions already set up. Deleting r0004-workplane crashes SolveSpace.
[workplane_arc.zip](https://github.com/solvespace/solvespace/files/4740623/workplane_arc.zip)

phkahler, 2020-06-10:
I can reproduce the crash, but I can not figure out how you created that file. If I use Sketch->Workplane to create a new workplane it says I need to start a new sketch-in-3d first. If I create a sketch-in-3d first, I get a new group in the text window and can then add a new workplane. Your file does not have a sketch-in-3d group. What were the steps used to create that file?

ruevs, 2020-06-10:
It is easy:
- `Ctrl+N`
- `3` ("Sketch | Anywhere in 3d")
- Rotate view with middle button drag
- "Sketch | Workpalne"
- Click on origin to create the workplane
- Select the created "workplane"
- `2` ("Sketch | In Workplane")
- `A` ("Sketch | Ark of a Circle")
- Make an ark

Done. Now delete the workplane:

- `ESC`
- Click `g002-sketch-in-plane`
- Click `r004-wokplane1`
- `Del`

Crash.

It's a stack overflow in (the recursive) `SolveSpaceUI::GenerateAll` but I have not had time to debug it. Probably it is caused by the ark remaining an "orphan" when the workplane it is located in is deleted, but how exactly?...

Call stack:
```
>	solvespace.exe!malloc(unsigned int size) Line 23	C++
 	[External Code]	
 	solvespace.exe!SolveSpace::IdList<SolveSpace::Entity,SolveSpace::hEntity>::Add(SolveSpace::Entity * t) Line 424	C++
 	solvespace.exe!SolveSpace::Group::Generate(SolveSpace::IdList<SolveSpace::Entity,SolveSpace::hEntity> * entity, SolveSpace::IdList<SolveSpace::Param,SolveSpace::hParam> * param) Line 471	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::GenerateAll(SolveSpace::SolveSpaceUI::Generate type, bool andFindFree, bool genForBBox) Line 243	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::GenerateAll(SolveSpace::SolveSpaceUI::Generate type, bool andFindFree, bool genForBBox) Line 383	C++
[skip 2174 more recursive calls like this]
 	solvespace.exe!SolveSpace::SolveSpaceUI::GenerateAll(SolveSpace::SolveSpaceUI::Generate type, bool andFindFree, bool genForBBox) Line 383	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::GenerateAll(SolveSpace::SolveSpaceUI::Generate type, bool andFindFree, bool genForBBox) Line 197	C++
 	solvespace.exe!SolveSpace::GraphicsWindow::DeleteTaggedRequests() Line 976	C++
 	solvespace.exe!SolveSpace::GraphicsWindow::DeleteSelection() Line 70	C++
 	solvespace.exe!SolveSpace::GraphicsWindow::MenuClipboard(SolveSpace::Command id) Line 322	C++
 	solvespace.exe!SolveSpace::GraphicsWindow::ActivateCommand(SolveSpace::Command cmd) Line 198	C++
 	solvespace.exe!SolveSpace::GraphicsWindow::KeyboardEvent(SolveSpace::Platform::KeyboardEvent event) Line 258	C++
 	[External Code]	
 	solvespace.exe!SolveSpace::Platform::WindowImplWin32::WndProc(HWND__ * h, unsigned int msg, unsigned int wParam, long lParam) Line 990	C++
 	[External Code]	
 	user32.dll![Frames below may be incorrect and/or missing, no symbols loaded for user32.dll]	Unknown
 	solvespace.exe!SolveSpace::PolylineBuilder::GenerateEdges(SolveSpace::SEdgeList * sel) Line 220	C++
 	[External Code]	
```

ruevs, 2020-06-10:
[NonCoplanarCirclesInOneWorkplaneExtruded.zip](https://github.com/solvespace/solvespace/files/4756604/NonCoplanarCirclesInOneWorkplaneExtruded.zip)

By the way using this "technique" one can create rather "interesting" (nasty) Solvespace files. See for example the attached one. It has two circles in the first group but they are not co-planar. One is in the "XY" ("sketch-in-plane") and the other is in "workplane". When you extrude this group... it does not quite work, but is is interesting :-)

I have never reported this, because I consider it a "feature" (and not a "bug"). For example imagine solving math problems in geometry - which Solvespace is great for even though it may not have been designed with this in mind. Also try adding constraints - e.g. a distance between the two centers - it can be added as a 3D constraint or as a 2D projected one in any of the workplanes . This is GREAT (always has been)!

ruevs, 2020-06-11:
Interesting how that slipped past me, maybe I was somewhere without internet access :-)
I added a comment explaining how to do slanted extrusions in https://github.com/solvespace/solvespace/issues/453 so that anyone searching can find it.

By the way I do not think that this feature is documented - maybe we should add it to the [documentation](https://github.com/solvespace/solvespace-web/blob/master/tutorial.pl) for 3.0.

I discovered it maybe 8-9-10(?) years ago when I had just discovered Sovespace and was amazed at how good it was and played a lot with all possible features. (Yes, I have been following "this" since the time when @jwesthues only had SketchFlat http://cq.cx/sketchflat.pl :-)

ruevs, 2020-06-11:
It is documented "a bit" https://github.com/solvespace/solvespace/issues/453#issuecomment-642574190. So we should update the reference manual https://github.com/solvespace/solvespace-web/blob/master/ref.pl

whitequark, 2020-06-23:
Thanks for the report!

commit 586b0477 (2020-06-23): Delete all entities on a workplane request that is being deleted.

Fixes #628.
closed 2020-06-23 by whitequark