solvespace/solvespace #629: SolveSpace won't import 3D circles/arcs from DXF

johreh, 2020-06-06:
### System information

SolveSpace version: 3.0~225f82a2 

Operating system: Arch 

### Actual behavior
Trying to import a DXF of this exported from Fusion 360
[image]
results in
[image]
The circles and arcs are imported and show up in the request list of the group, but are not drawn and cause the solver to fail.

### Expected behavior
Something like
[image]

### Additional information

DXF file manifesting problem.
[test1 v2.zip](https://github.com/solvespace/solvespace/files/4741051/test1.v2.zip)

johreh, 2020-06-06:
As could be guessed from the screenshot above, I have started on a patch for this 5c472fd568de548762ddc2c15a81c7ca2815fc75

whitequark, 2020-06-06:
Your modified import function creates entities in a way that is not reproducible via the UI. I'm not sure if this is the right approach, and I would be hesitant to merge this patch.

johreh, 2020-06-07:
Can you clarify what is not reproducible via the UI?

phkahler, 2020-06-07:
@johreh I think what whitequark is saying is:  The output .slvs file your code is going to produce can not be created with unmodified SolveSpace using the normal construction tools. In other words, you're creating new internal structures or functions just for this capability instead of reading the file and creating SS equivalent entities.

BTW, I appreciate the approach of opening the issue when you decided to work on it. This provides a place for the dialog that is about to happen to get it right. Not right - optimally useful.

johreh, 2020-06-07:
I still don't quite follow, as I don't see what could not be created using normal SolveSpace tools.

The patch introduces the following changes:
- Applies object coordinates to world coordinates transformation before creating SS objects using the extrusion vector as in the DXF specification
- Creates new workplanes for arcs not in the baseplane
- Applies a normal vector to circles (also to arcs, but I realize that is not actually necessary as they're locked to workplanes)

All of the operations are already achievable from the UI as far as I can tell.

Granted, there's also changes to createDiametric, but that was just to make the code compile for testing and needs to be figured out properly.

phkahler, 2020-06-07:
@johreh  I may be wrong and look forward @whitequark response.  But when the file is opened with the patch, I get a single group g002-sketch-in-plane with all the entities. But some of those are not in the plane. IDK how to construct that from scratch? It's also creating a quaternion in a new way, but I'm not sure how relevant that is. I don't think that plane can be created in SS without some prior entities to define it.

Also: If these things are defined as extrusions and you have the extrusion vector, why not create an actual extrusion in SolveSpace?  While I think that would be great, it would also raise peoples expectations for import which may not be a good thing ;-)

johreh, 2020-06-07:
You can create, move and reorient a workplane anywhere with SS. Just switch to sketch anywhere in 3d, create a new workplane and pick any point in space for its origin.

The added method of creating a quaternion is the [arbitrary axis algorithm](http://help.autodesk.com/view/OARX/2018/ENU/?guid=GUID-E19E5B42-0CC7-4EBA-B29F-5E1D595149EE) from the DXF specification. It is already in the dxfrw library, but unfortunately as a private methods so I had to rewrite it.

The extrusion vector in DXF files, as far as I understand it, is not necessarily the result of an actual extrusion from a 2d sketch to a solid 3d object. I don't think it should be taken to mean anything other than the normal of the plane the entity was created in. The [ezdxf](https://ezdxf.readthedocs.io/en/stable/tutorials/ocs_usage.html#placing-2d-circle-in-3d-space) library documentation does a fair job of illustrating how it works I think.

johreh, 2020-06-07:
Actually, I am certain that you could not use the extrusion vector in DXF to define a 3d-object as it only describes the direction and one point along that axis. For an extrusion you need two points to define both start and end of it.

whitequark, 2020-06-07:
> You can create, move and reorient a workplane anywhere with SS. Just switch to sketch anywhere in 3d, create a new workplane and pick any point in space for its origin.

That creates a new group, but your code doesn't appear to. Am I missing something?

johreh, 2020-06-07:
Sketch menu -> Workplane creates a new workplane in the current group.

whitequark, 2020-06-07:
Oh, I completely forgot about that feature. Then your patch looks great! Thanks for being patient with me here.

phkahler, 2020-06-07:
Learn something every day. But that ends up with a sketch-in-3d no?

phkahler, 2020-06-07:
> Actually, I am certain that you could not use the extrusion vector in DXF to define a 3d-object as it only describes the direction and one point along that axis. For an extrusion you need two points to define both start and end of it.

This I can clarify. When you extrude a 2D sketch in SolveSpace it does these steps:
1 create a new group (extrude)
2 copy all the previous sketch entities to the new group
3 make an additional copy of all the sketch entities with the extrusion offset (x,y,z)
4 construct line entities connective both copies of each sketch point
5 construct face entities (sides) from each sketch line
6 constrain the (x,y,z) from step 3 so the vector is perpendicular to the sketch plane.

If you want to set the extrusion depth that would be an additional distance constraint. It's typically done between 2 points, but it could be done internally for an import.

Just food for thought. I don't know the DXF spec and I see 2 angles in there related to extrusions. I just wanted to make sure you've got the SolveSpace model of extrusion down. Those 6 steps are all implemented if you create a 2D sketch-in-plane and then do an extrude of it.

johreh, 2020-06-23:
> Oh, I completely forgot about that feature. Then your patch looks great! Thanks for being patient with me here.

No problem.

I finally got around to complete the changes for a merge request now: #640

closed 2020-10-01 by phkahler

Symbian9, 2021-05-21:
> DXF file manifesting problem.

JFTR, I found that arc in linked `test1 v2.dxf` file has doubled (I'm not sure is it issue of DXF itself or is it issue of SolveSpace import).

I'll take a look on few other examples of 3D DXF, for example here are `cube.dxf`, `diamond.dxf` and `bridge.dxf`:

- https://people.math.sc.edu/Burkardt/data/dxf/dxf.html