solvespace/solvespace #630: SolveSpace crashes on launch (MacOS 10.15.5)

davehylands, 2020-06-12:
### System information

SolveSpace version: 
v2.3 - from https://github.com/solvespace/solvespace/releases/download/v2.3/solvespace.dmg

Operating system: 
MacOS 10.15.5

### Expected behavior

Program should have launched

### Actual behavior

Program crashed

### Additional information

Crashlog:
```
Process:               solvespace [94221]
Path:                  /Applications/solvespace.app/Contents/MacOS/solvespace
Identifier:            solvespace
Version:               2.3 (2.3)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           solvespace [94221]
User ID:               502

Date/Time:             2020-06-12 13:26:36.932 -0700
OS Version:            Mac OS X 10.15.5 (19F101)
Report Version:        12
Bridge OS Version:     4.5 (17P5300)
Anonymous UUID:        F036FB4D-DC64-6537-A39E-F2B855756A80

Sleep/Wake UUID:       12D8B8D6-5265-4C65-B771-FB58E7504094

Time Awake Since Boot: 420000 seconds
Time Since Wake:       17000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x4] Symbol missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Symbol not found: _inflateValidate
  Referenced from: /System/Library/Frameworks/ImageIO.framework/Versions/A/Resources/libPng.dylib
  Expected in: /Applications/solvespace.app/Contents/MacOS/libz.dylib
 in /System/Library/Frameworks/ImageIO.framework/Versions/A/Resources/libPng.dylib

Binary Images:
       0x10b15b000 -        0x10b591fff +solvespace (2.3 - 2.3) <225A132B-F702-34BF-8509-81A4175E4DE0> /Applications/solvespace.app/Contents/MacOS/solvespace
       0x10b6bc000 -        0x10b6dfffb +libpng.dylib (0) <E20FFBFB-7682-344D-8B29-F563F3F76603> /Applications/solvespace.app/Contents/MacOS/libpng.dylib
       0x10b6ec000 -        0x10b6fdff7 +libz.dylib (61.20.1) <B3EBB42F-48E3-3287-9F0D-308E04D407AC> /Applications/solvespace.app/Contents/MacOS/libz.dylib
       0x10b706000 -        0x10b785fff +libfreetype.dylib (19.3) <DCED11A8-2810-347A-86B4-FCFB9D215FC2> /Applications/solvespace.app/Contents/MacOS/libfreetype.dylib
       0x10fba6000 -        0x10fc37eff  dyld (750.5) <E4698FBD-806A-3396-B279-E685BA37430B> /usr/lib/dyld
    0x7fff31a1b000 -     0x7fff31a1bfff  com.apple.Accelerate (1.11 - Accelerate 1.11) <56DFF715-6A4E-3231-BDCC-A348BCB05047> /System/Library/Frameworks/Accelerate.framework/Versions/A/Accelerate
    0x7fff31a33000 -     0x7fff32089fff  com.apple.vImage (8.1 - 524.2.1) <17C93AB9-1625-3FDB-9851-C5E77BBE3428> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vImage.framework/Versions/A/vImage
    0x7fff3208a000 -     0x7fff322f1ff7  libBLAS.dylib (1303.60.1) <CBC28BE4-3C78-3AED-9565-0D625251D121> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib
    0x7fff322f2000 -     0x7fff327c5fef  libBNNS.dylib (144.100.2) <8D653678-1F9B-3670-AAE2-46DFB8D37643> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBNNS.dylib
    0x7fff327c6000 -     0x7fff32b61fff  libLAPACK.dylib (1303.60.1) <F8E9D081-7C60-32EC-A47D-2D30CAD73C5F> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libLAPACK.dylib
[... 243 lines cut ...]
USB Device: USB3.0 Hub
USB Device: USB3.0 Card Reader
USB Device: USB 3.1 Bus
USB Device: USB2.0 Hub
USB Device: Pyboard Virtual Comm Port in FS Mode
USB Device: USB2.0 Hub
USB Device: USB Optical Mouse
USB Device: HID to SPI Device
USB Device: USB2.0 Hub
USB Device: Natural® Ergonomic Keyboard 4000
USB Device: HD Pro Webcam C920
USB Device: USB Advanced Audio Device
USB Device: USB2.0 Hub
USB Device: Anker USB-C Hub Device
USB Device: Apple T2 Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.4
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.4
```

whitequark, 2020-06-12:
Duplicate of #398, see that issue for a workaround. We're aware of how irritating this issue is and are working on it.

davehylands, 2020-06-12:
FYI - I was able to build the latest master from source and it launches fine.

davehylands, 2020-06-12:
Closing this since its a duplicate (I searched for MacOS and didn't find any reported issues - I guess I should have searched for just Mac).

closed 2020-06-12 by davehylands

masonhensley, 2020-09-18:
I understand this is a duplicate, posting this for future internet searchers :)

Mac OS Big Sur Solve Space Crash

```
Process:               solvespace [16862]
Path:                  /Applications/solvespace.app/Contents/MacOS/solvespace
Identifier:            solvespace
Version:               2.3 (2.3)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           solvespace [16862]
User ID:               502

Date/Time:             2020-09-18 09:57:56.701 -0400
OS Version:            Mac OS X 10.16 (20A5364e)
Report Version:        12
Bridge OS Version:     5.0 (18P50370a)
Anonymous UUID:        2E6CCD1D-1257-852E-E84D-2A8ECB911228

Time Awake Since Boot: 690 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x4] Symbol missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Symbol not found: _inflateValidate
  Referenced from: /System/Library/Frameworks/ImageIO.framework/Versions/A/Resources/libPng.dylib
  Expected in: /Applications/solvespace.app/Contents/MacOS/libz.dylib
 in /System/Library/Frameworks/ImageIO.framework/Versions/A/Resources/libPng.dylib

Binary Images:
       0x101a2d000 -        0x101e63fff +solvespace (2.3 - 2.3) <225A132B-F702-34BF-8509-81A4175E4DE0> /Applications/solvespace.app/Contents/MacOS/solvespace
       0x101f8e000 -        0x101fb1fff +libpng.dylib (0) <E20FFBFB-7682-344D-8B29-F563F3F76603> /Applications/solvespace.app/Contents/MacOS/libpng.dylib
       0x101fba000 -        0x101fcbfff +libz.dylib (61.20.1) <B3EBB42F-48E3-3287-9F0D-308E04D407AC> /Applications/solvespace.app/Contents/MacOS/libz.dylib
       0x101fd1000 -        0x102050fff +libfreetype.dylib (19.3) <DCED11A8-2810-347A-86B4-FCFB9D215FC2> /Applications/solvespace.app/Contents/MacOS/libfreetype.dylib
       0x106092000 -        0x106129fff  dyld (831.1) <25F70DA4-3BC1-3C3F-A11B-86BA76423654> /usr/lib/dyld
    0x7fff21ece000 -     0x7fff21f23fff  libc++.1.dylib (904.4) <B49E52A8-5890-37F3-B293-E34CAD11C08F> /usr/lib/libc++.1.dylib
    0x7fff21f24000 -     0x7fff21f68fff  libdispatch.dylib (1271.40.4) <F78DAE75-7F92-3E5A-8F1B-950E29AAC36A> /usr/lib/system/libdispatch.dylib
    0x7fff21f69000 -     0x7fff21f97fff  libsystem_kernel.dylib (7195.40.65.0.2) <3E3C3209-ADB4-3C9A-B473-04D5222BC09B> /usr/lib/system/libsystem_kernel.dylib
    0x7fff21f98000 -     0x7fff21fcdfff  libxpc.dylib (2038.40.23.161.1) <4CDB88B7-8565-3279-84EE-E177A8720863> /usr/lib/system/libxpc.dylib
    0x7fff21fce000 -     0x7fff22069fff  libcorecrypto.dylib (1000.40.6) <81FDB929-550E-3979-82CE-0137EEF683A7> /usr/lib/system/libcorecrypto.dylib
    0x7fff2206a000 -     0x7fff22075fff  libsystem_pthread.dylib (454) <3A7C10DF-F4A2-3F13-8D57-C0FEDA39CA0D> /usr/lib/system/libsystem_pthread.dylib
    0x7fff22076000 -     0x7fff220b0fff  libdyld.dylib (831.1) <858A16D5-A205-3A68-BA4E-AF40C8535036> /usr/lib/system/libdyld.dylib
    0x7fff220b1000 -     0x7fff220b1fff  libsystem_product_info_filter.dylib (8.40.1) <2A69FB74-F6E5-3EA0-A910-B99F1AC7646B> /usr/lib/system/libsystem_product_info_filter.dylib
    0x7fff220b2000 -     0x7fff2213afff  libsystem_c.dylib (1439.40.5) <B3693487-9AD0-37BD-87AD-9E7C0E154D48> /usr/lib/system/libsystem_c.dylib
    0x7fff2213b000 -     0x7fff2213dfff  libsystem_featureflags.dylib (28.40.1) <6FB9D697-3422-3AB9-9C96-80D24E75E734> /usr/lib/system/libsystem_featureflags.dylib
    0x7fff2213e000 -     0x7fff22147fff  libsystem_platform.dylib (254.40.2) <B0E0B420-D6E6-3702-848D-58DC03CED2FE> /usr/lib/system/libsystem_platform.dylib
    0x7fff22148000 -     0x7fff22174fff  libsystem_malloc.dylib (317.40.1) <D9DD1EEE-9A9E-3562-A0A9-4EBA66E347FF> /usr/lib/system/libsystem_malloc.dylib
    0x7fff22175000 -     0x7fff22176fff  libsystem_blocks.dylib (78) <2DF0AF78-6E04-3400-B86B-BA6C925FFAAB> /usr/lib/system/libsystem_blocks.dylib
    0x7fff22177000 -     0x7fff221affff  libobjc.A.dylib (818) <706E4CF3-3950-3C3B-84A3-B4650C3AEFBC> /usr/lib/libobjc.A.dylib
    0x7fff221b0000 -     0x7fff221c7fff  libsystem_trace.dylib (1277.40.12) <562A0202-F0BD-34F0-8520-BBB30DD09A2A> /usr/lib/system/libsystem_trace.dylib
    0x7fff221c8000 -     0x7fff221e0fff  libc++abi.dylib (904.4) <1F5DBA15-70FC-388A-84C0-5BC47761166C> /usr/lib/libc++abi.dylib
    0x7fff221e1000 -     0x7fff2222cfff  libsystem_info.dylib (542.40.1) <961C2519-EEC0-39D1-B7F8-583F283577FB> /usr/lib/system/libsystem_info.dylib
    0x7fff2222d000 -     0x7fff22729fff  com.apple.CoreFoundation (6.9 - 1766.100) <1AEC4FBF-1C93-31A2-967D-5341FA79E25A> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
    0x7fff2272a000 -     0x7fff2296dfff  com.apple.LaunchServices (1116 - 1116) <A9D33178-4810-3C9F-97FC-E20EE92C9CBB> /System/Library/Frameworks/CoreServices.framework/Versions/A/Frameworks/LaunchServices.framework/Versions/A/LaunchServices
    0x7fff2296e000 -     0x7fff229e9fff  com.apple.SystemConfiguration (1.20 - 1.20) <00FB82B3-A6F4-3C56-B908-2F851470E7C2> /System/Library/Frameworks/SystemConfiguration.framework/Versions/A/SystemConfiguration
    0x7fff229ea000 -     0x7fff22a58fff  libSparse.dylib (106) <08BFA23B-03C9-3BE6-916D-B5D8F1165F68> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libSparse.dylib
[... cut ...]
    0x7fff35c00000 -     0x7fff35c07fff  com.apple.speech.recognition.framework (6.0.3 - 6.0.3) <13E8F251-320A-3BD4-A568-952C2F04DCF4> /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/SpeechRecognition.framework/Versions/A/SpeechRecognition

Model: Macmini8,1, BootROM 1554.40.13.0.0 (iBridge: 18.16.12370.5.1,0), 6 processors, 6-Core Intel Core i5, 3 GHz, 32 GB, SMC 
Graphics: kHW_IntelUHDGraphics630Item, Intel UHD Graphics 630, spdisplays_builtin
Memory Module: BANK 0/ChannelA-DIMM0, 16 GB, DDR4, 2667 MHz, 859B, CT16G4SFD8266.M16FE
Memory Module: BANK 2/ChannelB-DIMM0, 16 GB, DDR4, 2667 MHz, Micron, 16ATF2G64HZ-2G6E1
REDACTED

```