solvespace/solvespace #634: Tab accelerator key isn't supported by gtk

carrotIndustries, 2020-06-20:
### System information

SolveSpace version: git master 0da4a6b78a5d5bfd8f4e1c5666108d5b58696230

Operating system: Arch Linux, X11

### Expected behavior

Pressing <kbd>Tab</kbd> should toggle the property browser

### Actual behavior

Nothing happens. This is due to Gtk not considering <kbd>Tab</kbd> to to be a valid accelerator: https://gitlab.gnome.org/GNOME/gtk/-/blob/gtk-3-24/gtk/gtkaccelgroup.c#L982

### Suggested fixes

 - Use a different default accelerator
 - Roll our own `gtk_accel_groups_activate` and hook it up to the appropriate signal in `GtkWindow` or so to work around `gtk_accelerator_valid`

whitequark, 2020-06-20:
I'm confused. That file hasn't changed in years, and it works just fine on my machine. How did it break?

whitequark, 2020-06-21:
More to the topic, we already handle a few not-quite-accelerators in `OnKeyPress` or a related event; Tab could go there too.

whitequark, 2020-06-21:
Actually, we already do that in https://github.com/solvespace/solvespace/blob/0da4a6b78a5d5bfd8f4e1c5666108d5b58696230/src/graphicswin.cpp#L253-L260

You're going to have to figure out why that code path is never reached, since Tab works fine for me locally.

Symbian9, 2020-06-21:
> Nothing happens. This is due to Gtk not considering <kbd>Tab</kbd> to to be a valid accelerator:
>
> - OS: Arch Linux, X11,
> - GTK: 3.24

JFTR, Strange, because toggling *"Property Browser"* via <kbd>Tab</kbd> key works for me:

- OS: MX Linux MX-19.1 (Debian 10 buster), Xfce4
- GTK: 3.24.5

carrotIndustries, 2020-06-21:
> You're going to have to figure out why that code path is never reached, since Tab works fine for me locally.

Good to know that there's another code path.

https://github.com/solvespace/solvespace/blob/0da4a6b78a5d5bfd8f4e1c5666108d5b58696230/src/platform/guigtk.cpp#L568-L570

On my system apparently, state is GDK_MOD2_MASK even with no modifier pressed. Since the above code exits early if a modifier other than shift or ctrl is present `GraphicsWindow::KeyboardEvent` never gets called.

whitequark, 2020-06-21:
> On my system apparently, state is GDK_MOD2_MASK even with no modifier pressed.

Why does this happen?

carrotIndustries, 2020-06-21:
> Why does this happen?

On my system GDK_MOD2_MASK is set if Num Lock is on. If num lock is off the existing code works as intended.

As far as I can tell we can therefore safely ignore MOD2.

whitequark, 2020-06-21:
Does this help?
```patch
diff --git a/src/platform/guigtk.cpp b/src/platform/guigtk.cpp
index e68fd304..461ba60b 100644
--- a/src/platform/guigtk.cpp
+++ b/src/platform/guigtk.cpp
@@ -565,7 +565,7 @@ protected:
         KeyboardEvent event = {};
         event.type = type;
 
-        if(gdk_event->state & ~(GDK_SHIFT_MASK|GDK_CONTROL_MASK)) {
+        if(gdk_event->state & GDK_MOD2_MASK) {
             return false;
         }
 
```

whitequark, 2020-06-21:
Sorry, the earlier comment had a typo--I meant to use `GDK_MOD1_MASK` (fixed now).

carrotIndustries, 2020-06-21:
Depends on what we want to achieve:

 - Ignore MOD1 (Alt) specifically
 - Ignore all unknown modifiers such as super

From my point of view, we want the latter option since Super-tab shouldn't trigger our accelerator:

```diff
diff --git a/src/platform/guigtk.cpp b/src/platform/guigtk.cpp
index e68fd30..fe31466 100644
--- a/src/platform/guigtk.cpp
+++ b/src/platform/guigtk.cpp
@@ -565,7 +565,7 @@ protected:
         KeyboardEvent event = {};
         event.type = type;
 
-        if(gdk_event->state & ~(GDK_SHIFT_MASK|GDK_CONTROL_MASK)) {
+        if(gdk_event->state & ~(GDK_SHIFT_MASK|GDK_CONTROL_MASK|GDK_MOD2_MASK)) {
             return false;
         }
```

whitequark, 2020-06-21:
What other bugs remain in that code? Is there the same issue with ScrollLock and CapsLock?

carrotIndustries, 2020-06-21:
I had a closer look at the Gtk docs and I think I found what we need:

```diff
diff --git a/src/platform/guigtk.cpp b/src/platform/guigtk.cpp
index e68fd30..7ff0478 100644
--- a/src/platform/guigtk.cpp
+++ b/src/platform/guigtk.cpp
@@ -565,7 +565,8 @@ protected:
         KeyboardEvent event = {};
         event.type = type;
 
-        if(gdk_event->state & ~(GDK_SHIFT_MASK|GDK_CONTROL_MASK)) {
+        const auto mod_mask = get_modifier_mask(Gdk::MODIFIER_INTENT_DEFAULT_MOD_MASK);
+        if((gdk_event->state & mod_mask) & ~(GDK_SHIFT_MASK|GDK_CONTROL_MASK)) {
             return false;
         }
```

https://developer.gnome.org/gdk3/stable/gdk3-Windows.html#GdkModifierIntent
> The set of modifier masks accepted as modifiers in accelerators. Needed because Command is mapped to MOD2 on OSX, which is widely used, but on X11 MOD2 is NumLock and using that for a mod key is problematic at best.

commit d238f985 (2020-06-21): GTK: filter events more precisely.

Turns out that GDK_MOD2_MASK, at least on some systems, corresponds
to NumLock. Use the function provided by GDK to handle this.

Fixes #634.
closed 2020-06-21 by whitequark