solvespace/solvespace #635: Add "default solid model" configuration option

Symbian9, 2020-06-21:
### System information

SolveSpace version: `3.x`

Operating system: All supported

### Expected behavior

Add configuration option *"default solid model as: `union`|`assembly`|`difference`|`intersection`"* to be able set preferred default solid model type for new solid model groups.

> NOTE: Suggest, it would be better if by default "solid model" type would be `assembly` instead of `union`, as it prevent immediate boolean operations & it would be easier to tune next created solid before apply `union`/`difference`/`intersection` boolean operations with previous solid model groups.

### Actual behavior

Actually by default *"solid model as: "* set (hardcoded) to `union` for all new created groups.

As result, on complex models next created solid model group could cause whole model on `union` operation with previous solid models groups & its sometime hard to tune "united" solids.

### Additional information

whitequark, 2020-06-21:
This seems like it'd just annoy people who are used to the current behavior, and not do anything about the underlying performance issues.

Symbian9, 2020-06-21:
> This seems like it'd just annoy people who are used to the current behavior

That is why I request to add "option" instead of hardcoding single choice.

By default this "option" of course should be set to `union`, but user would be able to set other option on they wish.

whitequark, 2020-06-21:
Fair point.

Symbian9, 2020-06-21:
FTR, Maybe it would be enough just add checkbox configuration option *"☑ default solid model as `assembly`"*, which by default would be unchecked ☐ to provide `union` as "default solid model" (as it came from SolveSpace `v2.x`) for this case.

phkahler, 2020-06-21:
> This seems like it'd just annoy people who are used to the current behavior, and not do anything about the underlying performance issues.

AFAICT the entire "force to triangle mesh" system is a workaround for issues with NURBS booleans ;-)

I agree that I'd be very annoyed if this was a new default but I could see it as an option.

Symbian9, 2020-06-21:
> I could see it as an option.

Agree.

phkahler, 2023-08-10:
Closing.  The correct default should be union. Adding a user option to change it is probably some kind of work around. I'm not even a fan of allowing "assembly" as an option for non-assembly groups as it doesn't do a boolean at all and can lead to overlapping shells, and performance has improve significantly since then.

closed 2023-08-10 by phkahler