solvespace/solvespace #636: DXF export problems

ionapostol, 2020-06-21:
Hi,
My name is Ion Apostol and I downloaded your software which is great but...

I use Windows 10 laptop i7 7700HQ , 8Gb ram,  2x 256 SSD drive , 1050Ti nVidia 4Gb.
Solvespace v 2.3~7c1ca460

### Expected behavior
I tried both version 2.0 and 2.3 .
In v2.0 I exported a square with 50mm by 50mm and when I opened in eDrawings PRO from Dassault Systems the square was 1270mm by 1270mm.

In v2.3 the dxf export ...export only dimension lines the square is invisible.
I see that is an old problem. Is there a fix ?

[test.zip](https://github.com/solvespace/solvespace/files/4809848/test.zip)

phkahler, 2020-06-22:
Does this dxf file have the correct dimentions in eDrawings PRO?  I don't have access to it.
[test_export.zip](https://github.com/solvespace/solvespace/files/4810677/test_export.zip)

This was created with current master branch.

ionapostol, 2020-06-22:
Yes, from what I noticed is only a matter of colour of drawing. If is white is nit visible on exported dxf file ,but if I change colour to anything else then the line is visible in dxf file.

phkahler, 2020-06-22:
Is eDrawings PRO using a white background?  Is this just a white-on-white issue?

ionapostol, 2020-06-22:
No, I put one image with screenshot of eDrawings but I think the problem is on my side because now no matter if I change the colour I cannot see the square in eDrawings. I see only the dimension lines. Is strange because if I open the dxf file with  Lazycam for example the square is there and I can work with that file.
Unfortunatelly I cannot reinstall eDrawings because is part of Solidworks.( I will but in a distant future)
[dxf square white lines.zip](https://github.com/solvespace/solvespace/files/4814858/dxf.square.white.lines.zip)

phkahler, 2020-06-25:
Closing as requested by the OP who also seems to have solved their problem according to a post in the forum.

closed 2020-06-25 by phkahler