solvespace/solvespace #637: Add "digits after decimal point to show" for volume, area, perimeter & center of mass

Symbian9, 2020-06-23:
### System information

SolveSpace version: `3.0~d238f985`

Operating system: Debian 10 buster

### Expected behavior

There should be additional options to control *"digits after decimal point to show"* for `volume`, `area`, `perimeter` & `center of mass` (in same way as `distances` and `angles`)

### Actual behavior

Actually *"digits after decimal point to show"* for `volume`, `area`, `perimeter` & `center of mass` are hardcoded and there are no way to show those values preciser if need (especially for small solids & drawings).

### Additional information

whitequark, 2020-06-23:
> Actually _"digits after decimal point to show"_ for `volume`, `area`, `perimeter` & `center of mass` are hardcoded

That's not the case, which you would know if you read the code before making authoritative statements about the function of that code. Those values always use SI prefixes and they use the same amount of digits after the decimal point as distances do.

closed 2020-06-23 by whitequark

Symbian9, 2020-06-23:
> Those values always use SI prefixes and they use the same amount of digits after the decimal point as distances do.

At least not `center of mass` use SI prefixes.

Also `perimeter` follow the state of "use SI prefixes for distances".

whitequark, 2020-06-23:
> At least not `center of mass` entity use SI prefixes.

Fixed in 35c6af49.

> Also `perimeter` entities follow the state of _"use SI prefixes for distances"_ option.

A perimeter is a distance (alongside the contour, that is).

Symbian9, 2020-06-23:
> A `perimeter` is a `distance`(alongside the contour, that is)

If so, it should also use "digits after decimal point to show: `distances`", but actually `perimeter` is hardcoded to `3` digits after decimal point to show.

So, I propose to fix `perimeter` to follow "digits after decimal point to show: `distances`" option too.

whitequark, 2020-06-23:
> but actually `perimeter` is hardcoded to `3` digits after decimal point to show.

Of course it isn't.
[image]
[image]

Symbian9, 2020-06-23:
@whitequark, Sorry me here, I really mean `center of mass` in this comment above. Think, your latest commit on switching `center of mass` to hardcoded use of SI prefixes now fixes it.

Another one issue with `area` entity just found - is it hardcoded to `mm²`? Why it not recalculated to according SI prefixes (cm)for large areas?

[image]