solvespace/solvespace #638: Change style for "Center of Mass" entity

Symbian9, 2020-06-23:
### System information

SolveSpace version: `3.0~d238f985`

Operating system: Debian 10 buster

### Expected behavior

*"Center of Mass" entity* (which by design is *reference dimension*) should use `#def-constraint` style, which also used for *"show areas of closed contours" entities* and all other dimension-related entities (constraints).

### Actual behavior

*"Center of Mass" entity* use `#def-datum` style, which looks wrong.

### Additional information

whitequark, 2020-06-23:
The displayed center of mass is not an entity or a constraint; it's not a part of the sketch at all. The color is the same as for the origin of the plane shown when you export in 2d.

closed 2020-06-23 by whitequark

Symbian9, 2020-06-23:
> The displayed center of mass is not an entity or a constraint

OK, but I suggest it should be "entity".

May I request it here? It would be really useful for export drawings where "Center of Mass" could be shown.

whitequark, 2020-06-23:
It can't be an entity; given how SolveSpace currently works, entities (things you can constrain / constrain to) can't really be based on solid geometry.

On the other hand, exporting it seems reasonable.

Symbian9, 2020-06-23:
> It can't be an entity; given how SolveSpace currently works, entities (things you can constrain / constrain to) can't really be based on solid geometry.

Why not?

Suggest it `center of mass` entity is *similar* to `area` ("show are of closed contours") entity, but instead of `area` entity which position suggest is constrained to *center of mass of closed contour in 2D*, `center of mass` entity should be in 3D (similar to `comment` entities which already could be placed in 3D, but `center of mass` entity should not be dragged by mouse and its origin position should be always constrained to calculated center of mass position).

whitequark, 2020-06-23:
> Why not?

Because generating a sketch is a two-step process (first it is solved, second the solid geometry is generated based on the solution) and it would be difficult to change that. I'm also not really interested in any abstract discussion of features with you specifically, because you consistently fail to engage in such discussions in good faith.