solvespace/solvespace #641: Simple model and error when exporting to Export triangle mesch.

piotrb3d, 2020-06-24:
### System information
This is SolveSpace version 3.0~211ae8c4.
Operating system: Windows 7 x64

### Expected behavior
Simple model and error when exporting to Export triangle mesch.
[ochronnik.zip](https://github.com/solvespace/solvespace/files/4823516/ochronnik.zip)

[image]
### Additional information
Build in VS2017, no bugs in compilation.

ruevs, 2020-06-24:
I can not reproduce the crash. It exports and STL fine for me with 211ae8c44649532991c5fbedf40a735fa799ea81
What is your "export chord tolerance" ?

whitequark, 2020-06-24:
Yeah it sounds like a duplicate of #320

ruevs, 2020-06-24:
Ahh yes - with 0.01 chord tolerance and 200 segments:

```
Unhandled exception at 0x77289A09 (ntdll.dll) in solvespace.exe:
0xC00000FD: Stack overflow (parameters: 0x00000001, 0x00E02FFC).
```

So it is a stack overflow, and by the look of it exactly in bsp.cpp as in #320 

```
 	ntdll.dll!77289a08()	Unknown
 	ntdll.dll![Frames below may be incorrect and/or missing, no symbols loaded for ntdll.dll]	Unknown
 	[External Code]	
>	solvespace.exe!SolveSpace::Platform::AllocTemporary(unsigned int size) Line 694	C++
 	solvespace.exe!BspUtil::Alloc() Line 153	C++
 	solvespace.exe!SolveSpace::SBsp3::InsertConvex(SolveSpace::STriMeta meta, SolveSpace::Vector * vertex, unsigned int cnt, SolveSpace::SMesh * instead) Line 417	C++
 	solvespace.exe!SolveSpace::SBsp3::InsertConvexHow(SolveSpace::BspClass how, SolveSpace::STriMeta meta, SolveSpace::Vector * vertex, unsigned int n, SolveSpace::SMesh * instead) Line 401	C++
 	solvespace.exe!SolveSpace::SBsp3::InsertConvex(SolveSpace::STriMeta meta, SolveSpace::Vector * vertex, unsigned int cnt, SolveSpace::SMesh * instead) Line 421	C++
 	solvespace.exe!SolveSpace::SBsp3::InsertConvexHow(SolveSpace::BspClass how, SolveSpace::STriMeta meta, SolveSpace::Vector * vertex, unsigned int n, SolveSpace::SMesh * instead) Line 401	C++
[SKIP ~3000] 	
 	solvespace.exe!SolveSpace::SBsp3::InsertConvex(SolveSpace::STriMeta meta, SolveSpace::Vector * vertex, unsigned int cnt, SolveSpace::SMesh * instead) Line 421	C++
 	solvespace.exe!SolveSpace::SBsp3::InsertConvexHow(SolveSpace::BspClass how, SolveSpace::STriMeta meta, SolveSpace::Vector * vertex, unsigned int n, SolveSpace::SMesh * instead) Line 401	C++
 	solvespace.exe!SolveSpace::SBsp3::Insert(SolveSpace::STriangle * tr, SolveSpace::SMesh * instead) Line 506	C++
 	solvespace.exe!SolveSpace::SBsp3::InsertOrCreate(SolveSpace::SBsp3 * where, SolveSpace::STriangle * tr, SolveSpace::SMesh * instead) Line 461	C++
 	solvespace.exe!SolveSpace::SBsp3::InsertHow(SolveSpace::BspClass how, SolveSpace::STriangle * tr, SolveSpace::SMesh * instead) Line 90	C++
[SKIP ~600]
 	solvespace.exe!SolveSpace::SBsp3::Insert(SolveSpace::STriangle * tr, SolveSpace::SMesh * instead) Line 485	C++
 	solvespace.exe!SolveSpace::SBsp3::InsertOrCreate(SolveSpace::SBsp3 * where, SolveSpace::STriangle * tr, SolveSpace::SMesh * instead) Line 461	C++
 	solvespace.exe!SolveSpace::SBsp3::InsertHow(SolveSpace::BspClass how, SolveSpace::STriangle * tr, SolveSpace::SMesh * instead) Line 90	C++
 	solvespace.exe!SolveSpace::SBsp3::Insert(SolveSpace::STriangle * tr, SolveSpace::SMesh * instead) Line 485	C++
 	solvespace.exe!SolveSpace::SBsp3::InsertOrCreate(SolveSpace::SBsp3 * where, SolveSpace::STriangle * tr, SolveSpace::SMesh * instead) Line 461	C++
 	solvespace.exe!SolveSpace::SBsp3::FromMesh(const SolveSpace::SMesh * m) Line 27	C++
 	solvespace.exe!SolveSpace::SMesh::MakeFromDifferenceOf(SolveSpace::SMesh * a, SolveSpace::SMesh * b) Line 278	C++
 	solvespace.exe!SolveSpace::Group::GenerateForBoolean<SolveSpace::SMesh>(SolveSpace::SMesh * prevs, SolveSpace::SMesh * thiss, SolveSpace::SMesh * outs, SolveSpace::Group::CombineAs how) Line 180	C++
 	solvespace.exe!SolveSpace::Group::GenerateShellAndMesh() Line 417	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::GenerateAll(SolveSpace::SolveSpaceUI::Generate type, bool andFindFree, bool genForBBox) Line 275	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::ExportMeshTo(const SolveSpace::Platform::Path & filename) Line 814	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command id) Line 654	C++
 	[External Code]	
 	solvespace.exe!SolveSpace::Platform::WindowImplWin32::WndProc(HWND__ * h, unsigned int msg, unsigned int wParam, long lParam) Line 1041	C++
 	[External Code]	
 	solvespace.exe!SolveSpace::Platform::RunGui() Line 1679	C++
 	solvespace.exe!main(int argc, char * * argv) Line 29	C++
 	solvespace.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 40	C++
 	[External Code]	
```

whitequark, 2020-06-24:
Closing as duplicate

closed 2020-06-24 by whitequark