solvespace/solvespace #654: Point constrained to line has extra degree of freedom

socialhacker, 2020-07-18:
### System information

SolveSpace version: e.g. 3.0~211ae8c4

Operating system: Arch Linux x86_64

### Expected behavior

When a point is constrained to lie on a line that is perpendicular to the workplane I expect that would account for all of the degrees of freedom of the point.

### Actual behavior

The group that contains the point (and nothing else) reports one degree of freedom unconstrained.  But the point can't be moved.

### Additional information

Here's a minimal test case:

[point_on_line_dof.slvs.gz](https://github.com/solvespace/solvespace/files/4941128/point_on_line_dof.slvs.gz)

Symbian9, 2020-07-18:
Probably duplicate of https://github.com/solvespace/solvespace/issues/599

phkahler, 2020-07-18:
Yes, what @Symbian9 said. There is no way to know that the line will never be in-plane, in which case you'd have 1 DoF for the point. Constrain to a point instead of a line.

socialhacker, 2020-07-18:
Agreed, I missed that one because it was already closed and I only looked through the open issues.  :)

closed 2020-07-18 by socialhacker