solvespace/solvespace #657: adding mimalloc breaks running on Windows 10

steel-ne, 2020-07-22:
### System information

SolveSpace version: 3.0~cbac182e
next commit is broken

Operating system: Windows 10

### Expected behavior

solvespace runs and shows GUI on windows 10

### Actual behavior

solvespace is started and shows no GUI, task is present in task manager

### Additional information
commit cbac182edfa10743393cfba56ffd1ecd74006ca0 is good
commit c4ca4be91e0f38646fa391aaca6fd58e505be4c4 is bad

whitequark, 2020-07-22:
Can't reproduce. Could you provide more details?

nabijaczleweli, 2020-07-22:
Interesting, I did actually test this on (an admittedly ancient build of) Windows 10 and it certainly worked, insofar as showing models and spinning them around, at least

whitequark, 2020-07-22:
Works fine for me under Wine and a Win10 VM as well.

nabijaczleweli, 2020-07-22:
I nicked a Toshiba R700 and X40 with Windows version 1909 (build 18363.959) and 2004 (19041.329) respectively, and my GCC 9.3.0 build of the latter commit works there as well.

steel-ne, 2020-07-22:
I've downloaded the last snapshot build from https://ci.appveyor.com/project/whitequark/solvespace/build/artifacts
Doubleclick .exe file, nothing happened, but in the Task manager it appears as "SolveSpace, a parametric 2d/3d CAD (32 bit)"

So I start Git bisect, building on Windows with Visual studio 2019

Windows 10 Enterprise v1909 build 18363.959

It happens on HP notebook too (build 18363.778)

I have domain logons on both devices with some restrictions (can't install/run some programs, etc).

whitequark, 2020-07-22:
> building on Windows with Visual studio 2019

Since you have Visual Studio, can you attach a debugger and determine the backtrace for your hang?

steel-ne, 2020-07-22:
I do not so familiar with VS, here is call stack of thread main:
```

>	solvespace.exe!rx::DisplayWGL::terminate() Line 374	C++
 	solvespace.exe!egl::Display::terminate() Line 463	C++
 	solvespace.exe!egl::Terminate(void * dpy) Line 94	C++
 	solvespace.exe!eglTerminate(void * dpy) Line 118	C++
 	solvespace.exe!SolveSpace::Platform::WindowImplWin32::WindowImplWin32(SolveSpace::Platform::Window::Kind kind, std::shared_ptr<SolveSpace::Platform::WindowImplWin32> parentWindow) Line 647	C++
 	[External Code]	
 	solvespace.exe!SolveSpace::Platform::CreateWindow(SolveSpace::Platform::Window::Kind kind, std::shared_ptr<SolveSpace::Platform::Window> parentWindow) Line 1380	C++
 	solvespace.exe!SolveSpace::GraphicsWindow::Init() Line 413	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::AfterNewFile() Line 437	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::Init() Line 141	C++
 	solvespace.exe!main(int argc, char * * argv) Line 19	C++
 	solvespace.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 40	C++
 	[External Code]	

```

whitequark, 2020-07-22:
Thanks, that's useful.

steel-ne, 2020-07-22:
here is the callstack with "External code" unwrapped:

```
 	ntdll.dll!_NtWaitForSingleObject@12()	Unknown
 	ntdll.dll!LdrpDrainWorkQueue()	Unknown
 	ntdll.dll!_LdrUnloadDll@4()	Unknown
 	KernelBase.dll!FreeLibrary()	Unknown
 	ig9icd32.dll!686e7843()	Unknown
 	ig9icd32.dll![Frames below may be incorrect and/or missing, no symbols loaded for ig9icd32.dll]	Unknown
 	ig9icd32.dll!688950b0()	Unknown
 	ig9icd32.dll!6870fe96()	Unknown
 	ig9icd32.dll!686cf19e()	Unknown
 	opengl32.dll!6935d1df()	Unknown
>	solvespace.exe!rx::DisplayWGL::terminate() Line 374	C++
 	solvespace.exe!egl::Display::terminate() Line 463	C++
 	solvespace.exe!egl::Terminate(void * dpy) Line 94	C++
 	solvespace.exe!eglTerminate(void * dpy) Line 118	C++
 	solvespace.exe!SolveSpace::Platform::WindowImplWin32::WindowImplWin32(SolveSpace::Platform::Window::Kind kind, std::shared_ptr<SolveSpace::Platform::WindowImplWin32> parentWindow) Line 647	C++
 	solvespace.exe!std::_Ref_count_obj<SolveSpace::Platform::WindowImplWin32>::_Ref_count_obj<SolveSpace::Platform::WindowImplWin32><enum SolveSpace::Platform::Window::Kind &,std::shared_ptr<SolveSpace::Platform::WindowImplWin32>>(SolveSpace::Platform::Window::Kind & <_Args_0>, std::shared_ptr<SolveSpace::Platform::WindowImplWin32> && <_Args_1>) Line 901	C++
 	solvespace.exe!std::make_shared<SolveSpace::Platform::WindowImplWin32,enum SolveSpace::Platform::Window::Kind &,std::shared_ptr<SolveSpace::Platform::WindowImplWin32>>(SolveSpace::Platform::Window::Kind & <_Args_0>, std::shared_ptr<SolveSpace::Platform::WindowImplWin32> && <_Args_1>) Line 971	C++
 	solvespace.exe!SolveSpace::Platform::CreateWindow(SolveSpace::Platform::Window::Kind kind, std::shared_ptr<SolveSpace::Platform::Window> parentWindow) Line 1380	C++
 	solvespace.exe!SolveSpace::GraphicsWindow::Init() Line 413	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::AfterNewFile() Line 437	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::Init() Line 141	C++
 	solvespace.exe!main(int argc, char * * argv) Line 19	C++
 	solvespace.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 40	C++
 	solvespace.exe!invoke_main() Line 99	C++
 	solvespace.exe!__scrt_common_main_seh() Line 253	C++
 	solvespace.exe!__scrt_common_main() Line 296	C++
 	solvespace.exe!WinMainCRTStartup() Line 17	C++
 	kernel32.dll!@BaseThreadInitThunk@12()	Unknown
 	ntdll.dll!__RtlUserThreadStart()	Unknown
 	ntdll.dll!__RtlUserThreadStart@8()	Unknown

```

whitequark, 2020-07-22:
@nabijaczleweli I have a suspicion of what's happening here. It looks like it hangs in FreeLibrary. This likely happens because a mimalloc heap is created (because of the way C++ thread locals work) for every thread, including those in the Intel integrated graphics driver, and for some reason this causes a deadlock. I think we can fix it by lazily creating a heap.

whitequark, 2020-07-22:
@steel-ne Does this help?
```diff
diff --git a/src/platform/platform.cpp b/src/platform/platform.cpp
index 934220ca..a0809821 100644
--- a/src/platform/platform.cpp
+++ b/src/platform/platform.cpp
@@ -684,20 +684,21 @@ void DebugPrint(const char *fmt, ...) {
 //-----------------------------------------------------------------------------
 
 struct MimallocHeap {
-    mi_heap_t *heap = mi_heap_new();
-
-    MimallocHeap() {
-        ssassert(heap != NULL, "out of memory");
-    }
+    mi_heap_t *heap = NULL;
 
     ~MimallocHeap() {
-        mi_heap_destroy(heap);
+        if(heap != NULL)
+            mi_heap_destroy(heap);
     }
 };
 
 static thread_local MimallocHeap TempArena;
 
 void *AllocTemporary(size_t size) {
+    if(TempArena.heap == NULL) {
+        TempArena.heap = mi_heap_new();
+        ssassert(TempArena.heap != NULL, "out of memory");
+    }
     void *ptr = mi_heap_zalloc(TempArena.heap, size);
     ssassert(ptr != NULL, "out of memory");
     return ptr;
```

steel-ne, 2020-07-22:
Yes, it works!

closed 2020-07-22 by whitequark

nabijaczleweli, 2020-07-22:
Interestingly, none of the laptops nor my desktop currently have dedicated GPUs in them; just driver weirdness, I guess?

whitequark, 2020-07-22:
> just driver weirdness, I guess?

Yeah. I think it actually oughtn't crash even the way we wrote it, it might be actually a bug in the driver. But it's good to be defensive here because drivers are full of bugs :)

commit b0fc1912 (2020-07-22): Create mimalloc heaps only when necessary.

The mimalloc temporary heap is a thread-local object that uses RAII
to manage heap lifetimes even in threads that are created implicitly,
e.g. by OpenMP. However, not all threads are necessarily created by
the application; graphics drivers may create their own threads, and
this can lead to deadlocks when combined with library unloading.

Fixes #657.