solvespace/solvespace #660: HIDPI: GDK_SCALE is not being respected

cheuksing, 2020-07-25:
### System information

SolveSpace version: 2.3-3 (solvespace AUR)

Operating system: Manjaro i3 (arch linux)

### Expected behavior

Icon size and font size should be larger.

### Actual behavior

They are too small.

### Additional information

1. GDK_SCALE is set to 2 within ~/.profile
2. Xft.dpi is set to 192.
3. This is a 4k display.
4. Running `env GDK_SCALE=1 solvespace` has no diff

When using snapd version, system gtk theme is not using, but the scaling is correct.
When using manually build version, gtk theme is applied, but icon and font size are small.

Here's a side-to-side comparison.

[image]

whitequark, 2020-07-25:
Version 2.3 does not support HiDPI. Either use snaps, or build from source until 3.0 is released.

ruevs, 2020-10-22:
@cheuksing did you try building master? Does it work?

closed 2021-05-16 by phkahler