solvespace/solvespace #663: Nested assembly keeping the inner-most "suppress this solid" setting?

rpavlik, 2020-07-26:
### System information

SolveSpace version: 3.0~060a3280

Operating system: Debian Buster

### Expected behavior

Three models, A, B, and C.

- A ("syringe plunger") is "the environment" - the thing I'm attaching to, so not part of what I'm making.
- B ("carriage plunger part") is the part that interacts with the environment.
- C ("carriage") was to be the overall part. C includes B, and B includes A but suppresses from the solid. I'd expect to see the solids of B in C.

### Actual behavior

A in B looks right - I only see the lines, not the solid, or "part A"

[image]

B shows up in C as wireframe-only as if it were also "suppress from the solid".

[image]

Here are the files:

[testcase.zip](https://github.com/solvespace/solvespace/files/4977989/testcase.zip)

### Additional information

I did note a weird drawing of "chords" in the face of the cylindrical part of my model B ("carriage plunger part") so I fear it might be slightly numerically unstable or something. Just backed up a single extrude group to see it:

[image]

I tried making a minimal repro and it didn't show the issue... Not sure why since I made all of these this morning using the same build (and actually, the same execution) of SolveSpace.

Here are my sample files that do **not** show the problem, presumably there's some subtle difference between them and my "real" models I didn't identify.  Maybe the same thing causing the "chords" ? I did end up having to turn to trimeshed instead of nurbs fairly early ("g007-extrude" of "g006-top-pocket", essentially my second sketch) because apparently adjacent curved faces aren't a lot of fun for the nurbs engine, even with slightly different diameters.

[these-work-fine.zip](https://github.com/solvespace/solvespace/files/4977990/these-work-fine.zip)

whitequark, 2020-07-26:
In the "Actual behavior" screenshot, can you show the group list?

rpavlik, 2020-07-27:
sure

[image]

I think it's something about my "A" model (syringe plunger) - I made another design using it (tried to surround it to make a "negative" to subtract from a larger solid), and that showed the same issue - it definitely didn't include the solid because it didn't cut any holes. It's not just a rendering thing, I exported as STL and the holes were missing there too.

[image]

[carriage-try2.zip](https://github.com/solvespace/solvespace/files/4982789/carriage-try2.zip)