solvespace/solvespace #666: Unwanted red edges

phkahler, 2020-08-02:
### System information

SolveSpace version: 3.0~fd2dfe87

Operating system: Fedora 31

### Expected behavior

Model should display with no red edges.

### Actual behavior

Model has red edges, or even stray red segments.

### Additional information
When the 8 circles cut into the center hole or sometimes the outer ring, these red edges appear. Analyze indicates no naked edges. I'm opening this issue to document it before submitting a PR that was not intended to fix it but does.

[image]

phkahler, 2020-08-03:
Looking through the text window I see several "can't find a ray that doesn't hit on edge!" messages. They go away with the patch.

ruevs, 2020-08-03:
Please attach the model.

phkahler, 2020-08-03:
[A_wheel.zip](https://github.com/solvespace/solvespace/files/5016551/A_wheel.zip)

Remember, random numbers are a platform dependent implementation so you may not see this problem on Windows. OTOH it is very easy to get multiple failures on my setup.

closed 2020-08-03 by phkahler