solvespace/solvespace #669: Unable to set arc centers coincide contraints

dims12, 2020-08-07:
### System information

SolveSpace version: 2.3~7c1ca460

Operating system: Windows 10 Pro

### Expected behavior

Constaint should be possible to set.

### Actual behavior

Impossible.

### Additional information

https://youtu.be/2rRu4qKlFw4

whitequark, 2020-08-07:
Please share the file you're working on.

nabijaczleweli, 2020-08-07:
The model seems to have been added ("icon.zip").

phkahler, 2020-08-08:
@dims12  You might try selecting both arcs and setting them to equal diameter "d". Once you do that I think the pair of arcs only has one degree of freedom, so each one independently has 1 right now and constraining the center point would require 2DoF for each arc.

When I do things like this, I usually make a construction line/circle. Then constrain arc end points to it and set their diameters equal to that of the construction circle, this avoids constraining the center point which seems to over-constrain in unintuitive ways. If you want coincident centers, another option is to constrain those first, then you'll also be less likely to run into this.

ruevs, 2025-09-04:
This is normal behaviour as explained. Either "allow redundant constraints" or (better) use a different set of constraints.

MikePryadko, 2025-12-24:
Hi there!
I'm new with SolveSpace (switched from FreeCAD because I was tired of its bugs) and the very first steps caused me pain 😐 
There is an `Arc`, it has 3 points: center and start/end -- but when you try to constrain them you got error (workspace become red and red text says `SOLVE FAILED!`) in any cases 🤯 even more: SolveSpace provide a list of problematic constrains, but deleting any of it get `1 DOF` 😅  is it kind of crazy joke?
@phkahler advises to use additional (construction) line/circle, but why additional lines is better than redundant constrains? Looks like `allow redundant constrains` is not an option but a necessity 😟 

Another pain is strict rules where constraints can be applied 😵 I can't just select 3 point and click `symmetric` -- why?

P.S. and first frustrating thing I encountered: you can't select workplane for new group by flat face or by two lines or by three points 😒 why?
SolveSpace looks cool but user  experience is same like with FreeCAD, except it has restrictions instead of bugs 😣

phkahler, 2025-12-24:
> [@phkahler](https://github.com/phkahler) advises to use additional (construction) line/circle, but why additional lines is better than redundant constrains? Looks like `allow redundant constrains` is not an option but a necessity 😟

@MikePryadko  an arc defined by 2 endpoints and a center point only has 5 degrees of freedom, not 6. I find that mathematical fact annoying too. If you constain the 2 end points, then you'll need to either constrain the radius/diameter or the arc length, or find a way to constrain the center point that reduces it by 1 degree of freedom. I often use a Vertical or Horizontal constraint between the center and one of the end points. Or you can constrain it on-line.

> Another pain is strict rules where constraints can be applied 😵 I can't just select 3 point and click `symmetric` -- why?

This took me a long time to figure out what you mean. I'm guessing you want to select 3 points A-B-C and have one of them become the mid-point between the other two. Is that right? We might be able to add an option to apply Mid-point constraint to 3 points instead of just a point and line.
 
> P.S. and first frustrating thing I encountered: you can't select workplane for new group by flat face or by two lines or by three points 😒 why?

Because internally solvespace doesn't currently store enough information about a "face" to put a U,V coordinate system on it. This was an oversite when it was first created. I hope to fix it one day, but it might break file compatibility.