solvespace/solvespace #670: Suggest to remove special handling of closed contours

dims12, 2020-08-07:
### System information

SolveSpace version: 2.3~7c1ca460

Operating system: Windows 10 Pro

### Expected behavior

I think any "graph loop" (in graph theory sense) should be treated as closed contour.

### Actual behavior

Only contours, built by closing tool, are treated as closed.

### Additional information

In attached file there are 2 closed contours, but the program doesn't understand

 it.

[not closed contours.zip](https://github.com/solvespace/solvespace/files/5042753/not.closed.contours.zip)

whitequark, 2020-08-07:
There's no such thing as a "closing tool". SolveSpace considers any contour with exactly one cycle a "closed contour". This is also the only kind of contour that is usable with 3d operations like extrusion, which is why a message is shown when the condition isn't satisfied.

closed 2020-08-07 by whitequark

dims12, 2020-08-07:
See an attachement. It has 2 closed contours, but the program say they aren't.

[image]

dims12, 2020-08-07:
If extrusion needs some conditions, namely application of extrusion tools should fail or warn. No need to indicate any errors if exctrusion tool was not called. Also no need to fail area calculation tool.