solvespace/solvespace #671: Redundant constraints are treated as errors

dims12, 2020-08-07:
### System information

SolveSpace version: 2.3~7c1ca460

Operating system: Windows 10 Pro

### Expected behavior

Redundant (not contradicting, but telling the same) constraints should be ok.

### Actual behavior

In case of redundant constraints, the program says "solve failed", which is wrong.

### Additional information

Redundant triangle attached

[redundant.zip](https://github.com/solvespace/solvespace/files/5042782/redundant.zip)

whitequark, 2020-08-07:
In group options you can tick the "allow redundant constraints" checkbox to suppress the error. However, SolveSpace's constraint solver does not handle many kinds of redundant constraints very well, which is why this is an error by default. Because it may significantly impair convergence, I recommend avoiding this option other than for assemblies.

closed 2020-08-07 by whitequark

phkahler, 2020-08-08:
@dims12 You can also select one of the angle constraints and convert it to a reference dimension "REF" on the toolbar. Then it will show the angle but not add the redundant constrain. Version 3 will do this automatically when the 3rd constraint is added.