solvespace/solvespace #674: Build broken on Windows via VC++

sherief, 2020-08-13:
### System information

SolveSpace version: f952991

Operating system: Windows 10 x64

### Expected behavior

Building with Visual Studio should be possible from CMake file

### Actual behavior

The generated project file for zlibstatic defines ASMINF and ASMV, but it doesn't assemble and link inffasx64.asm or gvmat64.asm, leading to undefined symbols for inflate_fast(), longest_match(), and others

### Additional information

N/A

whitequark, 2020-08-13:
Which Visual Studio version?

sherief, 2020-08-13:
Community 2019 Version 16.7.0

ruevs, 2020-10-15:
I build with VS 2019 with no problems all the time...

ruevs, 2020-11-23:
@sherief were you able to compile?

ruevs, 2020-11-26:
> @sherief were you able to compile?

Write here if you still have problems. Closing.

closed 2020-11-26 by ruevs