solvespace/solvespace #676: Not fully constrained shape when dragging with mouse

starsolaris, 2020-08-14:
### System information

SolveSpace version: 3.0 bc4924ae

Operating system: Windows 10

### Expected behaviour
When i try to move fully constrained shape by dragging upper-left vertex, nothing should happen, regardless of whether I move the mouse fast or slow.
[image]

### Actual behaviour
When I move the shape behind the vertex slowly, it doesn't move; if I move it quickly, it moves.
[image]

### Additional information
[quad.zip](https://github.com/solvespace/solvespace/files/5077359/quad.zip)

phkahler, 2020-08-14:
This is a known behaviour. When it flips the other direction all the constraints are still satisfied, so in some sense it is not fully constrained, but it takes effort to make it flip.

whitequark, 2020-08-15:
Yeah, duplicate of #76.

closed 2020-08-15 by whitequark

starsolaris, 2020-08-15:
I understand that second case is fully satisfy constraints also, but different options depending on the speed of movement of the mouse is very annoying. (i cannot flip shape if i will move mouse with slow speed)

whitequark, 2020-08-15:
I'm aware that it's annoying but you must understand that properly fixing this requires a significant redesign of the constraint system.