solvespace/solvespace #680: second revolve breaks the surface

zzj1213, 2020-08-16:
As shown on the test file:  the **second** revolve, based on the workplane resulted from the first resolve, breaks the surface.

Solvespave version: **Version 3.0 (3.0~bc4924ae)**,
macOS Catalina Version 10.15.2

[image]

[image]

[second_revolve_test.slvs.zip](https://github.com/solvespace/solvespace/files/5079984/second_revolve_test.slvs.zip)

zzj1213, 2020-08-16:
related to this issue: #439 

This consecutive  "revolve" workaround for Extrude along an arbitrary path seems a very important feature.

The path in #439 also has the same issue.

[image]

[path.zip](https://github.com/solvespace/solvespace/files/5079994/path.zip)

phkahler, 2020-08-16:
This looks like a problem with coincident surfaces - the end face of one revolve and the "bottom" face of the next one. Unfortunately this is common. As we learn more about the NURBs kernel some of these issues might get fixed.

If you don't need STEP export this does work if you check "force NURBS to triangle mesh" on the 3rd revolve group and continue construction.

Path extrusion would offer a more reliable way to make this, but it's still a long way off. This is exactly the kind of thing I could see being created as one piece (hence no NURBS problems), but you'd still need a difference or intersection to cut the bottom flat.

zzj1213, 2020-08-17:
The first end and second begin faces are carbon copy, but the NURBS kernel treats them different.   Is it because of approximation?

This brings up another important feature: **stitch**: filling the gap between NURBS faces when the faces are not a plane. 

After using Solvespace intensively for two weeks, really love it.  But in the meantime,  found out these two major limitations to handle serious jobs:
- path extrusion: revolve can be a workaround in some cases;
- stitch: no workaround for now.

zzj1213, 2020-08-17:
Ran into this: https://github.com/wsteffe/solvespace_occ 

First tests with **opencascade** bezier curves: 

https://github.com/wsteffe/solvespace_occ/commit/f186b74006feb40814917e6b7b5502b3f56d6a68

phkahler, 2020-10-14:
The problem here is now understood. There is an extra edge being created at the intersection of two planes - the vertical one at the end of the previous revolve and the top surface of the new one. This "curve" is going beyond the area of the surface polygon, as can be seen in the original picture. A couple of hacks confirm this.

Commenting out the call to MakeFirstOrderRevolvedSurfaces() after Revolve operations "fixes" the problem at the expense of leaving those surface with order 1x2 which prevents merging planes and constraining to them - so not a fix. Also, going in SSurface::IntersectAgainst() and aborting surface-surface intersection:
```
    if(degm == 1 && degn == 1 && b->degm == 1 && b->degn == 1) {
        return;
```
Makes the problem go away, but again that's not a solution.

That intersection edge is not being split in two and if it was I'm not sure if the outside portion would be discarded.

Unrelated: That "path.slvs" sketch mentioned above as a crude substitute for extrusion along a path works really well with current SolveSpace master branch. No more red or missing surfaces.

closed 2020-10-20 by phkahler

zzj1213, 2020-10-24:
It works fine now, thanks.