solvespace/solvespace #681: Button click area is shifted left of icon

developex-dmytro-i, 2020-08-20:
### System information

SolveSpace version: 3.0~bb56daf3

Operating system: Windows 10

### Expected behavior

Clicking Occluded Lines button must switch mode

### Actual behavior

The right side of the button is non-operate, no hovering or clicking events do nothing.

Seems the hover-and-click area is shifted left (see attached pic)

Solid model buttons are shifted too, but less. I think the width of spacers are not calculated right.

### Additional information

[image]

nabijaczleweli, 2020-08-20:
norepro on Windows at HEAD (6894b0c9c3e5ba3e080cf364d5f6e68a760b2b8c) or the reporter's SHA (bb56daf3015da4619766daca79a4200cd51749b9)

In case it's something to do with the GL config, mine is thus:
[image]

developex-dmytro-i, 2020-08-20:
[image]

ruevs, 2020-08-20:
Works for me on Windows as well at 04b332df HD Graphics 630 the rest as @developex-dmytro-i

developex-dmytro-i, 2020-08-20:
addition:
try to minimize width of Property browser window  (just behind Occlusion lines buttons), close Solvespace, run it again. The buttons are spread slightly widely, so that last button is shifted. When you try to resize browser's window, buttons rearranges normally and works good.

Before close:
[image]
After reopen:
[image]

nabijaczleweli, 2020-08-20:
Weirdly, I can't see that distortion either

developex-dmytro-i, 2020-08-20:
Ok, it will be my pain ))

kk6mrp, 2020-08-20:
I am using Gnome 3.34.4 and when I first open Solvespace the property window is slightly more narrow than the buttons but you can click them fine.
[image]

developex-dmytro-i, 2020-08-20:
> I am using Gnome 3.34.4 and when I first open Solvespace the property window is slightly more narrow than the buttons but you can click them fine.
You can click all the button surface or the right side is unclickable? For example green button "Show triangle..."

kk6mrp, 2020-08-20:
The buttons are all working correctly even after resizing the window. I posted that because it looked similar to your "After Reopen" picture.

ruevs, 2020-12-02:
@developex-dmytro-i  does your problem still happen on a recent build?
Please try the "Edge" from here https://github.com/solvespace/solvespace/releases

ruevs, 2021-04-07:
I **_think_** that I could reproduce this a few days back on Windows and then it disappeared?!?! Needs testing with all options cleared from the registry/json - just a guess...

ruevs, 2023-01-13:
I can consistently reproduce this on Windows by deleting the registry settings of SolveSpace (`Computer\HKEY_USERS\S-xxxx\SOFTWARE\SolveSpace` registry key).

After this the contents of the property browser window are scaled incorrectly - they are very slightly enlarged. This is the reason the icons and the "click" areas do not line up. As soon as the window is resized the scaling fixes itself.

In the screen shots below:
- The top one is the window after deleting the registry settings - the scaling is incorrect.
- The bottom one is after slightly resizing the window.

I added the vertical red lines to emphasize the problem.
[image]

The slight anti-aliasing of the top one is not an artifact of the screen shot - it is caused by the scaling and is visible on screen.

I did try to debug this a bit but did not immediately find the problem and ran out of time, I'll come back to it.

phkahler, 2023-02-01:
@ruevs   You might try calling TextWindow::Resize()  sometime after startup since resizing fixes the issue. If that works you can maybe track down something being not initialized correctly.

phkahler, 2023-12-11:
It's as if changing visibility of the scrollbar changes the width of the window but we don't change the size of our offscreen bitmap unless we get an actual window resize event. Changing scrollbar visibility by resizing the window works fine since we are getting resize events. Changing scrollbar visibility by changing text screens blurs the text because we don't get the resize event.

Do we explicitly change scrollbar visibility, or does that happen automatically as a side effect of changing its parameters? Maybe we should do a resize after changing scrollbar metrics then?

I have to leave this to someone compiling on windows, but I think I've narrowed it down quite a bit  ;-)

phkahler, 2023-12-14:
@ruevs Does this help?

ruevs, 2023-12-14:
I'm certain it does, but I'm swamped with other stuff... will try to look during the weekend.

phkahler, 2025-01-28:
@ruevs  If I put a bounty on this one would that help?  ;-)

ruevs, 2025-01-28:
> [@ruevs](https://github.com/ruevs) If I put a bounty on this one would that help? ;-)

No :-) I'll just have to force myself to debug it...

phkahler, 2025-02-17:
Since [WM_SIZE](https://github.com/solvespace/solvespace/blob/f09bbb96168120dd7a91802d5a0d246312df35dd/src/platform/guiwin.cpp#L791) does very little and resizing the window vertically "fixes" the problem until the scroll bar visibility changes, might a fix be to copy that one line into [SetScrollbarVisible](https://github.com/solvespace/solvespace/blob/f09bbb96168120dd7a91802d5a0d246312df35dd/src/platform/guiwin.cpp#L1352) ?

I have no way to develop on Windows, so I can't even test a one-liner like this.

ruevs, 2025-02-17:
I'll try.

ruevs, 2025-02-19:
Changing to:
```c++
    void SetScrollbarVisible(bool visible) override {
        scrollbarVisible = visible;
        sscheck(ShowScrollBar(hWindow, SB_VERT, visible));
        SS.TW.window->Invalidate();
    }
``` 
does not help.

phkahler, 2025-02-27:
Tried to reproduce this bug on Linux running the Windows version of Solvespace on Wine. The problem is does not seem to happen...

phkahler, 2025-07-29:
Just want to share a thought. Is it possible this is a bug in Angle? Seems unlikely, but our version of Angle is very very old.

ruevs, 2025-08-11:
> Just want to share a thought. Is it possible this is a bug in Angle? Seems unlikely, but our version of Angle is very very old.

It is not. The OpenGL 1 version behaves the same.

phkahler, 2025-10-15:
I had another idea a while back. Since it seems to come and go based on the visibility of the scroll bar... Could we just always leave the scroll bar visible on Windows? It'd be a hack, but if it keeps the text clear and the click area aligned it'd be better than today.