solvespace/solvespace #684: Bug: Windows "Asterisk" sound is played after adjusting dimension

sherief, 2020-08-26:
### System information

SolveSpace version: f952991

Operating system: Windows 10 x64

### Expected behavior

When I double-click a dimension, type a number then press enter I don't expect the Asterisk sound as specified in Control Panel (Classic) / Sound / Sounds to play

### Actual behavior

Every time I adjust the dimension and press enter, SolveSpace plays the Asterisk sound. It's counter to UI / UX expectation on the platform as this is not an exceptional event or indicative of an error.

### Additional information

N/A

sherief, 2020-08-26:
BTW, I wouldn't mind trying to fix this myself if you can provide pointers to where in the codebase I should start looking.

whitequark, 2020-08-26:
Take a look here: https://github.com/solvespace/solvespace/blob/master/src/platform/guiwin.cpp#L1051

sherief, 2020-08-26:
EditorWindowProc() handles WM_KEYDOWN, but Windows also sends a WM_CHAR message with scancode == 0x1c (enter key) and once that gets into DispatchMessage()  the unhandled sound plays. 

Check out https://github.com/sherief/solvespace/commit/0f6e47c808a955e81c29fe75d6e343494bfe778f - I'm not sure whether that's a fix or a hack, is it safe to consume WM_CHAR into the void here?

whitequark, 2020-08-26:
I took a quick look--seems that this is the correct way to do it, conceptually speaking.

whitequark, 2020-08-26:
Actually, I think that if you change `WM_KEYDOWN` to `WM_CHAR` here, you'll get the same effect in a slightly nicer way. https://github.com/sherief/solvespace/commit/0f6e47c808a955e81c29fe75d6e343494bfe778f#diff-99db0ce0039e3ba6b61036af47ff544fR1062

sherief, 2020-08-26:
OK, probably not gonna go through and read the CLA right now but I submitted a pull request and it's a trivial change that I just feel like throwing in the public domain - feel free to accept, reject, or re-implement it. Thanks for all the help!

PR #727 (2020-10-01): Fix issue 684 as whitequark suggested via WM_KEYDOWN->WMCHAR change.
resolves #684
closed 2020-10-01 by phkahler