solvespace/solvespace #689: Snap solvespace RGBA pixel error

joellapointe1717, 2020-08-31:
### System information

SolveSpace version:  3.0~d3da2c8b

Operating system:  Ubuntu 20.04

### Expected behavior

 Solvespace should just open and work!

### Actual behavior

 Solvespace has an error : No available configuration for the given RGBA pixel format. (see picture)
[image]

### Additional information

whitequark, 2020-09-01:
What GPU do you have?

joellapointe1717, 2020-09-01:
Intel hd4000 integrated on i7. Lenovo T430s

Téléchargez Outlook pour iOS<https://aka.ms/o0ukef>

whitequark, 2020-09-01:
Can you post the output of `glxinfo`?

joellapointe1717, 2020-09-01:
Sure, I’ll do it tomorrow morning because it is 3:12 AM un Montreal, Canada. Is it the complete command fromage terminal? (I’m not a programmer.)

whitequark, 2020-09-01:
Yes, it is the complete command.

joellapointe1717, 2020-09-01:
Here it is :
`# glxinfo
name of display: :0.0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_no_error, 
    GLX_ARB_create_context_profile, GLX_ARB_create_context_robustness, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_libglvnd, GLX_EXT_no_config_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, GLX_OML_swap_method, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_buffer_age, GLX_EXT_create_context_es2_profile, 
    GLX_EXT_create_context_es_profile, GLX_EXT_fbconfig_packed_float, 
    GLX_EXT_framebuffer_sRGB, GLX_EXT_import_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_no_error, 
    GLX_ARB_create_context_profile, GLX_ARB_create_context_robustness, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_query_renderer, GLX_MESA_swap_control, GLX_OML_swap_method, 
    GLX_OML_sync_control, GLX_SGIS_multisample, GLX_SGIX_fbconfig, 
    GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel Open Source Technology Center (0x8086)
    Device: Mesa DRI Intel(R) HD Graphics 4000 (IVB GT2) (0x166)
    Version: 20.0.8
    Accelerated: yes
    Video memory: 1536MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 4.2
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) HD Graphics 4000 (IVB GT2)
OpenGL core profile version string: 4.2 (Core Profile) Mesa 20.0.8
OpenGL core profile shading language version string: 4.20
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
    GL_3DFX_texture_compression_FXT1, GL_AMD_conservative_depth, 
[... 430 lines cut ...]
0x107 24 dc  0  32  0 r  y .   8  8  8  8 .  .  0  0  0  0  0  0  0  8 1 None
0x108 24 dc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  4 1 None
0x109 24 dc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  8 1 None
0x10a 24 dc  0  24  0 r  y .   8  8  8  0 .  .  0  0  0  0  0  0  0  4 1 None
0x10b 24 dc  0  24  0 r  y .   8  8  8  0 .  .  0  0  0  0  0  0  0  8 1 None
0x10c 24 dc  0  24  0 r  y .   8  8  8  0 .  .  0 24  8  0  0  0  0  4 1 None
0x10d 24 dc  0  24  0 r  y .   8  8  8  0 .  .  0 24  8  0  0  0  0  8 1 None
0x10e 24 dc  0  32  0 r  y .   8  8  8  8 .  s  0  0  0  0  0  0  0  4 1 None
0x10f 24 dc  0  32  0 r  y .   8  8  8  8 .  s  0  0  0  0  0  0  0  8 1 None
0x110 24 dc  0  32  0 r  y .   8  8  8  8 .  s  0 24  8  0  0  0  0  4 1 None
0x111 24 dc  0  32  0 r  y .   8  8  8  8 .  s  0 24  8  0  0  0  0  8 1 None
0x112 24 dc  0  24  0 r  y .   8  8  8  0 .  s  0  0  0  0  0  0  0  4 1 None
0x113 24 dc  0  24  0 r  y .   8  8  8  0 .  s  0  0  0  0  0  0  0  8 1 None
0x114 24 dc  0  24  0 r  y .   8  8  8  0 .  s  0 24  8  0  0  0  0  4 1 None
0x115 24 dc  0  24  0 r  y .   8  8  8  0 .  s  0 24  8  0  0  0  0  8 1 None
0x116 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
0x117 32 tc  0  32  0 r  . .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
0x118 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x119 32 tc  0  32  0 r  . .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x11a 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x11b 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x11c 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
0x11d 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
0x11e 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x11f 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None`

whitequark, 2020-09-01:
@ppd can you take a look? seems like a Snap-related bug

whitequark, 2020-09-01:
> Well, the snap/platform ships its own Mesa based on the Ubuntu 18.04 distribution, so I'm not sure how conclusive that glxinfo is in relation to the snap.

I was worried about Intel HD4000 being too old to support OpenGL 3.2 (which IIRC is the version we require). But no, that seems fine, though it does seem that it used to be a problem in the past.

gtxi8f4, 2020-09-17:
Hi,
It is not only a problem with the older Intel HD4000. My notebook has a UHD Graphics 620 installed and I get the same error

ppd, 2020-09-17:
What host system do you have? Do you have a dual GPU setup?

gtxi8f4, 2020-09-17:
no, i have a notebook with a Core i7-8565U and the built-in GPU. No other one built in

ppd, 2020-09-17:
And your OS is also Ubuntu 20.04?

gtxi8f4, 2020-09-17:
Ubuntu Budgie 18.04

ppd, 2020-09-17:
Wayland or X11? 

@joellapointe1717 The same question: is it an X11 session (I believe that's the default), or is it a Wayland session?

gtxi8f4, 2020-09-17:
It is an X11 session (default, like you said)

gtxi8f4, 2020-09-17:
can I somehow create a debug log?

gtxi8f4, 2020-09-17:
when i run the snap from the command line i get this output:

/snap/bin$ ./solvespace 
Gtk-Message: 15:20:00.103: Failed to load module "canberra-gtk-module"
Gtk-Message: 15:20:00.104: Failed to load module "canberra-gtk-module"
SolveSpace!

Missing (absent) translation for '&Line Styles...'
Missing (absent) translation for '&View Projection...'
Missing (absent) translation for 'Con&figuration...'
Missing (absent) translation for '&Helix'
Missing (absent) translation for 'Re&volve'
Missing (absent) translation for 'Show &Underconstrained Points'

(solvespace:11902): glibmm-CRITICAL **: 15:20:00.162: 
unhandled exception (type Glib::Error) in signal handler:
domain: gdk-gl-error-quark
code  : 1
what  : No available configurations for the given RGBA pixel format

(solvespace:11902): glibmm-CRITICAL **: 15:20:00.168: 
unhandled exception (type Glib::Error) in signal handler:
domain: gdk-gl-error-quark
code  : 1
what  : No available configurations for the given RGBA pixel format

ppd, 2020-09-17:
Maybe try: `snap run --strace solvespace &> solvespace.log` and attach here.

gtxi8f4, 2020-09-17:
[solvespace.log](https://github.com/solvespace/solvespace/files/5239518/solvespace.log)

gtxi8f4, 2020-09-17:
[solvespace.log](https://github.com/solvespace/solvespace/files/5239610/solvespace.log)

Christoph-D, 2020-09-17:
I have the same issue, also on Ubuntu 18.04.
GPU: Intel(R) HD Graphics 520
CPU: i5-6200U

What I did:

```shell
$ sudo snap install --edge solvespace
$ solvespace
[...]
(solvespace:27195): glibmm-CRITICAL **: 17:53:02.604: 
unhandled exception (type Glib::Error) in signal handler:
domain: gdk-gl-error-quark
code  : 1
what  : No available configurations for the given RGBA pixel format

(solvespace:27195): glibmm-CRITICAL **: 17:53:02.611: 
unhandled exception (type Glib::Error) in signal handler:
domain: gdk-gl-error-quark
code  : 1
what  : No available configurations for the given RGBA pixel format
```

Christoph-D, 2020-09-17:
solvespace works fine on the same system after I followed the "Build from source" instructions (built at 094eff7). It looks like only the snap package shows the error.

ppd, 2020-09-17:
There's something funky going on. I'll publish a snap with mesa-utils included for debugging. We must be doing something wrong.

ppd, 2020-09-18:
[solvespace_3.0~79457971_amd64.snap.zip](https://github.com/solvespace/solvespace/files/5244486/solvespace_3.0.79457971_amd64.snap.zip)

That's current master with glxinfo added on top (see https://github.com/ppd/solvespace/tree/glxinfo). Glxinfo is run exactly with the environment that the main solvespace app uses.

Install like so

```shell
wget https://github.com/solvespace/solvespace/files/5244486/solvespace_3.0.79457971_amd64.snap.zip
unzip solvespace_3.0.79457971_amd64.snap.zip
snap install --dangerous ./solvespace_3.0.79457971_amd64.snap
```

Run

```shell
solvespace.glxinfo
```

gtxi8f4, 2020-09-18:
um, if you get only the output of glxinfo with solvesapce.glxinfo, here is the log
[solvespace.glxinfo.log](https://github.com/solvespace/solvespace/files/5245194/solvespace.glxinfo.log)

gtxi8f4, 2020-09-23:
Hi,
is there already something new to report?

ppd, 2020-09-23:
> is there already something new to report?

I'm still investigating, but at this time, I'm a bit swamped with other stuff. Your glxinfo looks rather fine though. Maybe there's some strange corner case with GDK/Gtk not picking up the correct config when initializing. 

Could you do `sudo /usr/lib/snapd/snap-discard-ns solvespace` and then try to run it again?
And I'd like to see `snap connections solvespace`, too.

gtxi8f4, 2020-09-23:
should I run your solvespace.glxinfo or the original snap?

gtxi8f4, 2020-09-23:
after snap-discard-ns ... with original solvespace

snap connections solvespace
Benutzeroberfläche        Plug                        Slot                             Notizen
content[gnome-3-34-1804]  solvespace:gnome-3-34-1804  gnome-3-34-1804:gnome-3-34-1804  -
content[gtk-3-themes]     solvespace:gtk-3-themes     gtk-common-themes:gtk-3-themes   -
content[icon-themes]      solvespace:icon-themes      gtk-common-themes:icon-themes    -
content[sound-themes]     solvespace:sound-themes     gtk-common-themes:sound-themes   -
desktop                   solvespace:desktop          :desktop                         -
desktop-legacy            solvespace:desktop-legacy   :desktop-legacy                  -
gsettings                 solvespace:gsettings        :gsettings                       -
home                      solvespace:home             :home                            -
network                   solvespace:network          :network                         -
opengl                    solvespace:opengl           :opengl                          -
removable-media           solvespace:removable-media  -                                -
unity7                    solvespace:unity7           :unity7                          -
wayland                   solvespace:wayland          :wayland                         -
x11                       solvespace:x11              :x11                             -

ppd, 2020-09-24:
It's failing here: https://gitlab.gnome.org/GNOME/gtk/-/blob/master/gdk/x11/gdkglcontext-x11.c#L1159

I'm looking for a simple reproducer to check whether we're doing something wrong in solvespace's snap.

Therefore, I've adapted Emmanuele Bassi's glarea-example to run as a snap: https://github.com/ppd/glarea-example/tree/snap
At this time, it's not published in the store because of the dbus slot. 

You can get the snap right now at https://github.com/ppd/glarea-example/releases/tag/snap1

And install like so: `snap install --dangerous glarea-example_22f1de8_amd64.snap`.

Running `glarea-example` should exercise the same codepath. Maybe we'll gain a little more insight that way.

ppd, 2020-09-25:
Another option: Run solvespace with GtkInspector: ` GTK_DEBUG=interactive solvespace`. Info about the display config and the GL environment should be available under the "General" tab.

gtxi8f4, 2020-09-29:
HI ppd,
i have installed glarea. It comes the same error
[image]

ppd, 2020-09-29:
@gtxi8f4 Thank you very much. Could you do the  `GTK_DEBUG` stuff and the `snappy-debug` thing for me too, please?

gtxi8f4, 2020-09-29:
[image]

gtxi8f4, 2020-09-29:
and finally from /var/log/syslog with snappy-debug.security scanlog :
snappy-debug[1536]: 1601366464.4726288: Starting scanlog with --follow
solvespace[1539]: Failed to load module "canberra-gtk-module"
solvespace[1539]: Failed to load module "canberra-gtk-module"
budgie-panel.desktop[2036]: SolveSpace!
solvespace[1539]: #012unhandled exception (type Glib::Error) in signal handler:#012domain: gdk-gl-error-quark#012code  : 1#012what  : No available configurations for the given RGBA pixel format
budgie-wm.desktop[2026]: Window manager warning: Invalid WM_TRANSIENT_FOR window 0x5600016 specified for 0x5600003 (Property B).
solvespace[1539]: #012unhandled exception (type Glib::Error) in signal handler:#012domain: gdk-gl-error-quark#012code  : 1#012what  : No available configurations for the given RGBA pixel format
budgie-wm.desktop[2026]: Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x5600016 ((new sketc)
budgie-panel[2036]: button_wrapper_set_orient: assertion 'self != NULL' failed
budgie-panel[2036]: gtk_revealer_set_reveal_child: assertion 'GTK_IS_REVEALER (revealer)' failed
budgie-panel[2036]: button_wrapper_set_orient: assertion 'self != NULL' failed
budgie-panel[2036]: gtk_revealer_set_reveal_child: assertion 'GTK_IS_REVEALER (revealer)' failed
budgie-panel[2036]: button_wrapper_set_orient: assertion 'self != NULL' failed
budgie-panel[2036]: gtk_revealer_set_reveal_child: assertion 'GTK_IS_REVEALER (revealer)' failed
tuxbook budgie-panel[2036]: button_wrapper_set_orient: assertion 'self != NULL' failed
budgie-panel[2036]: gtk_revealer_set_reveal_child: assertion 'GTK_IS_REVEALER (revealer)' failed

 sudo journalctl --output=short --follow --all | sudo snappy-debug
kernel.printk_ratelimit = 0

ppd, 2020-09-29:
I think that's not quite right. You have to start `snappy-debug` in one terminal, then launch solvespace from wherever you want.
It should look like this:

```shell
snappy-debug.security scanlog
INFO: Following '/var/log/syslog'. If have dropped messages, use:
INFO: $ sudo journalctl --output=short --follow --all | sudo snappy-debug
= AppArmor =
Time: Sep 29 10:11:52
Log: apparmor="DENIED" operation="sendmsg" profile="snap.solvespace.solvespace" pid=18622 comm="solvespace" family="unix" sock_type="dgram" protocol=0 requested_mask="send" denied_mask="send" addr=none peer_addr="@7661722F72756E2F2F6E76696469612D786472697665722D303662656636383100000000000000000000000000000000000000000000000000000000000000" peer="unconfined"

= AppArmor =
Time: Sep 29 10:11:52
Log: apparmor="DENIED" operation="open" profile="snap.solvespace.solvespace" name="/var/lib/snapd/hostfs/usr/lib/x86_64-linux-gnu/libOpenGL.so.0.0.0" pid=18622 comm="solvespace" requested_mask="r" denied_mask="r" fsuid=1095800001 ouid=0
File: /var/lib/snapd/hostfs/usr/lib/x86_64-linux-gnu/libOpenGL.so.0.0.0 (read)
Suggestions:
* adjust program to read necessary files from $SNAP, $SNAP_DATA, $SNAP_COMMON, $SNAP_USER_DATA or $SNAP_USER_COMMON
* adjust snap to use snap layouts (https://forum.snapcraft.io/t/snap-layouts/7207)
```

gtxi8f4, 2020-09-29:
I have started it in the terminal:
snappy-debug.security scanlog 
INFO: Following '/var/log/syslog'. If have dropped messages, use:
INFO: $ sudo journalctl --output=short --follow --all | sudo snappy-debug

No further output.
Then I started solvespace. But even then no output was generated. Then I closed solvespace and stopped the debug with ctrl-C

What I sent was on the syslog.

but i have started the "original" solvespace from the snap. Not the one provided by you. But that doesn't make any difference, does it?

And thanks for the hint with the code :-)

ppd, 2020-09-29:
It makes no difference. So you have no denials, and GTK is rather happy with your GL environment. It just doesn't seem to be able to find a valid pixel format. 

The GTK build in snapcraft's gnome-3-34 extension is not debug enabled, and I can't figure out how to run an X11 snap program with GDB atm. I'm not really sure how to debug this without it turning into a large effort.

gtxi8f4, 2020-09-29:
what is the difference between the last running snap version and this one? Apparently there were some major changes?

ppd, 2020-09-29:
Do you have a last running revision? That'd be interesting. I feel that the last relevant snap change wrt GTK would be this one: https://github.com/solvespace/solvespace/commit/f5415b3fe64ca1288736ebe1471ae1794469b168#diff-c33a969de50ed245e7d91f6b05968401

gtxi8f4, 2020-09-29:
well, not anymore :-)

ppd, 2020-09-29:
Well, here's 13b461d, the last commit before gnome-3-34:
[solvespace_3.0~13b461d7_amd64.snap.zip](https://github.com/solvespace/solvespace/files/5298074/solvespace_3.0.13b461d7_amd64.snap.zip)

and here's f5415b3, the first commit with gnome-3-34:
[solvespace_3.0~f5415b3f_amd64.snap.zip](https://github.com/solvespace/solvespace/files/5298078/solvespace_3.0.f5415b3f_amd64.snap.zip)

gtxi8f4, 2020-09-29:
Unfortunately neither of them work. But I know that I had a working version 3 of solvespace via snap

phkahler, 2020-10-19:
Are you not running your desktop in 32bit color? Try typing the following:

```
xwininfo -root | grep Depth
```

joellapointe1717, 2020-10-20:
Here it is :

```
jmasque@lmde4:~$ xwininfo -root | grep Depth
  Depth: 24
jmasque@lmde4:~$
``` 
Thank you, I see that people are very motivated to rule out the problem...

gtxi8f4, 2020-10-20:
xwininfo -root|grep Depth
  Depth: 24

ppd, 2020-10-20:
@joellapointe1717  @gtxi8f4  I'm quite amazed to discover that the platform ships a GTK build with debug enabled. So you could try to see if GDK has some nicer debug messages for us:

`solvespace --gdk-debug all &> solvespace-gdk-debug.log`

gtxi8f4, 2020-10-20:
What did you do? I just installed solvespace again via snap to call it with gdk-debug. But I don't need to run the debug anymore, because it runs again without problems :-)
The logfile is of no use to you now, is it?
The installed version from the snap is: 3.0~3af8127e

ppd, 2020-10-20:
> What did you do? 

Nothing. Literally nothing.

I wonder whether I could be related to state, i.e. something that's stored in `$SNAP_USER_DATA`. Did you ever fully remove the snap and all its data before re-installing when you had the problem?

If it has just started magically working for all users, it could  also point to an (implicit) fix in a recent update of `snapd` or `gnome-3-34-1804`.

gtxi8f4, 2020-10-20:
I had deleted solvespace with purge. But that did not help so far. Maybe it was because of a Linux-Update? There was something coming in several times during the days. If no explanation helps anymore, magic is always a possibility ;-)

gtxi8f4, 2020-10-20:
Thank you for your support and your time

ppd, 2020-10-26:
@joellapointe1717 I'd appreciate it very much if you could try the following once more and give us feedback:
1. I'd like to see the output of `snap version`.
1. A fresh install: `snap remove --purge solvespace` and `snap install --edge solvespace`. Try if that solves it already.
2. `solvespace --gdk-debug all &> solvespace-gdk-debug.log` ← that logfile would be interesting if it's still not working

joellapointe1717, 2020-10-29:
My Lenovo T430s decided to not power up anymore (silent key as they say in ham radio jargon). So I’ll get access to a computer in the next few days. Sorry for the delay.

ruevs, 2021-04-07:
@joellapointe1717 can you still reproduce this problem with recent builds?

tomjnixon, 2022-01-17:
I found this issue while trying to get the solvespace build from nixpkgs running on arch with intel graphics (do not recommend).

The most helpful debug output i got was through running with `LD_DEBUG=files,libs` -- the opengl driver is dynamically loaded, and it seems like being in a snap might get in the way of that, just as nixos' rpath fiddling does.

This suggests that the driver needs to be included in the snap; it looks like it currently isn't, but that doesn't really explain why it works for anyone: https://forum.snapcraft.io/t/adding-opengl-gpu-support-to-a-snap/6273

Apologies if that isn't useful or relevant info...

ppd, 2022-01-18:
@tomjnixon That's great info; thank you for contributing. The snap itself does not include the drivers, that's true. However, the connected platform snap does (in the Intel case), or it gets mounted from the host into the snap namespace (in the Nvidia case). 

I suspect this bug has been fixed by our recent move to the core20 base in 34efb6de77edbb4ae2b4d9252124240987895bbf.

Quedale, 2022-10-05:
I'm not directly affected by this bug, but I did encounter this EXACT same error on my own custom made application.... using gstreamer library. 
The error is displayed by the GtkGstGlWidget.
I attempted to convert a deprecated use of GtkDrawingArea.

I just thought I would help people look in the right direction.
I'm most likely getting this error because I'm working off WSL 22.04, but this should help narrow down the root cause.

Hint: Somewhere along gstreamer and its dependencies by my understanding.

I'll share my findings if I manage to resolve my problem.

[image]

wolfpackmars2, 2023-03-05:
I receive this error off and on. I am running Linux Mint, using the current stable flatpak version of Solvespace. Unfortunately, I haven't figured out the cause or the definitive fix. I had just attempted to run Solvespace which wouldn't run. I had previously upgraded my OS (Linux Mint 21.0 > 21.1) and installed what I thought were all the available updates, and tried to run solvespace again which still didn't work.

I noticed that Mint was notifying me of new updates. I installed them and then tried running solvespace again (not expecting it to work) but it actually worked and loaded fine.

This photo shows the updates that were installed.

[image]

When I previously had this problem, I may have fixed it in a similar way.

Something to note - the nvidia driver I am running is 515.86.01, and the update history shows that I just installed nvidia 515.86.01 "flatpak runtime". I'm not very familiar with the details of how snap/flatpak works, but in my case, I am leaning towards a problem with graphics drivers being mismatched between the OS and the flatpak "runtime". Its possible that some have updated their graphics drivers, but a flatpak runtime isn't available yet for the new driver version.

I also happened to take and save a glxinfo before and after the update (system glxinfo) but it doesn't show any differences from before and after. It's available if wanted.

I hope this helps.

MrFaul, 2023-06-19:
In order to prevent this prehistoric laptop with a GeForce GT 330M from burning down while watching YT vids in the summer heat I thought it would be a good idea to install the Nvidia driver to get the HW acceleration/decoder running.

Now getting the latest driver would be easy... if Nvidia hadn't dropped the support for the relic.
No biggie I thought, get the legacy driver... what a bummer doesn't support the kernel I'm on.
Luckily for me someone patched the driver ([nvidia-legacy multi kernel repo](https://github.com/MeowIce/nvidia-legacy)) now I could watch YT without heat death yay.
(Only took four hours, because reasons...)

Guess now what greeted me today when I started up solvespace to design a small adapter plate.
Snap gives the "RGBA pixel error" and the flatpak version just freezes and doesn't render at all.
Flatpak loaded the correct legacy driver "runtime" but that could be very well incompatible with the kernel, I dunno.

Is there a possible fall back to software rendering?
So you get at least something?
Because I really don't want to go back to nouveau... this potato is then only 480p... :sweat_smile: