solvespace/solvespace #696: Coincident edges show as "Naked Edges"

phkahler, 2020-09-11:
### System information

SolveSpace version: 3.0~d8f5a8da

Operating system: Fedora 32

### Expected behavior

Make a sketch with 2 rectangles with one point coincident. Extrude. Analyze->Show Naked Edges

There should be no naked edges.

### Actual behavior

The coincident edges are shown red and there are 4 naked edges found.

### Additional information

You get similar behaviour even if the 2 extrusions are from different sketches as shown here
[image]

This is similar to the second problem reported in issue #303. It is also loosely related to issue #171.

phkahler, 2020-09-12:
My recent PR to fix this causes the following model to show zero naked edges at default CT settings. I never figured out why this sketch has any bad edges, or if it's actually misreporting them. Either way it doesn't make sense that the fix for this issue would affect it.
[gear90.zip](https://github.com/solvespace/solvespace/files/5213303/gear90.zip)

phkahler, 2020-09-13:
The gear90.slvs is still having issues, so I don't think this fix is hiding anything. Everything else I've looked at seems fine too.

closed 2020-09-14 by phkahler

ruevs, 2020-10-15:
@phkahler tag this "mesh" please.

commit e74185b6 (2020-09-14): Fix #696. Account for multiple coincident edges when looking for naked edges.