solvespace/solvespace #699: Dragging in assembly doesn't work

Christoph-D, 2020-09-17:
### System information

SolveSpace version: Built from source at 094eff7.
Operating system: Ubuntu 18.04

### Expected behavior

I followed the tutorial on http://solvespace.com/box.pl. I expected the following behavior after importing `base.slvs`:

> **We can grab any point with the mouse, and left-drag; the entire part will move along with us**, within the plane of the screen.
> We also can grab a point with the mouse, and Shift+left-drag. This rotates the part into and out of the plane of the screen, **about that point**.

### Actual behavior

I can click on only some points of the part, and no matter which point I drag, and no matter if I press shift or ctrl, the part rotates about its origin.

I found no way to drag the part or to rotate about anything else than the part's origin.

### Additional information

I used the file `base.slvs` from the zip file provided on http://solvespace.com/box.pl.

Christoph-D, 2020-09-17:
git bisect claims that f9529916c4913875af2408c5670815d10b2a50ae is the culprit. THe commit message also indicates that this commit changed drag behavior:

```
Fix issue161 and 174. Use the same logic for selction and dragging, but ignore entites that are inherently undraggable.
```

Christoph-D, 2020-09-17:
@phkahler as the author of f9529916c4913875af2408c5670815d10b2a50ae.

Christoph-D, 2020-09-17:
f9529916c4913875af2408c5670815d10b2a50ae introduced a new function `IsDraggable`. This function returns `false` for `Entity::Type::POINT_N_ROT_TRANS`. This entity type appears to be used only in `Group::Type::LINKED` groups, because linking is the only place that calls `CopyEntity(..., CopyAs::N_ROT_TRANS)`.

Maybe `Entity::Type::POINT_N_ROT_TRANS` should always be considered draggable in `IsDraggable`? This fixes the issue for me but I'm not sure if this would be a correct fix.

phkahler, 2020-09-17:
@Christoph-D thanks for tagging me. Incomming pull request 700 should fix it without breaking things.

Oh, your comment just updated. Yes, that is the fix.

closed 2020-09-18 by phkahler