solvespace/solvespace #712: Can not use Extrude

jatmn, 2020-09-21:
### System information

SolveSpace version: 3.0~a7f3b7e3

Operating system: Windows 10 Pro 2004 19041.508

### Expected behavior

Sketched a square "Sketch Rectangle" tool
New Group > Extrude 
Extrude square? Prompt for Values to drive Extrude? I don't know first time I have ever used the software.

### Actual behavior

Sketched a square "Sketch Rectangle" tool
New Group > Extrude 
Completely froze up the application requiring me to force endtask from task manager

### Additional information

There is no save file, I literally sketched a retangle on a new sketch.
See video at this tweet of the process:
 https://twitter.com/THEJATMN/status/1308108020665126912

phkahler, 2020-09-21:
> SolveSpace version: 3.0~a7f3b7e3

@jatmn  Where did you get that from?

jatmn, 2020-09-21:
> > SolveSpace version: 3.0~a7f3b7e3
> 
> @jatmn Where did you get that from?

https://github.com/Symbian9/SolveSpace-Daily-Engineering#solvespace-3x-slvs

But I was directly linked to report it here?

phkahler, 2020-09-21:
@jatmn that is unfortunate. @Symbian9  is linking directly to the CI builds, which are NOT nightly builds, are not official releases, and may be broken. I searched the history for the particular version you reported a problem with but could not find it. My guess would be that it was temporarily available but didn't actually work? That doesn't make sense either because I'm not aware of any recent problems of that magnitude, though I can't test on Windows myself.

@Symbian9 Please don't link directly to those builds, that's not what they are for.

nabijaczleweli, 2020-09-22:
3.0~a7f3b7e3 is probably https://github.com/solvespace/solvespace/commit/a7f3b7e3e250c98c620fbf9e0aa4120f94d05589 which is a merge of 0ba3bb13d70d3cfe72beea70d6d1e6d50623d785, which is currently the head of #711.

phkahler, 2020-09-22:
@nabijaczleweli  I applied that as a patch locally (git apply) and all I got was:
warning: 1 line adds whitespace errors.

The program build and runs fine though (Linux version). The CI tests all passed too. I see no reason not to merge it.

phkahler, 2020-09-22:
Closing, as not a relevant build version.

closed 2020-09-22 by phkahler