solvespace/solvespace #714: Stepping down as a maintainer

whitequark, 2020-09-22:
Effective immediately I am stepping down as a maintainer of SolveSpace. This doesn't mean that I will stop contributing to it completely, but it does mean that I will no longer be responsible for the project for the time being. As I'm sure everyone already noticed, I have not been able to complete any significant amount of work for quite a while.

To clarify things a bit, I am doing this for two reasons:
  1. Maintaining a relatively popular OSS project with a legacy codebase is not an easy or straightforward task. When combined with my other work and OSS responsibilities as well as a recent series of unfortunate events in my life, this results in an enormous amount of stress and a major impact on my mental health. Regardless of my desire to continue I lack the capability to do so.
  2. Although the vast majority of the community that formed around SolveSpace has been excellent to work together with, a few people behave in a way that is unconstructive and directly leads to maintainer burnout. I have not encountered this level of entitlement and persistence when working on any other OSS, of which I maintain plenty, and I do not wish to encounter it ever again.

After this, I believe the project will still be in good hands. For example, @phkahler has been able to fix several longstanding issues I could not meaningfully approach. I wish the project and its other maintainers only the best and I hope I'll be able to contribute to it again at some point in the future.

ppd, 2020-09-23:
Although I did not contribute to SolveSpace's core itself, I still felt that you made a huge effort to review and scrutinize any of my changes while displaying commendable amounts of politeness and patience.

Almost nothing is meant to last forever, and you certainly don't need to light yourself on fire to keep others warm. I can empathize with your feeling of burning out and completely understand the need to focus on fewer responsibilities in the future.

Finally, thank you for your selfless leadership, hard work, and also your efforts in everything non-technical. I appreciate it very much.

vcaputo, 2020-09-28:
Thanks for your help here on github and on #solvespace irc over the years, it's hugely appreciated.

ruevs, 2020-10-01:
Thank you for being very demanding on code quality for contributions. It has made the "bleeding edge" master branch of SolveSpace so stable over the years, that I have almost never used 2.3! This is not an easy thing to achieve and I really appreciate your consistency in enforsing it.

whitequark, 2020-10-04:
After some recent public and private interactions with both current and past developers, I change my mind. I have no desire to touch SolveSpace ever again, and I have removed myself from the organization. Have fun.

closed 2020-10-04 by whitequark

jwesthues, 2020-10-16:
So I regret that I'm sufficiently removed from SolveSpace's development that I didn't become aware of this until just now.

Without @whitequark's work building and maintaining SolveSpace over the last six years, it would still be the research-grade codebase that I released back in 2013. Her efforts both on user-facing enhancements and to make that codebase accessible to contributions by others are the only reason why this project exists in a form beyond "frozen in time, with scattered forks". I greatly appreciate that work, and hope to work with you again (on something, if not SolveSpace) in future.

Evil-Spirit, 2020-10-19:
I need to mention, I was one who a reason of previous @whitequark message. But now we solved issues between us, and I also should say: It was a pleasure to work together, and if she decided to work on SolveSpace again, I will be happy to work together.