solvespace/solvespace #716: Set up crowdfunding stream?

traverseda, 2020-09-22:
I've been told that this project has developers working on it commercially, but presuming this is going in a more community-led direction it would probably be a good idea to figure out some way to actually *give this project money*. Might also help dealing with this " level of entitlement and persistence" if you could point those people towards the collection bucket.

It also gives people who are unhappy with the pace of development, but who don't have the time or the skills to contribute code, a way to contribute to the project in some small way, and makes the project more sustainable in general.

This issue is a dupe of #265 , but I figure it's been long enough that it's worth revisiting the issue.

phkahler, 2020-09-22:
@traversda  I'm not aware of anyone currently getting paid to work on SolveSpace, though as I understand it that has been the case at times in the past. I'm not sure who funded the work or who all got paid, but it was temporary. I believe at least one person has been allowed to put in some time "on the clock" but that's IMHO a job perk more than anything like funding.

As for myself, I've had the luxury of taking an extended time off work and have found some satisfaction working on SolveSpace in that time. I plan to return to the job world soon (currently looking) but I still want to implement a number of features after 3.0 is released and will likely continue at whatever pace I can. I don't think the project is dying, but we have lost an incredibly valuable contributor and leader. BTW she has a Patreon and deserves whatever contribution you might want to give as backpay for years of work here.

traverseda, 2020-09-22:
>BTW she has a Patreon and deserves whatever contribution you might want to give as backpay 

Nah, as per #265 that kind of thing isn't necessary. Still, setting up some funding streams going forward would probably be a good idea. I think there's a lot of people who wanted to support solvespace but found that it was pretty difficult to actually do that. Making it easy for people to contribute would be good, even if it's only $1 per month or whatever.

Evil-Spirit, 2020-10-01:
Oh.. Have you read an [artricle](https://thenewstack.io/how-solvespace-is-retooling-its-3d-cad-open-source-community/) about @whitequark? She have to pay me... Yes, for development, not for thing you just imagine...

phkahler, 2020-10-01:
@Evil-Spirit  Please keep whatever issues or disputes you have off github, no good can come from it. As someone once said, "Not my monkeys, not my circus."  Your comment does explain some of the tension that I noticed, but again it's not my problem and I don't want to know.

whitequark, 2020-10-01:
@phkahler I don't believe there is any issue or dispute, at least there isn't to my knowledge. If there is, @Evil-Spirit is yet to tell me about it.

traverseda, 2020-10-01:
I don't think there's any real disputes, I'm just trying to politely say I'm not going to start paying whitequark *now* when she was uninterested in setting up any sort of sustainable revenue stream for the project in the past.

I think @phkahler might be talking about the thread at #265 when they mention "tension"? Not a big concern either way though.

whitequark, 2020-10-01:
Oh, I've missed the start of the thread somehow.

> I've been told that this project has developers working on it commercially,

This hasn't been true for several years.

> Might also help dealing with the " level of entitlement and persistence" if you could point those people towards the collection bucket.

Back when I actively worked on the project, the amount of money I could have collected in donations for it was dwarfed by the amount of cash *from my own pocket* that I was spending on it (over a thousand USD per month), *and* I didn't have a particularly viable way of actually collecting donations. So you can see why I wasn't interested in it. Of course, it's not relevant anymore, that's just some historical context.

traverseda, 2020-10-01:
I mean that's a discussion we've already had, so there's no point in going over it again. I do think if you make it hard for people to support a project, and to support you, you shouldn't be surprised when you don't really get supported.

I'd encourage the new maintainer to make it easier for people to support them, it might be a waste of time but if you don't take that first step it's impossible to build up. The amount of money you can collect from donations will always start at basically nothing, but that at least includes the possibility of it building up over time.

I mean take a look at https://github.com/uraimo/awesome-software-patreons

Some of those people even make a reasonable wage from donations. You've got to start somewhere.

whitequark, 2020-10-01:
> I do think if you make it hard for people to support a project, and to support you, you shouldn't be surprised when you don't really get supported.

It is a mistake to think that the only, or even the primary, kind of support a maintainer needs is money. For me, it wasn't, and this discussion only reaffirms my decision, if anything.

traverseda, 2020-10-01:
Sure. Hopefully the new maintainer will have an easier time with it.

phkahler, 2020-10-01:
Sorry about that, I misunderstood Evil's comment and thought it indicated an unpaid debt. It seems I was mistaken.

> Sure. Hopefully the new maintainers will have an easier time with it.

@traverseda As far as I can tell there are no maintainers at the moment. I'm just trying to wrap up a 3.0 release by closing out issues that whitequark had tagged for it and I happen to have write access ;-)  Some of those will likely get bumped, but until I figure out how to make a proper release they'll just slowly get resolved. Maybe.

Along the way I might have some fun too.  @whitequark, I hope you like the IDF file linking feature.  ;-)

ruevs, 2020-10-01:
Way off-topic:
And I just learned that Whitequark is "she" after contributing to SolveSpace from time to time since 2015 :-)

whitequark, 2020-10-01:
> I hope you like the IDF file linking feature. ;-)

I've planned to add IDF linking for *years*. I'll check it out next time I'm doing PCB layout!

> For example I do it for fun and (in a maintainer's shoes) would not want to be paid, since that would make it a "have to" instead of "want to" proposition.

Yup. For me, the fact that I'm a maintainer already presents enough of an obligation. The logic in #265 was that once you put a monetary value on it, it ceases being something I do because I'd like to see the project succeed, and becomes something I do for $5/month... which is a much less interesting proposition.

Evil-Spirit, 2020-10-02:
> Please keep whatever issues or disputes you have off github

We have no any disputes and issues. I prefer to think @whitequark is my friend. Article is just a joke.

Evil-Spirit, 2020-10-02:
> If there is, @Evil-Spirit is yet to tell me about it.

Only one issue is what I don't know how to help you to feel you better. I'm worried about you.

Evil-Spirit, 2020-10-02:
> And I just learned that Whitequark is "she"

@whitequark I don't know it is better to say them or not... @whitequark is a new kind of gender. Quantum one! Schrödinger talked about cat, but he don't say about gender of the cat. So, now there should be quantum experiment called "Whitequark's gender" as like "Schrödinger's cat". You can think whitequark is man or woman, but only asking him can let wave function to collapse!

whitequark, 2020-10-02:
> Only one issue is what I don't know how to help you to feel you better. I'm worried about you.

You're not helping.

Evil-Spirit, 2020-10-02:
Sorry

Evil-Spirit, 2020-10-02:
I didn't know you are serious about this. Sorry.

pentacular, 2020-10-02:
@Evil-Spirit you're making me feel uncomfortable enough to stop lurking for a bit.

I hope it is not intended, but it feels like your behavior is verging on harrassment.

whitequark, 2020-10-02:
It is not "verging on harassment", it is harassment, and I regret ever contracting Aleksey to work on SolveSpace, or frankly working on it at all. I don't know what it is about mCAD packages that brings out all the assholes, but I'm done with them.

Evil-Spirit, 2020-10-02:
@whitequark
I am sorry for this, I thought it was just a joke (as I understand from Discord), but seems things is very serious. I can just make them to remove an article. I am sorry for my misunderstanding of situation. I never know how things is bad here.

Evil-Spirit, 2020-10-02:
That was just a bad joke, I can't ever say how bad fellings now and I am sorry again.

ruevs, 2020-10-02:
@whitequark I sincerely apologise for bringing up the subject, I should have thought twice.

As for Aleksey - I think/hope he did not have ill intentions. It probably boils down to cultural differences - being eastern European myself, but having significant "experience" with the USA/"west", I can say that many things that would be considered blatant harassment or violation of privacy in the "west" are, if not perfectly OK, at least completely acceptable in the "east".

In many Slavic languages there is no word "privacy" - it has to be explained verbosely and older people may still not understand...

Again, I apologise.

traverseda, 2020-10-02:
I'm just confused, I thought the article was quite flattering. Maybe I missed an edit or something, I mostly read these through email.

Anyway, I think everything relevant has been said. Closing this issue as it went off-topic. I've said everything I feel I need to.

closed 2020-10-02 by traverseda