solvespace/solvespace #719: New error message dialog is annoying.

Evil-Spirit, 2020-09-27:
### System information

0a061b6f9ea8ea10414e04a6a87c14c6b12ad04b

Operating system: Windows 7

### Expected behavior

Dialog should be just info dialog (as it been before), not the error with ugly sound and big red sign.

### Actual behavior

big red sign soundy annoying dialog
[image]

ruevs, 2020-11-26:
It is annoying, I totally agree. When I have some time I'll try to dig out how it was done before and "restore" it.

ruevs, 2021-01-08:
Well before it was done with a custom window. Now it is a standard message box. And (selectively) turning off the "beep" for them is... not supported...

https://stackoverflow.com/questions/1738980/how-do-i-disable-the-c-sharp-message-box-beep/42419494

phkahler, 2021-01-08:
@ruevs the comments in your links suggest that getting rid of the icon will make the sound go away.

It's been over 20 years since I wrote serious Windows code, but my recollection is that there was a parameter to indicate what kind of dialog box you got - the big red X warning being just one.

ruevs, 2021-01-09:
There is. But I'll have to experiment a bit.
It comes to this
https://github.com/solvespace/solvespace/blob/3e3ccdca8d6c66080078c46c274cbde645f43cf4/src/platform/guiwin.cpp#L1521

https://docs.microsoft.com/en-us/windows/win32/api/winuser/nf-winuser-messageboxindirectw

I'll have to enable all sounds on my dev. system (normally everything is disabled) and see... I think I'll try to put the SolveSpace icon in there.

ruevs, 2021-01-09:
Note to self
https://webcache.googleusercontent.com/search?q=cache:y18TNQDowgwJ:https://www.garybeene.com/code/gbsnippets_gbs_00332.htm+

closed 2021-01-09 by phkahler