solvespace/solvespace #720: When drawing segment, viewport is not redrawing while mouse movement

Evil-Spirit, 2020-09-27:
### System information

SolveSpace version: 0a061b6f9ea8ea10414e04a6a87c14c6b12ad04b
Operating system: Windows 7

### Expected behavior

Viewport should be redrawn every mouse move event

### Actual behavior

While movement, viewport isn't redrawn so we can't see the actual entity position

### Additional information

Click  line segment button then click first point and move mouse. Second segment point stays in the previous position, but it should follow mouse cursor.
It seems, actual only for segment, but need to be tested more

phkahler, 2020-10-15:
This issue needs confirmation, and then it should be traced to a specific commit because it would be a regression.

ruevs, 2020-10-16:
I can not reproduce it on d72eba8039c52f5e8d3ab36ec2d8a7598dfd578b Windows 10

phkahler, 2020-10-16:
> I can not reproduce it on [d72eba8](https://github.com/solvespace/solvespace/commit/d72eba8039c52f5e8d3ab36ec2d8a7598dfd578b) Windows 10

@Evil-Spirit  What version of Windows are you on? How to reproduce it?

phkahler, 2020-10-20:
Is it maybe a problem when holding down the mouse button while dragging?

kk6mrp, 2020-10-23:
I have one of the AppVeyor CI builds from before and drawing segments do fail to refresh immediately when moving the cursor fast, but when moving slow it refreshed.

ruevs, 2020-10-23:
Can you zip and upload this version here. Also what version does it show in Help | About ?

kk6mrp, 2020-10-23:
3.0~12b7c8be

[solvespace.zip](https://github.com/solvespace/solvespace/files/5431337/solvespace.zip)

Evil-Spirit, 2020-10-26:
@kk6mrp 
You have exactly the same issue.

Evil-Spirit, 2020-10-26:
@phkahler I use Windows 7

kk6mrp, 2020-10-26:
I can't reproduce the issue on the latest Edge release on Windows 10.

ruevs, 2020-10-26:
@kk6mrp yes, I can reproduce the problem with your version and it is awful. However the commit seems harmless to me 12b7c8be strange...
If you try with the current edge/master (https://github.com/solvespace/solvespace/releases) how does it work for you (it is not a debug build)?

phkahler, 2020-10-26:
@Evil-Spirit   Please try with a build from edge and report back: https://github.com/solvespace/solvespace/releases

ruevs, 2020-11-26:
The "new" SolveSpace is definitely slow compared to 2.x when dragging lines/rectangles.

I built a recent master after 3.0rc1 for Windows XP (x86, Release, OpenGL 1 renderer, LTO, 141_xp platform toolset, VS2019 19.27.29112 compiler).

On my ancient 2008 laptop with XP dragging a line looks just like the screen shot @Evil-Spirit posted above.

To rule out the renderer and XP I ran the above version as well as a "normal" build from the same source on a slow 2010 Windows 7 laptop. Both versions are slow compared to 2.3. If anything the GL3 one may be a bit slower (or I am imagining).

Debug builds show this much more noticeably.

On faster machines the difference between 3 and 2 becomes almost imperceptible (also on a faster XP machine).

phkahler, 2020-11-26:
This is when dragging a 2d sketch? Did you try without OMP? Are you comparing with the same CT settings? I dont think those should affect 2d sketching though.

ruevs, 2020-11-26:
Yes. Empty file, 2D sketch in the default first workplane, single line segment. Exactly like the above animation https://github.com/solvespace/solvespace/issues/720#issuecomment-715465479

No OMP.

I did not pay attention to the CT settings - at least three of the machines involved have seen SolveSpace before, so they may not be default - my mistake.
But as you said - I don't think they affect literally `Ctrl-N` (or run freshly), `S`, `click`, drag.

Rotating the view in 3D with 3d objects and many groups present works fine - visibly as fast as 2.x

Evil-Spirit, 2020-11-27:
@ruevs There is no something slow in dragging only one line. You can't see the difference. This is just a bug - probably, with logic of updating screen inside segment tool (since the other tools work fine)

robnee, 2021-01-20:
I also see this in 3.0 including 3.0 rc2.  It only affects sketching with the line or rectangle tool.   I don't know if this is related but I have a 4K monitor and run 125% scaling.  I will try resetting scaling to 100% to see if this matters.

robnee, 2021-01-20:
Nevermind.  It appears this is not related to HiDPI scaling.

If I drag the mouse very slowly while sketching the view updates normally.  When moving the mouse faster, at some point it's as if the mouse messages are coming in too fast and some are being ignored.  Then, If I move the mouse just slightly faster than that at a consistent speed, I can get the view to not update at all, indefinitely,  until I stop moving the mouse.  It's as if the event loop is trying to keep up by discarding mouse events if there is another mouse event later in the queue as a too-aggressive optimization.

There was a refactoring that broke 6 DOF mice which was thankfully just fixed.  Could this be another problem with dispatching events?

If there is anything I can test to assist I'm happy to try.

phkahler, 2021-01-20:
> If there is anything I can test to assist I'm happy to try.

@robnee  Thank you for the detailed description of the problem ;-)  I don't do Windows development, but this might enable me to see a possible cause by reading that area of the code. If you can C++, you are more than welcome to have a look too.

To any Windows devs, this issue is most likely in the one Windows specific file here:
https://github.com/solvespace/solvespace/blob/master/src/platform/guiwin.cpp

robnee, 2021-01-20:
I've been feeling my way through the code.  I wasn't able to find any obvious blames. Perhaps not mentioned yet is that this problem only appears when drawing new requests.  For instance if I finish a rectangle with right click and then immediately drag the corner again it resizes at normal draw speed.  If I had to guess it seems to have something to do with the constraint suggestions for horizontal and vertical.  When I drag a new rectangle the "H" constraint on the bottom of the rectangle as well as a tiny purple dot (at top of mouse pointer) seems to follow the mouse when dragging while the other 3 constraints,the 4 segments, and 4 datum points don't draw until the mouse stops moving.  Could this be a draw problem?  Perhaps the view is being invalidated too often while drawing constraint suggestions?

phkahler, 2021-01-21:
@robnee The H and V constraints for a rectangle are not suggestions, they are real from the start. You can disable automatic line constraints with a checkbox on the configuration screen though. One other thing that is different between creating and dragging a rectangle is related to hovering - it's looking for points to snap the corner to. If you drag near an existing point, it will highlight that point, then if you click it will constrain to the hovered-over point.

phkahler, 2021-01-21:
Could this just be mouse-move events getting priority over whatever triggers redrawing?

robnee, 2021-01-21:
> Could this just be mouse-move events getting priority over whatever triggers redrawing?

I think so according to this: 

https://stackoverflow.com/questions/22039413/moving-the-mouse-blocks-wm-timer-and-wm-paint

phkahler, 2021-01-23:
@robnee That sounds about right and there are several solutions suggested at that link.

If I had a windows machine... The idea of checking the queue for more mouse events and processing them all seems like a good idea, as does reading the position using GetCursorPos() instead of the mouse event data.

But first I like the suggestion:
1)  Store the mouse movement info in a structure.
2)  Write an OnIdle handler that checks for mouse move data and handles it there.

The reason they suggest that is that Idle has the lowest priority, so will not delay window repaints.

We already put all the mouse data into a struct (declared in line 862 of guiwin.cpp) and pass that to the platform agnostic  onMouseEvent() at line 960/961.

862:
```
                MouseEvent event = {};
```
960:
```
                if(!consumed && window->onMouseEvent) {
                    window->onMouseEvent(event);
                }
```
So make "consumed" and "event" memebers of WindowImplWin32.  Actually event will need to be renamed to "lastMouseEvent" or something so as not to conflict with other local variables named event. Move that check for consumed into the Idle handler and it might be good to go.

At least it's something to try.

robnee, 2021-01-23:
@phkahler I took a crack at this.  My Windows programming is (seriously) 30 years old. So ,a little rusty.  The problem I ran into is the OnIdle handler seems to be an MFC concept vs. a low-level Windows API thing.  It's possible to hand-craft an idle loop it seems but that would be a task for a real Windows dev.

I did come up with a rather ghastly hack to rate-limit the WM_MOUSEMOVE messages to no faster than one every 20ms.  This works.  Line segments and rectangles draw smoothly.  10ms on my machine was still a problem.  I'm not sure if this is really an appropriate solution though.

`    diff --git a/src/platform/guiwin.cpp b/src/platform/guiwin.cpp
    index d885195..0947db4 100644
    --- a/src/platform/guiwin.cpp
    +++ b/src/platform/guiwin.cpp
    @@ -513,6 +513,7 @@ public:
         HWND hTooltip = NULL;
         HWND hEditor  = NULL;
         WNDPROC editorWndProc = NULL;
    +    LONG lastMoveTime = 0;

     #if HAVE_OPENGL == 1
         HGLRC hGlRc = NULL;
    @@ -937,6 +938,14 @@ public:
                             event.type = MouseEvent::Type::LEAVE;
                             break;
                         case WM_MOUSEMOVE: {
    +                        // rate limit mouse move messages to no faster than one per 20ms
    +                        LONG mt = GetMessageTime();
    +                        if(mt - window->lastMoveTime > 20) {
    +                            window->lastMoveTime = mt;
    +                        } else {
    +                            consumed = true;
    +                        }
    +
                             event.type = MouseEvent::Type::MOTION;

                             if(wParam & MK_LBUTTON) {`

phkahler, 2021-01-23:
@ruevs  Could you have a look at this fix?  Maybe you have an improvement, but I'm kind of happy with something that resolves the issue.

ruevs, 2021-01-24:
Monday at the earliest.

ruevs, 2021-01-27:
I think the [PR](https://github.com/solvespace/solvespace/pull/908) just hides the real problem.

Only the initial dragging of a rectangle or a line when they are created is "slow". Resizing them later is "fast", which  proves that there is some bottleneck in the first case. So we should find and fix the root cause.

Slowing down the handling of WM_MOUSEMOVE to once every 20ms just hides the problem. On a (much) slower system it will still appear.

Just to check that there are no unnecessary WM_MOUSEMOVE events with unchanging coordinates I tried the code below. The only time they appear is when you `Alt-Tab` into the window.
```C++
diff --git a/src/platform/guiwin.cpp b/src/platform/guiwin.cpp
index d8851958..00eb2c24 100644
--- a/src/platform/guiwin.cpp
+++ b/src/platform/guiwin.cpp
@@ -937,6 +937,15 @@ public:
                         event.type = MouseEvent::Type::LEAVE;
                         break;
                     case WM_MOUSEMOVE: {
+                        static MouseEvent previousEvent = {};
+
+                        if((previousEvent.x == event.x) && (previousEvent.y == event.y)) {
+                            consumed = true;
+                            dbp("x: %f y: %f", event.x, event.y);
+                        } else {
+                            previousEvent = event;
+                        }
+
                         event.type = MouseEvent::Type::MOTION;

                         if(wParam & MK_LBUTTON) {
```

robnee, 2021-01-27:
@ruevs Thank you.  I'm frankly a little relieved to hear you say this.  After I messed up the posting of the diff above I did a PR just so my patch would be clear.  I worried that limiting paints to 50Hz might not even be enough for very slow machines while unnecessarily impacting fast machines that otherwise could keep up.  It's really just an experiment.

I could try looking at the event handling in mouse.cpp to see if I can generate some timing info for these steps in my environment to see if one is obviously the bottleneck (perhaps hit testing?).  I haven't tried gprof on a Windows executable to see if that can be useful.

Rob

```
case Pending::DRAGGING_NEW_LINE_POINT:
            if(!ctrlDown) {
                SS.GW.pending.hasSuggestion =
                    SS.GW.SuggestLineConstraint(SS.GW.pending.request, &SS.GW.pending.suggestion);
            } else {
                SS.GW.pending.hasSuggestion = false;
            }
            // fallthrough
        case Pending::DRAGGING_NEW_POINT:
            UpdateDraggedPoint(pending.point, x, y);
            HitTestMakeSelection(mp);
            SS.MarkGroupDirtyByEntity(pending.point);
            orig.mouse = mp;
            Invalidate();
            break;
```

robnee, 2021-01-27:
@ruevs I was experimenting with your change but I can't seem to get the dbp messages to display.  I did:

cmake .. -DCMAKE_BUILD_TYPE=Debug
make

and confirmed it does suppress the -DNDEBUG compiler option so I think DebugPrint should work.  I expected to see the stderr messages in the console I launched bin/solvespace.exe from.  I don't know if this is a problem with using mingw builds or if I'm looking in the wrong place.  Any suggestion as to what I am doing wrong?

Rob

ruevs, 2021-01-27:
The debug messages only show up if you have a debugger (e.g. in the "Output" window of VisualStudio) or if you apply this patch:
https://github.com/ruevs/solvespace/commit/3ef465984689ffb28edfcabfd69800d892d37933
https://github.com/ruevs/solvespace/commits/Win32DebugConsole
that explicitly creates a console for stderr. Otherwise Win32 GUI processes do not have a console (even when started from cmd.exe) and thus the messages go to the `dev/null` equivalent of Windows (whatever that is).

phkahler, 2021-01-27:
It is worth noting that a complex extrusion can be dragged with low frame rates and it still updates (apparently, I don't have Windows). There is at least one flag in mouse.cpp that seem related to weather a repaint is needed. The difference with this issue might be that for a new line segment or rectangle we're dragging with the mouse button UP.

robnee, 2021-01-27:
@ruevs Thanks! I will look at the patch.

@phkahler Right...  Also, I notice a similar problem with text requests.

robnee, 2021-01-27:
@ruevs/@phkahler  I did some more investigation.  

1) It does not appear that the problem is how long the call to onMouseEvent takes.  It takes less than 1ms on my machine.  

2) Mouse messages on my machine max out at a rate of about 60Hz (one every 16ms or so)  when moving the mouse around the window.  I believe my rate limiting patch works (at 20ms vs 10ms) only because it then causes some calls to onMouseEvent to be skipped.  

I believe there is something about calling onMouseEvent or subsequent calls to things like Invalidate or ClearPending that ensures that another WM_MOUSEMOVE message will be delivered before the next WM_PAINT thus starting the cycle over again.  skipping dispatching at least the occasional onMouseEvent breaks this cycle allowing the message queue to drain and some paint messages to be processed.  

Unfortunately, I'm not sure what causes this behavior.  I tried to play with havePainted flag but so far to no avail.

robnee, 2021-01-28:
@ruevs/@phkahler

It seems the problem is related to the timers on Windows.  WM_TIMER messages are like WM_PAINT in that they get deferred until there is no other messages to process.  There is a generateAllTimer that tries to set a timeout of zero, presumably to have a generateAll "run soon".  Windows doesn't call the timer callback if there are more mouse messages to process, so no "generation" happens until mouse events die down.  I see WM_PAINT messages but without a generateAll call there is nothing to paint.

I think SS is expecting more reliable timers and probably gets them on GTK and Mac but not Windows.

Rob

robnee, 2021-01-28:
After a little searching I came across an [blog post on WM_TIMER priority](https://devblogs.microsoft.com/oldnewthing/20191108-00/?p=103080) with a technique for boosting the priority of timer messages to keep them from starving in the message queue.  If involves synthesizing WM_TIMER messages in the message pump.  It seems to work.

```void RunGui() {
    MSG msg;
    // The return value of the following functions doesn't indicate success or failure.
    while(GetMessage(&msg, NULL, 0, 0)) {
        TranslateMessage(&msg);
        DispatchMessage(&msg);

        // this makes WM_TIMER a higher priority message
        MSG timer_msg;
        PeekMessage(&timer_msg, nullptr, WM_TIMER, WM_TIMER, PM_NOREMOVE);
    }
}```

phkahler, 2021-01-29:
I had a chance to run version 3 RC2 on a Windows 10 machine today and it worked fine. This issue isnt universal. I suppose that follows from iit being sensitive to timing?

robnee, 2021-01-29:
I agree that this does not seem to be universal and I can't explain that.  I have two win 10 machines one recent and both have the issue.  It seems to line up with discussions of the behavior of timer and paint messages on windows and problems with redraws or timeouts being starved.  That is to say it seems reasonable that my fix would work but it's really strange that only some setups seem to require it.

kk6mrp, 2021-01-29:
What I don't understand is why arcs/circles do not have the same issue. If it is a timing issue, wouldn't if affect all the drawing elements the same?

robnee, 2021-01-29:
@kk6mrp i may not be the best to explain this but what I see is the generateAll step is needed to prepare the geometry for drawing.  Without it nothing gets updated when painting lines, TrueType or rectangles even though paint messages are being received and processed.  Since generateAll is triggered by a timer and timers on Windows have the lowest priority (lower than the paint message even) the generateAll step is being skipped when the event loop gets busy.

Maybe circles and arcs don't require generateAll in order to update while drawing.  The arcs and  circles code path in mouse move processing is simpler than lines and rectangles.  There are no H and V suggestions.  Maybe something about that *requires* invalidating and regeneration in a way arc and circles don't.  I don't understand the common SS core quite well enough to say for sure.

I think this boils down to unpredictability (on Windows) of requesting that a generateAll be performed after a delay of 0ms which is what generateAllTimer tries to do.  It probably expects that it will happen immediately when the next message is processed but certainly before a paint message tries to draw the geometry.  But timers on Windows are guaranteed to behave this way and a 0ms delay can be stretched indefinitely if you can generate enough mouse messages to keep the timer event from being delivered.  SS in that case is repeatedly drawing the stale geometry.

robnee, 2021-01-29:
@kk6mrp Dragging a new arc or dragging a new circle does not call Invalidate() during its processing of mouse movement messages.  Dragging a new line, rectangle and ttf text do call Invalidate each mouse movement.  Calling Invalidate() presumably will require a generateAll before processing a paint message and redrawing.  

@phkahler What is the difference between Invalidate() and Invalidate(true)?  I.e. what are "persistents?"

robnee, 2021-01-29:
I think I might have finally gotten to the bottom of this.  Thanks @kk6mrp for convincing me to take another look.

robnee, 2021-01-29:
This may be the [commit](https://github.com/solvespace/solvespace/commit/f324477dd06c15022670a4486bf83be3abbf48b5) that created the problem.

closed 2021-01-31 by phkahler