solvespace/solvespace #738: All NURBS and mesh bugs and their test models in one place

ruevs, 2020-10-15:
This is a list of all SolveSpace issues related to [NURBS](https://github.com/solvespace/solvespace/labels/NURBS) or [mesh](https://github.com/solvespace/solvespace/labels/mesh) bugs.
Links to the `.slvs` models reproducing the problems and the fixes may be useful for regression testing.
If the "Fixed on" column is empty the issue is probably still open (or I did not update this list :-))

Source | Model | Author | Fixed on | Fixed in
------ | ----- | ------ | -------- | --------
https://github.com/solvespace/solvespace/issues/32 <br/> https://github.com/solvespace/solvespace/issues/540<br/>https://github.com/solvespace/solvespace/issues/683<br/>https://github.com/solvespace/solvespace/issues/683#issuecomment-683274604<br/>https://github.com/solvespace/solvespace/issues/732 | [failed.lathe.slvs.zip](https://github.com/solvespace/solvespace/files/408873/failed.lathe.slvs.zip)<br/>[adapter-plate.tar.gz](https://github.com/solvespace/solvespace/files/4054379/adapter-plate.tar.gz)<br/>[wrong.zip](https://github.com/solvespace/solvespace/files/5112393/wrong.zip)<br/>[wrong2.zip](https://github.com/solvespace/solvespace/files/5145107/wrong2.zip)<br/>[coincident4.zip](https://github.com/solvespace/solvespace/files/5357050/coincident4.zip) | @jockbert<br/>@baryluk<br/>@starsolaris<br/>@starsolaris<br/>@phkahler | 2020-10-09 | https://github.com/solvespace/solvespace/pull/733 https://github.com/solvespace/solvespace/commit/d72eba8039c52f5e8d3ab36ec2d8a7598dfd578b
https://github.com/solvespace/solvespace/issues/35#issuecomment-531606474 | [NURBS_issues.zip](https://github.com/solvespace/solvespace/files/3614377/NURBS_issues.zip) | @phkahler |  | 
https://github.com/solvespace/solvespace/issues/171 | [bug.zip](https://github.com/solvespace/solvespace/files/728328/bug.zip) | @whitequark | https://github.com/solvespace/solvespace/issues/652#issuecomment-696931793 | 
m https://github.com/solvespace/solvespace/issues/175 | [aaa-minus-weirdly-broken.slvs](https://gitlab.com/anelok/mexp/raw/016367ea70960daac4d8bb94a5740d84a3f5e900/alt13/aaa-minus-weirdly-broken.slvs) | @wpwrak |  | 
m https://github.com/solvespace/solvespace/issues/177 | [an-top.slvs](https://gitlab.com/anelok/mexp/raw/8951232e87fa6c7a8f2820436a2ae9c247dd26a6/alt13/an-top.slvs) | @wpwrak |  | 
m https://github.com/solvespace/solvespace/issues/259 | [case.zip](https://github.com/solvespace/solvespace/files/1016276/case.zip) | @jonas-schievink | 2017-05-22 | 
m https://github.com/solvespace/solvespace/issues/295 | 16 sided small circles |  |  | https://github.com/solvespace/solvespace/pull/626
https://github.com/solvespace/solvespace/issues/296 <br/> https://github.com/solvespace/solvespace/issues/693 | [table.zip](https://github.com/solvespace/solvespace/files/1341417/table.zip) <br/> [polygon_failures.zip](https://github.com/solvespace/solvespace/files/5185284/polygon_failures.zip) | @tgirod <br/> @phkahler | 2020-09-08 | https://github.com/solvespace/solvespace/pull/694 https://github.com/solvespace/solvespace/commit/0413c1b9262902e9d8a0b41d7bb56530b72fba9c https://github.com/solvespace/solvespace/commit/d8f5a8da32b48a8490bf509e2757249896e06659
https://github.com/solvespace/solvespace/issues/297#issuecomment-352584589 | [bug-repro.zip](https://github.com/solvespace/solvespace/files/1569707/bug-repro.zip) | @rpavlik |  | 
https://github.com/solvespace/solvespace/issues/303 | [tangent-circles-break-contour-fill.zip](https://github.com/solvespace/solvespace/files/1485089/tangent-circles-break-contour-fill.zip) | @Natronium | 2020-09-13 | https://github.com/solvespace/solvespace/pull/697 https://github.com/solvespace/solvespace/commit/e74185b63940b4b977cc2f36cc8aeb15d79ac6dc
https://github.com/solvespace/solvespace/issues/315 | [clip.zip](https://github.com/solvespace/solvespace/files/1748684/clip.zip) | @wavexx | 2020-08-13 | https://github.com/solvespace/solvespace/pull/673 https://github.com/solvespace/solvespace/commit/04b332dfd0d885a45c9bb52d3e76294a3e5eb847
https://github.com/solvespace/solvespace/issues/317 | [Archive.zip](https://github.com/solvespace/solvespace/files/1865582/Archive.zip) | @AndyPhil | ca. 2020-08-20? | 
https://github.com/solvespace/solvespace/issues/320#issuecomment-681055803 m| [320_death_star.zip](https://github.com/solvespace/solvespace/files/5131641/320_death_star.zip) | @phkahler |  | 
https://github.com/solvespace/solvespace/issues/408 | [screwdriver-sheath-with-taper.zip](https://github.com/solvespace/solvespace/files/3184113/screwdriver-sheath-with-taper.zip) | @rpavlik | between 2019-05-13 and 2020-05-08 | [between 201e15e6 and 7baf585 174 revisions](https://github.com/solvespace/solvespace/issues/408#issuecomment-648864401)
https://github.com/solvespace/solvespace/issues/470 | [revolve-extrude-union-error.slvs](https://gist.github.com/Symbian9/d38ce6749ba48c27a96fd6e9ac1ddbd3) | @Symbian9 |  | 
https://github.com/solvespace/solvespace/issues/471 <br/> https://github.com/solvespace/solvespace/issues/472 | no model <br/> no model | @phkahler <br/> @phkahler | 2020-03-27 | https://github.com/solvespace/solvespace/pull/473/ https://github.com/solvespace/solvespace/commit/f7b6f6930e06680d7429bb130a0fb5b77708bff7
https://github.com/solvespace/solvespace/pull/473#issuecomment-554161047 | [fillet.zip](https://github.com/solvespace/solvespace/files/3849334/fillet.zip) | @maz3max | Try? | 
https://github.com/solvespace/solvespace/issues/489#issuecomment-533671509 <br/> https://github.com/solvespace/solvespace/issues/550 | [model.zip](https://github.com/solvespace/solvespace/files/3637334/model.zip) <br/> [helix.zip](https://github.com/solvespace/solvespace/files/4176517/helix.zip) [gear90.zip](https://github.com/solvespace/solvespace/files/4176518/gear90.zip) | @kalle <br/> @andres-erbsen | 2020-07-01 | https://github.com/solvespace/solvespace/pull/643 https://github.com/solvespace/solvespace/commit/9802b5d1ee0d450f68f58dc5f7ad4199d8333cb1
https://github.com/solvespace/solvespace/issues/497#issuecomment-545632009 | [issue_497.zip](https://github.com/solvespace/solvespace/files/3764380/issue_497.zip) | @phkahler |  | 
https://github.com/solvespace/solvespace/issues/537 <br/> https://github.com/solvespace/solvespace/issues/537#issuecomment-669068154 <br/> https://github.com/solvespace/solvespace/issues/537#issuecomment-689079430 <br/> https://github.com/solvespace/solvespace/pull/594#issuecomment-625587508 | [jaw-.zip](https://github.com/solvespace/solvespace/files/3995496/jaw-.zip) [example2.zip](https://github.com/solvespace/solvespace/files/3995520/example2.zip) <br/> [Prisms_EarFail.zip](https://github.com/solvespace/solvespace/files/5027208/Prisms_EarFail.zip) <br/> [537_2D.zip](https://github.com/solvespace/solvespace/files/5190493/537_2D.zip) <br/> [CouldntFindAnEar_NURBSTests.zip](https://github.com/solvespace/solvespace/files/4596617/CouldntFindAnEar_NURBSTests.zip) | @rpavlik <br/> @ruevs <br/> @phkahler <br/> @ruevs | [Relevant code](https://github.com/solvespace/solvespace/blob/d72eba8039c52f5e8d3ab36ec2d8a7598dfd578b/src/srf/triangulate.cpp#L492) | 
https://github.com/solvespace/solvespace/issues/577 | wishes | | | 
m https://github.com/solvespace/solvespace/issues/632 | [texttest.zip](https://github.com/solvespace/solvespace/files/4781801/texttest.zip) | @ tjader0 | mesh | 
m https://github.com/solvespace/solvespace/issues/647 | no model | @phkahler | 2020-07-12 | https://github.com/solvespace/solvespace/pull/648 https://github.com/solvespace/solvespace/commit/c75b2e473e20499cb2775236073f6a86bf4c983c
https://github.com/solvespace/solvespace/issues/652 <br/> https://github.com/solvespace/solvespace/issues/652#issuecomment-695808103 <br/> https://github.com/solvespace/solvespace/issues/652#issuecomment-707183309  | [spherecut.zip](https://github.com/solvespace/solvespace/files/4915837/spherecut.zip) <br/> [triangle.slvs.zip](https://github.com/solvespace/solvespace/files/5252089/triangle.slvs.zip) <br/> [PrismConeNURBSNormalsTangents.zip](https://github.com/solvespace/solvespace/files/5366061/PrismConeNURBSNormalsTangents.zip) | @xzcvczx <br/> @Christoph-D <br/> @ruevs | 2020-10-12 <br/> 2020-10-13 | https://github.com/solvespace/solvespace/pull/734 https://github.com/solvespace/solvespace/commit/5d29f6533696bd5697aa90179c50609147779b88 <br/> https://github.com/solvespace/solvespace/pull/736 https://github.com/ruevs/solvespace/commit/d900578ccdc92e35cfc850ad122a6b1f0df7456b
[... cut ...]
https://github.com/solvespace/solvespace/issues/728 | [gusseted-truss.zip](https://github.com/solvespace/solvespace/files/5315184/gusseted-truss.zip) | @vcaputo | | 
https://github.com/solvespace/solvespace/pull/486 | A math improvement | @phkahler | 2019-09-20 | https://github.com/solvespace/solvespace/commit/162897eca70c1a7de02119aff596992787835555
https://github.com/solvespace/solvespace/pull/623 | A math improvement | @phkahler | 2020-06-02 | https://github.com/solvespace/solvespace/commit/94a3cfdeb74aa3cbfd91ffe26ffa57c63a1094b2

maz3max, 2020-10-20:
[image]
My error persists :/

ruevs, 2020-10-20:
It does :-)

phkahler, 2020-10-20:
> My error persists :/ (Just tried the most recent master)

@maz3max  It looks like my old comment was wrong - that was not the same issue as 315. One thing I noticed just now is that your final Lathe operation creates a cylinder coincident with the existing hole. That will cause problems. I tried changing the sketch so it was just the 2 arcs and the short segements needed to cut the fillets but that didn't work either. Also tried making the segments so they are not coincident with the hole and plane and that didn't work either. After those changes I'm not sure why it still fails. That is a difficult problem. As a work around I tried leaving out the hole extrusion and making a similar cross-section to cut the entire hole at once. I will probably open a new issue including that and another sketch I have - both work for intersection but not difference.

I really hate to pollute this long-term tracking issue with this, but here's a super cool pro-tip on sketching inside another object:
[image]
After the original extrusion I dropped a datum point on the line highlighted in yellow. Then selected a pair of lines on the end of the extrusion and that point to create a new workplane. You can slide the point off the end of the line - it is the one at the bottom of the picture - and the workplane will follow. You can constrain against the entities from the other groups even though they are behind the workplane- it will constrain projected to workplane. You can even create the lathe group out there in space and then move that original datum point back to somewhere along the edge to move the hole into position. Go back to a previous group to constrain the position of the point along that edge and you're done. Like I said earlier, in this case it does not work - the hole works as intersection but not difference. Just wanted to point out the construction method.

Also, lastly. I had one more idea. Use that technique but don't create a cutting surface coincident with the top or bottom. This is how to work around the issue without going to triangle mesh:
[437_work_around.zip](https://github.com/solvespace/solvespace/files/5411489/437_work_around.zip)
The hole can still need to be constrained in the first extrusion group.

maz3max, 2020-10-30:
Thanks for looking into that! :)

Symbian9, 2021-01-20:
Another case, inspired by ["nucleus" OpenClipArt](https://openclipart.org/detail/326932/nucleus)

- SLVS: [nucleus_slvs.zip](https://github.com/solvespace/solvespace/files/5839600/nucleus_slvs.zip)

NURBS operation says that all is `ok`, but real mesh is failed:

[image]

giannissc, 2021-04-07:
Another one causing issues
[dove-table.zip](https://github.com/solvespace/solvespace/files/6272982/dove-table.zip)

JiangXL, 2021-07-07:
Another NURBS Fail case: [NURBS_Fail.zip](https://github.com/solvespace/solvespace/files/6775127/NURBS_Fail.zip)
[image]

phkahler, 2021-07-07:
@JiangXL  That is a common failure. There are some known work-arounds. Try creating the sketch plane for the second hole by using lines that are not orthogonal to the original box - make them at some arbitrary angle. Another option that may not be OK (depends on your design requirements) is to make one hole 0.001 larger or smaller than the other.

JiangXL, 2021-07-08:
Thank for your kind reply! I decide to make one hole larger than other. I'm curious whether it is a NURBS limitation. If not, it will be so great that solvespace can solve it in the future.

phkahler, 2021-07-08:
> I'm curious whether it is a NURBS limitation. If not, it will be so great that solvespace can solve it in the future.

It is a problem in the NURBS code. A cylinder is composed of 4 surfaces of 90 degrees each.  The problem seems to be failure to find the intersection of trim curves with the edge of those surfaces. I'd like to add code to split trim curves against trim curves which I suspect will fix this issue. Either that or figure out why the curve-surface intersection is failing on those edges and find a fix for that.

Symbian9, 2021-07-11:
> It is a problem in the NURBS code. A cylinder is composed of `4` surfaces of 90 degrees each.

Could be this value increased to `8` (45° stepping) for "Lathe" operations?

Symbian9, 2021-09-27:
- SolveSpace version: `3.0~`https://github.com/solvespace/solvespace/commit/6bc63e92b0d908aa751ae06063b2ec00fec4cc57 (Debian 10.x)
- SLVS - [tube-turns.zip](https://github.com/solvespace/solvespace/files/7232521/tube-turns.zip)
[image]

phkahler, 2021-09-27:
@Symbian9 this issue is not meant to be a dumping ground for NURBS issues. @ruevs created an index to other issues.

Symbian9, 2021-09-27:
> That is a common failure. There are some known work-arounds. Try creating the **sketch plane for the second hole by using lines that are not orthogonal to the original box - make them at some arbitrary angle**.

Here is sample of such *solution* (not a bug case).

- SLVS: [cube-orthogonal-holes_slvs.zip](https://github.com/solvespace/solvespace/files/7236631/cube-orthogonal-holes_slvs.zip)

ConductiveInsulation, 2022-01-17:
I was told [here](https://twitter.com/SolveSpace/status/1483141214446698504) to post and upload it here too. The "red hole" moved by a bit when i saved the file at another angle but its still there. Dont fully understand why it's there and how to remove it

[hook.slvs.tar.gz](https://github.com/solvespace/solvespace/files/7884145/hook.slvs.tar.gz)
[image]

phkahler, 2022-01-17:
@ConductiveInsulation  Did you add the threads as a helix around an existing cylinder? If so, try having some (more) overlap so the threads cut into the cylinder an let the boolean union clean it up. That's just a suggestion to try, not a sure thing. I have a bolt model that does not use a boolean and has some red in the threads - that one is a a real mystery to me.

Symbian9, 2022-01-18:
>  I have a bolt model that does not use a boolean and has some red in the threads

@phkahler, May I take a look on this SLVS?

Symbian9, 2022-01-31:
Another one NURBS fail on intersection (not sure which issue is related here)

- [slvs3x-nurbs_cut.zip](https://github.com/solvespace/solvespace/files/7973952/slvs3x-nurbs_cut.zip)

<summary><details>
There are two variants:</details>

- Variant 1 (`slvs3x-nurbs_cut.slvs`)
[image]

- Variant 2 (`slvs3x-nurbs_cut-2.slvs`)
[image]

</summary>

Both variants resulted solids are metric equal.

phkahler, 2022-01-31:
@Symbian9  Please open new issues for new NURBS problems so they get unique issue numbers. We can add them to the table at the top of this tracker.

Also, the second version seems to work fine for me (on Windows:  Version 3.x commit 5cb299b2be)

Symbian9, 2022-01-31:
And another one NURBS fail:

- [slvs3x-hex-hole_asm.zip](https://github.com/solvespace/solvespace/files/7974122/slvs3x-hex-hole_asm.zip)

<details><summary>Sreenshot:</summary>

[image]

</details>

Instead of https://github.com/solvespace/solvespace/issues/738#issuecomment-1026173253, here I used corrected cut segment solid as segment for hexagonal solid.

Symbian9, 2022-01-31:
> Please open new issues for new NURBS problems so they get unique issue numbers. We can add them to the table at the top of this tracker.

OK.

MayeulC, 2023-08-02:
Here is my humble contribution to this list:

<details><summary>Screenshots and file</summary>

[image]
[image]

[nurbs-fail.slvs.tar.gz](https://github.com/solvespace/solvespace/files/12241698/nurbs-fail.slvs.tar.gz)

</details>

eternal-sorrow, 2025-01-13:
This kind of issue is a plague. It is what prevents SolveSpace from being a usable CAD program. Every time you use boolean difference to make some cutout in your model, you throw dice if this will happen or not. This makes exported STEP files unusable.

ruevs, 2025-01-13:
> This kind of issue is a plague. It is what prevents SolveSpace from being a useful CAD program. Every time you use boolean difference to make some cutout in your model, you throw dice if this will happen or not. This makes exported STEP files unusable.

@eternal-sorrow 
Thank you for your keen insight. It will be immensely helpful in quickly fixing all the NURBS bugs and making SolveSpace the number one CAD software - among both open and closed source competitors. Your feedback is invaluable to us.

ruevs, 2025-01-13:
Irony aside - what did you expect to achieve with your original comment? :-)

These are complex problems - they are not "design problem[s]" and in most cases do not have a "general fix (every case should be looked into separately)". They boil down to complex math, many corner cases and singularities - yes "singularities" - in the geometry the user creates.

If you are interested in the topic you can read the discussions in the liked issues - this is part of the reason this "overview issue" exists. Here are a few good starting points:
https://github.com/solvespace/solvespace/issues/470#issuecomment-527175273
https://github.com/solvespace/solvespace/issues/470#issuecomment-527175910
https://github.com/solvespace/solvespace/issues/470#issuecomment-527178752

By educating yourself you will become better at avoiding geometry in your models that hits one of these bugs. Which is not to say the bugs should not be fixed - on the contrary - however it is not an easy thing and people who have the knowledge, desire and time to do it are few.

eternal-sorrow, 2025-01-13:
>Irony aside - what did you expect to achieve with your original comment? :-)

Nothing. Just to vent my frustration that this otherwise quite a good piece of software is plagued by this kind of issues, which makes it much less useful than it could be.

phkahler, 2025-01-13:
> Nothing. Just to vent my frustration that this otherwise quite a good piece of software is plagued by this kind of issues, which makes it much less useful than it could be.

@eternal-sorrow  You are not alone. There are IMHO just a few remaining NURBS bugs - scenarios that are not handled properly or at all - causing the majority of failures.  I listed my original suspicions in a [comment](https://github.com/solvespace/solvespace/issues/297#issuecomment-1731383459) But I've since gotten more specific and have these cases as tests:

3 surfaces tangent at an edge where one is curved:
https://github.com/solvespace/solvespace/issues/1291
This should be fixed by taking curvature into account when surfaces are classified as coincident along an edge - conceptually easy but the code is complex with many special cases to handle.

Curve-curve intersection issues:
https://github.com/solvespace/solvespace/issues/1091#issuecomment-901393520
The problem here is edges that need to be split in 2 but the surfaces are tangent at the intersection - or something like that.

General edge classification bugs:
https://github.com/solvespace/solvespace/issues/1367
This may well be the root of most boolean failures with all flat faces.

What I call "proximity problems" because it may go away by changing the lengths of extrusions without changing the topology:
https://github.com/solvespace/solvespace/issues/1091
I have a feeling this one is in the same family:
https://github.com/solvespace/solvespace/issues/907

BTW, I'm not sure why I wrote this comment... It begs the question "So you (me) know what to do, what's the holdup?"

ruevs, 2025-01-14:
> BTW, I'm not sure why I wrote this comment... It begs the question "So you (me) know what to do, what's the holdup?"

The answer - for me at least - is: It takes a lot of time and "fresh brain power" to "swap in" the context of this - rather complex - code. And then the thinking/debugging is not easy. And my math is fairly rusty. As a result it takes a lot of time and brain power - which I usually either don't have or prefer to spend on something else.

nccvector, 2025-05-13:
Linking my threads here for reference:
- Problem: https://github.com/solvespace/solvespace/issues/470#issuecomment-2874785514
- Work-around: https://github.com/solvespace/solvespace/issues/470#issuecomment-2874816570

I was trying to create a T or plus shaped pipe:

Boolean operation failed:
[image]

After work-around:
[image]