solvespace/solvespace #739: The continuous integration is gone/broken

ruevs, 2020-10-15:
Since @whitequark [left](https://github.com/solvespace/solvespace/issues/714) (and even it seems [deleted his fork](https://github.com/whitequark/solvespace.git)) the AppVeyor project at https://ci.appveyor.com/project/whitequark/solvespace/build/artifacts is also gone and it used to run the Windows continuous integration builds and test suite.
Also the Mac OS target in TravisCI (https://travis-ci.org/github/solvespace/solvespace/builds/) is failing.

Can anyone familiar with these systems help? There was also a discussion of moving to GitHub actions here https://github.com/solvespace/solvespace/issues/618 @ppd  ?

griwes, 2020-10-15:
Her fork.

phkahler, 2020-10-15:
First, you should probably not tag quark with an @ any more. Someone with that level of negative feeling toward the project almost certainly doesn't want to be notified that they've been mentioned.

As for the CI, yes Appveyor is down. The configuration file is still here in the repo, but the plan was to move away from it anyway using GitHub actions as you pointed out. I have no familiarity with it and would appreciate some help. We do need builds on all 3 platforms and having CI do the sanitizers sure is nice.

There are no recent updates that should affect any platform specifically, but there are 3 remaining issues for the 3.0 milestone that are all on Windows.  All other issues that were targeted at 3.0 have either been resolved, deprioritized or move out to 4.0 (as usual there's no plan just priorities).

There are a couple NURBS fixes in progress. Then those Windows issues. and the CI change. Then we should have our 3.0-RC1.  I myself do not have a Windows machine and no familarity with GH actions or release process. I'll find my way if necessary but it could all go very quickly with some help.

ppd, 2020-10-16:
A few things about CI:

1. GitHub Actions does not support ARM64 runners (except self-hosted). Atm we're building the arm64 snap on Travis, so in case of migration to GitHub Actions, we'd need to drop that.
2. Travis could do the Windows build as well (afaik)
3. The solvespace snapcraft account should probably add @phkahler as a collaborator to regain a bit of access redundancy.
4. A key point regarding Travis vs. GitHub Actions was artifact uploading. I think that could be achieved by using Travis -> GitHub releases.
5. I haven't ever touched the OSX target. It builds on a rather old macOS & XCode version though, so investigation is necessary. Preferably from someone actually owning a Mac.

My conclusion: Our Travis integration is in OK shape and could do much of what we want. The path of least resistance would be to integrate the Windows target into it.

ruevs, 2020-10-16:
> I myself do not have a Windows machine and no familarity with GH actions or release process. I'll find my way if necessary but it could all go very quickly with some help.

I work on Windows and can (and do) build with 
- VisualStudio 2019 ( v16.7.2, C/C++ Compiler Version 19.27.29111 - can update)
- clang (currently 10.0.0 - can update) 
- MinGW (version 6.3.0 MinGW.org GCC-6.3.0-1 - can not update it)

but have no familiarity with any of the CI tools. I'll take a look at the [3.0 milestone issues](https://github.com/solvespace/solvespace/milestone/2) and see if I can help. 

> A few things about CI:
> 
>     1. GitHub Actions does not support ARM64 runners (except self-hosted). 

For Widnows we build 32 bit only.

>     5. I haven't ever touched the OSX target. It builds on a rather old macOS & XCode version though, so investigation is necessary. Preferably from someone actually owning a Mac.

I have a Mac laptop from 2008 with a 32bit MacOS ([Leopard?](https://en.wikipedia.org/wiki/Mac_OS_X_Leopard)) with 32 bit XCode... I do not think that will work - the current 64bit MacOS versions do not even support my model (I think).

phkahler, 2020-10-16:
> A few things about CI:
>     1. GitHub Actions does not support ARM64 runners (except self-hosted). Atm we're building the arm64 snap on Travis, so in case of migration to GitHub Actions, we'd need to drop that.

I'd like to keep ARM. In fact, I want it with OpenMP and LTO to get the most out a Raspberry Pi. Why drop something that works... BTW do you have any stats on how many people use the ARM version?

>     2. Travis could do the Windows build as well (afaik)

Good. Using one system for everything seems better IMHO. Also see point 5 below.

>     3. The solvespace snapcraft account should probably add @phkahler as a collaborator to regain a bit of access redundancy.

Ooof. I already feel like I'm left holding the bag here, but I understand the concern.

>     4. A key point regarding Travis vs. GitHub Actions was artifact uploading. I think that could be achieved by using Travis -> GitHub releases.

I don't really want debug builds to be generally available anyway. Making a proper 3.0 release available will eliminate the need for people to have the latest from master. I like that a number of people build from source, but having the non-developer masses use "nightly" builds has already caused problems.

>     5. I haven't ever touched the OSX target. It builds on a rather old macOS & XCode version though, so investigation is necessary. Preferably from someone actually owning a Mac.

Well there was a bunch of effort here:  https://github.com/solvespace/solvespace/pull/517  Not sure why that stalled. Maybe @vespakoen is stilll willing to offer some help here.

> My conclusion: Our Travis integration is in OK shape and could do much of what we want. The path of least resistance would be to integrate the Windows target into it.

I'm all for "least resistance" in this area ;-)

On a related note, someone once said releases will not be built with OpenMP because it would break the Windows experience of downloading a single executable file. OpenMP adds a dependency on a DLL that Windows doesn't ship with, so I'm asking @jwesthues  what do you think? I agree that the single file model is awesome, but what about pointing to the required installer and saying "you'll need to install this first" ? Or what if the Windows build just don't use it?

BTW,  If I sound like someone in charge of the project that's just my desire for this thing to have an official update after 4 years. My level of contribution is going to decline very soon but I'll still be around, there are still a few things I'd like to add.

ppd, 2020-10-16:
> I'd like to keep ARM. In fact, I want it with OpenMP and LTO to get the most out a Raspberry Pi. Why drop something that works... BTW do you have any stats on how many people use the ARM version?

I'm not involved with the snap's administration as I'm not listed as a collaborator. The snap's ownership was transferred last year: https://github.com/solvespace/solvespace/issues/482 Therefore, I can't look up the statistics. @jwesthues should be able to do so.

> Good. Using one system for everything seems better IMHO. Also see point 5 below.

I'll play around with it.

> Ooof. I already feel like I'm left holding the bag here, but I understand the concern.

Well, you're the only organization member aside from @jwesthues.

> Well there was a bunch of effort here: #517 Not sure why that stalled. Maybe @vespakoen is stilll willing to offer some help here.

Thanks for the reminder.

jwesthues, 2020-10-16:
@ppd I added you back to the snap; feel free to decline if you don't want the access of course. @phkahler Shall I add you too?

ARM is 5/1684 snap downloads for the most recent month, small share. I agree that it's interesting beyond that small share though. No strong feelings on the single-file distribution. As you say, you could distribute with the DLL, or disable OpenMP just under Windows by default (or switch to a non-Microsoft compiler, which I believe permits static linking?), and any of those options seems reasonable to me.

ppd, 2020-10-16:
> @ppd I added you back to the snap; feel free to decline if you don't want the access of course.

[image]

That didn't work. I think we've just found a bug.

jwesthues, 2020-10-16:
@ppd Ugh. I guess only the Ubuntu people can fix this? As a workaround I could try adding to youremail+something@domain, but then you'd have two Ubuntu accounts.

ppd, 2020-10-16:
@jwesthues No worries, I'm typing the bug report as we speak (write?). This is probably going to be resolved quite soon. It's not terribly urgent anyway.

vespakoen, 2020-10-16:
Hey, I worked on macOS Catalina support and the Travis build.

I'd happily take some time to bring it up to date again and the offer for sharing my macOS Developer account is still up.

The macOS build works great, i've been using it a lot over the last year.

You can send me an email at hello [at] koenschmeets.nl for the macOS certificate (I need a developer account for my own projects for the foreseeable future anyways).

I can also make a fork that does the macOS voodoo with a Github action with a cronjob or something like that...

phkahler, 2020-10-17:
@vespakoen  Thank you!  Please issue whatever PRs you need. I wasn't sure of the status on the pending one, so I'm happy to merge it if it's ready. I'll take the cert discussion over to email.

ppd, 2020-10-17:
@vespakoen Getting the OSX build going again for the test suite would already be a very nice thing. The signing process is more relevant for stable/beta releases as far as I understand, isn't it? Could a competent user still manually force macOS to run an unsigned app for testing purposes?

phkahler, 2020-10-17:
@jwesthues  It looks like the CLA signing check is gone as well. The only checks being done on PRs are Travis builds.

jwesthues, 2020-10-17:
Unless anyone else feels otherwise, I'm good just eliminating the CLA. In the past, we'd occasionally licensed either my 2.0 release or a special clean branch with Linux support that whitequark prepared under commercial terms. The goal of the CLA was to eventually make that possible for our main development branch too, to avoid maintaining that special clean branch, and to give ourselves the option to move to a different free software license (e.g., LGPL) later if desired. This wouldn't be possible now, because that main development branch also contains changes from before the CLA.

But disentangling the main development branch would have been a lot of trouble before, and now it would be yet more. So I'm inclined just to say this codebase is GPLv3-only forever, and avoid the distraction of a CLA.

phkahler, 2020-10-18:
Does the CLA help to make clear that contributions are GPLv3? It seems like a decent thing to have even if it didn't allow later relicensing. IANAL but some form of ack that the contributor knows what they're doing is nice.

OTOH there's a chinese translation waiting on the CLA which may not happen.

jwesthues, 2020-10-18:
Just by sending us the contribution, the contributor "convey[s] a work based on the Program, or the modifications to produce it from the Program, in the form of source code under the terms of section 4" (quoting from the GPLv3). At that point, either they must license that contribution under the GPLv3, or they're violating the license of the original work. So even if a confused contributor somehow later wants to rescind their contribution, I believe the GPLv3 alone means there's negligible risk, even with no CLA.

A typical CLA does contain terms where the contributor promises that the contribution is their own work, or that they otherwise have the right to license it. So that CLA could give us a cause of action against someone who submitted code that violated a third party's copyright; but a cause of action against some random, possibly-pseudonymous stranger on the Internet probably isn't worth much. There's also usually patent terms, but that's unlikely to be relevant except for big corporate contributors (which we've never had; to be clear, I mean contributions by the big corporation itself, not contributions in their off hours by people who also happen to work for a big corporation).

So my suspicion is that it's not worth the trouble and barrier to entry, unless we think we'd ever go back and clean up the pre-CLA contributions (which I doubt). No strong feelings, though.

ppd, 2020-10-18:
The snap store shenanigans are filed here:
https://forum.snapcraft.io/t/i-cant-be-re-added-as-a-collaborator-after-having-been-removed-before/20622
https://bugs.launchpad.net/snapstore-server/+bug/1900179

jwesthues, 2020-10-18:
@ppd So maybe the snap store issue was my fault; on closer inspection, there's a separate path to add new contributors vs. re-enabling old ones. I just tried the latter, and maybe that resolves?

ppd, 2020-10-19:
@jwesthues that worked

phkahler, 2020-10-26:
So are we good to close this one?

ppd, 2020-10-26:
Can we consider this solved? CI is back and far better than it's ever been. Further improvements and additional build jobs are beyond the scope of this dash to 3.0.

closed 2020-10-26 by phkahler