solvespace/solvespace #744: migrate to travis-ci.com before 31-12-2020

ppd, 2020-10-17:
Travis-ci.org will shut down on 31-12-2020: https://mailchi.mp/3d439eeb1098/travis-ciorg-is-moving-to-travis-cicom

Migrating to travis-ci.com is easy, free and also gives us a more robust (in my tests) arm64 target: arm64-graviton2 in a VM instead of an LXD-container.

Migration howto: https://docs.travis-ci.com/user/migrate/open-source-repository-migration
Example build on arm64-graviton2: https://travis-ci.com/github/ppd/solvespace/jobs/401308051

ppd, 2020-10-17:
So I can't get the arm64 builds on LXD to fail any more, which is good. Sometimes it truncates the log, which is obviously bad. However, arm64 on VM is coming to bionic images, and that env seems to be more stable regarding log truncation.

It seems already a lot better than the arm64 arch on travis-ci.org, and it will surely improve in the future (either we move to focal images, or they bring VM support to bionic images). :tada:

phkahler, 2020-10-17:
Better to move sooner, not later. Did your access to Travis get worked out?

ppd, 2020-10-17:
Are you thinking of the snapcraft collaboration problem? I think that has to wait till after the weekend. 

I've just migrated my fork to travis-ci.com to play around with the new arm64 options.

ppd, 2020-10-19:
@phkahler If you do the migration (log in & click migrate), I can push the switch to `arm64-graviton2`.

ppd, 2020-10-19:
Alright. I think we're almost set for the migration if we so desire.

Required steps:

1. As I'm not an owner (or admin in Travis lingo) of the SolveSpace organization, I need @phkahler to push a button for use:
At https://travis-ci.com/organizations/solvespace/migrate click `Sign up for the beta`, add `SolveSpace` in the drop-down box and click `Join the beta`.
2. We (I) transfer the `solvespace` repository to travis-ci.com. All secrets should migrate automatically.
3. Rejoice and throw some confetti.

phkahler, 2020-10-19:
That shoud be a GO.

ppd, 2020-10-19:
Oh no. I promised too much. Seem like you need to go all the way: 
[image]

phkahler, 2020-10-19:
Would have been nice if they popped that up before :-) should be good now.

ppd, 2020-10-19:
Perfect. Looks good and builds. All secrets seem to be there. If it doesn't explode in our faces in the next few days, this can be closed.

closed 2020-10-19 by phkahler